linuxboot/heads #931: Flashrom non-reproducible

tlaurion, 2020-12-14:
@Tonux599 @Thrilleratplay @MrChromebox : comparing [master](https://github.com/osresearch/heads/commit/a81ae6ed5b7068b6b5752f6856fa29e69e7a6bdd) build vs local build:
```
user@x230-master:~$ git clone https://github.com/osresearch/heads heads2
user@x230-master:~$ cd heads2
user@x230-master:~/heads2$ make BOARD=x230-hotp-maximized CPUS=2
user@x230-master:~/heads2$ wget https://277-64810881-gh.circle-artifacts.com/0/build/x230-hotp-maximized/hashes.txt -O remote_hashes.txt
user@x230-master:~/heads2$ egrep '^[0-9a-f]{64}' build/x230-hotp-maximized/hashes.txt | while read line; do HASH_REF=$(echo $line|awk -F " " {'print $1'}); FILE_REF=$(echo $line|awk -F "/" {'print $NF'}); if ! grep -q "$HASH_REF" ./remote_hashes.txt; then echo "$FILE_REF doesn't match";fi; done
tools.cpio doesn't match
flashrom doesn't match
gawk doesn't match
```

_Originally posted by @tlaurion in https://github.com/osresearch/heads/issues/927#issuecomment-744457179_

MrChromebox, 2020-12-15:
able to extract the flashrom binary and vbindiff it with your locally built one? comparing one built within Heads to my local host shows that there are some embedded library paths that differ, so not hugely surprising to see it differ between a local Heads instance and a CI one

osresearch, 2021-03-02:
Is `flashrom` still a problem?  I just did a build on Fedora 33 and Ubuntu 20.04 and produced the same results.  It looks like thi smight have been fixed in aba13a9c55b4c90a026299fad28892cd9861d2c0 by pinning the module.

tlaurion, 2021-06-15:
Not in latest tests! Thanks @osresearch !

closed 2021-06-15 by tlaurion