linuxboot/heads #935: Parallel make doesn't work on a clean checkout

osresearch, 2020-12-16:
There are some components that do not correctly depend on the cross compiler, so a make for the clean checkout fails.  Easy fixes are the various tools in `$(COREBOOT_UTIL_DIR)` like `cbmem`, `superiotool` and `inteltool`, which need to wait until the cross compiler is available.

bwachter, 2021-01-04:
I'm currently running into this trying to build one of the x230 pull requests. -j isn't propagated to the subdirectories, and setting MAKE_JOBS breaks bootstrapping of the toolchain. Without anything set everything runs as -j1, though, which takes ages. 

At least some of that seems to have been introduced in the past 6 months - I remember having some initial issues to make it use parallel builds back in summer, but once I had that working it took about 5 minutes for a full build (including bootstrapping the tool chain). So far I've been sitting on this now for about 1 hour, and still didn't get past the tool chain bootstrapping.

tlaurion, 2021-01-05:
@bwachter documentation has been updated [here](https://github.com/osresearch/heads-wiki/commit/fbf11667a8a2becdf05439754d18c2df871ffdea), rendered [here](http://osresearch.net/general-building/), to specify that CPUS=YY should be specified on the make BOARD=XXX statement

I'm still a bit confused here, since the main Makefile of Heads is taking [the ouptut of nproc if not specified and populates CPUS variable which is then passed along if not defined on make initial call](https://github.com/osresearch/heads/blob/master/Makefile#L33).

That hack was implemented, because some modules won't like to have -j forced, where others play along. The idea here was to pass CPUS down in other makefiles (modules/*) where they play fair, and go single threaded for when modules don't play well.

@osresearch?

osresearch, 2021-03-02:
The issue with `CPUS=xxx` on the make command line is that it does not spawn multiple top-level jobs.  So while the individual components might be built in parallel, there is no parallelism across the modules.  On a rebuild after a `make real.clean` (so the cross compilers are intact) only one module is built at a time.  On my build machine this makes the difference between a 90 second rebuild and a much longer process process.

```
make real.clean && time make V=1 CPUS=128
...
real	30m2.657s
user	25m18.422s
sys	7m0.333s
```

versus

```
make real.clean && time make V=1 -j128
...
real	1m27.378s
user	30m26.307s
sys	6m12.245s
```

tlaurion, 2021-03-23:
@osresearch The problem with -j24 is that there is no locking between interdependent tasks. 

[Here](https://github.com/osresearch/heads/pull/977#issuecomment-789178534), I launch the same build with [CPUS=24(functional)](https://app.circleci.com/pipelines/github/tlaurion/heads/712/workflows/b6621766-09fc-4ec3-aa3f-ade2e0c12b97/jobs/854) and [-j24(failing badly with files missing)](https://app.circleci.com/pipelines/github/tlaurion/heads/713/workflows/25dabad5-9275-497a-9fb8-3912df0f4f90/jobs/855) over CircleCI.

The [line](https://app.circleci.com/pipelines/github/tlaurion/heads/713/workflows/25dabad5-9275-497a-9fb8-3912df0f4f90/jobs/855/parallel-runs/0/steps/0-109) `rm -rf build/x230-hotp-maximized/* build/log/* && make -j24 V=1 BOARD=x230-hotp-maximized || touch /tmp/failed_build` in CircleCI permits all logs created to be outputed in next task on CI, which otherwise would not be readable since -j24 ruins the output.

So the next task picks up created logs and outputs them on CI with delimitators: `if [[ -f /tmp/failed_build ]]; then find ./build/ -name "*.log" -type f -mmin -1|while read log; do echo ""; echo '==>' "$log" '<=='; echo ""; cat $log;done; exit 1;else echo "Not failing. Continuing..."; fi`

The [resulting logs are concatenated here](https://app.circleci.com/pipelines/github/tlaurion/heads/713/workflows/25dabad5-9275-497a-9fb8-3912df0f4f90/jobs/855/parallel-runs/0/steps/0-110) with ==> and <== separators.

tlaurion, 2021-03-23:
@osresearch : i'm redoing a build [without decompressing your cache file](https://github.com/tlaurion/heads/commit/9f6e86d078571d056a3566d3b2f8b9c672f1e878), which [failed on CircleCI](https://circleci.com/gh/osresearch/heads/311?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link) for your latest commit, [since host binaries were not found](https://circleci.com/api/v1.1/project/github/osresearch/heads/311/output/108/0?file=true&allocation-id=603f7df086b146495ebac997-0-build%2F181CF9F5) on fed config.status (sed, gawk and other host binaries were not found on provided paths).

A CI build is happening [here](https://circleci.com/gh/tlaurion/heads/858?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link)

tlaurion, 2021-11-30:
@osresearch seems like the missing culprit was in commenting MAKE_JOBS under global Mafefile on your PR?

Then a weird race condition happens only under newt module build, which can be hacked by forcing that module to be built only with one job. Will do a replacement PR once working around new CircleCI limitations pass.

tlaurion, 2021-12-01:
@osresearch @bwachter : I was successful into fixing partly this issue in that commit: https://github.com/osresearch/heads/pull/1015/commits/5e4309c1e19b58cb6c2ca9cddd11893c83afaa71#diff-a7aa45cf31f3d34533d95a154a86338c1a99e7134864e2f65834c6ecf3497e00R12-R14

Where MAKE_JOBS was commented out into the main Makefile.
Improvements welcome.

tlaurion, 2022-02-05:
Tag me to reopen. CircleCI builds in parallel with 36 cores now under 45 minutes on clean checkout.

closed 2022-02-05 by tlaurion