linuxboot/heads #937: Can't see console output

jtmoree-github-com, 2020-12-20:
I am having problems with heads in a pureboot setup on a librem 15.  I see messages on the console but then the graphical environment hides them and I cannot find a way to get to the messages.  Even if there is a way to use the recovery shell it would be better if the console allowed me to see the messages in real time.

DETAILS:  I have just re-installed the OS and trying to get the system to boot.  I reset the TPM owner in the process.  

At the PUreBoot Boot Menu:

Default boot)  States that I dont have a default boot so I choose YES: show me a menu.    Then it fails quickly and flashes stuff on the console before going back to the PureBoot Boot Menu where I can choose default boot again and repeat the whole thing.

After a few flashes I can see there is a BAD signature in the text.  OK.  I'll try to Refresh TOTP/HOTP but it does the same thing.  fails and goes back to boot menu.  I have a system that is making it really hard to fix itself.

tlaurion, 2020-12-20:
@jtmoree-github-com : your system doesn't initially have signed /boot detached signature after reboot.
Please go to options, and sign /boot content. You are basically in a state where your system has been tampered with, by you. :)

jtmoree-github-com, 2020-12-20:
@tlaurion   sorry.  got distracted before i finished the post.  I tried to sign files and it does the same thing: fails and flashes past the messages.

jtmoree-github-com, 2020-12-20:
The only place that waits for me to see the console is after trying to generate a new TOTP secret.  That's probably because of the QR code that is displayed.  OP is about the fact that I cannot see messages.  I'm going to tackle the specific problems in other forums.

jtmoree-github-com, 2020-12-20:
I just realized that this may not be the right forum for this post.  I am using Pureboot which may be a fork of this heads project.  Is this project using the same graphical interface as PureBoot from Librem?  That support forum seems to be here: https://source.puri.sm/coreboot/pureboot/-/issues

MrChromebox, 2020-12-22:
@jtmoree-github-com sounds like you're using an older version of Pureboot which has a bug when trying to sign the default OS selection. Update to the latest release and try again please

jtmoree-github-com, 2020-12-23:
I just updated coreboot/heads at the beginning of this ordeal.   pureboot-librem_15v3-Release-15.1.rom.gz

jtmoree-github-com, 2020-12-23:
I just looked in the documentation (wiki included) for a recovery shell reference.  If I can manually run the same commands that the gui is running I can see all of the messages.  Didn't this project only have a command line two years ago?  where are the docs for that?

MrChromebox, 2020-12-23:
gui-init was created on 20 Feb 2018, so almost 3 years ago. Documentation is linked from the wiki tab on the project here.

jtmoree-github-com, 2020-12-23:
wiki link in source code is broken.  wiki link from this project goes to http://osresearch.net/.  I can't find any references to gui-init in the search on that site or in the menu on the left of that site.

jtmoree-github-com, 2020-12-24:
I have parsed out these commands from the wiki docs and the recovery shell.  Is there no command line reference for this project?  
- kexec
- kexec-boot
- kexec-insert-key
- kexec-iso-init
- kexec-parse-bls
- kexec-save-default
- kexec-save-key
- kexec-seal-key
- kexec-unseal-key
- kexec-select-boot
- kexec-sign-config
- kexec-check-config (not found in my heads environment)

As mentioned in the OP I want to run everything in the recovery shell so I can see all the output for debugging.  I have tried to figure out the order to run various steps based on the wiki.

I could really use a troubleshooting doc.  The wiki refers to choosing menu entries in some cases which correspond to the gui-init but other times references pressing keys such as 'd' but I dont know what environment that is referring to.  Perhaps that is all in the qubes section.

tlaurion, 2020-12-26:
@jtmoree-github-com actually, there isn't. The project forked its way from generic-init to gui-init based on Purism main effort, where generic-init id now considered legacy.

There is an issue for documentation, which is seperated from this project, under heads-wiki, rendered over osresearch.net. 

Any help welcome.

jtmoree-github-com, 2020-12-31:
I started a fork to add to the FAQ.  Will continue to research and fill information before making a pull request.  https://github.com/osresearch/heads-wiki/compare/master...jtmoree-github-com:patch-1

I see 3 doc issues in heads-wiki...digging through them.  Will work on them as I study the heads boot process.

jtmoree-github-com, 2021-01-10:
I may have found a bug as I'm tooling around in the recovery shell.  Inside of gui-init the default boot is set to the correct partition.  When I use the recovery shell and type 'mount' it shows a different partition mounted on /boot.   I have changed boot setting and flashed the bios multiple times and restarted multiple times.  Each time the default boot in gui-init says the correct one and the recovery shell has the wrong one.

If they can/should be different I'll add it to docs.  If it's a bug I'll open a new issue.  Any thoughts?

MrChromebox, 2021-01-10:
what do you mean 'right one' and 'wrong one'? you're going to need to be more specific. 

What is CONFIG_BOOT_DEV set to when building?
via config?
how are you accessing the recovery shell?

jtmoree-github-com, 2021-01-10:
- right one = the partition that actually has files for booting
- wrong one = any other partition
I'm not building heads.  I'm using pureboot on a Librem 15.  It is broken.  I'm troubleshooting in the recovery shell by using 'r' during boot and choosing it in the gui-init menus.  (I have already asked for help from purism.  It has been slow and painful and stalled--through no fault of anyone.  This is unsupported software after all.)

For an example: in the recovery shell environment  **CONFIG_BOOT_DEV=/dev/sda1**  but in gui-init
`  Options => Change Configuration Settings => Change the Boot Device`
shows '**Currently set to /dev/sda2.**'

Should they be the same on a fresh boot?  I'm assuming both come from the flashed BIOS image.

MrChromebox, 2021-01-10:
> right one ... wrong one...

I'm trying to understand if it's automatically picking another device/partition at boot, as part of gui-init

> I'm using pureboot on a Librem 15

version? Release 15.1 is the latest, and is displayed via the 'System info' option from the main menu

> This is unsupported software after all.

no, Pureboot is supported

> For an example ...

So, when you boot to the recovery console via 'r' at boot, it completely skips the gui-init and applying any user-saved configs, and so `echo CONFIG_BOOT_DEV` will show the default set when building (/dev/nvme0n1p1).

Now, as part of the GUI init, Pureboot will attempt to mount CONFIG_BOOT_DEV and locate grub.cfg. If grub.cfg can't be found on CONFIG_BOOT_DEV, then Pureboot will iterate thru all available internal storage devices (NVMe and SATA) in an attempt to locate a valid boot partition. It will save that to CONFIG_BOOT_DEV but not to flash. And it won't apply if you drop to a recovery shell from the GUI, since it's a separate environment

jtmoree-github-com, 2021-01-10:
Librem 15 v3
| Pureboot will iterate thru all available internal 
That explains the wrong setting.  I have multiple partitions with linux kernels in them.  It's grabbing the first one which is the wrong one.
If the environment will not record the correct partition in flash where can I put it? Is heads/pureboot ONLY ever going to auto boot the first partition with linux kernels?

tlaurion, 2021-01-10:
@MrChromebox : I am not sure I follow why recovery shell is having a different "environment" since it is supposed to use defined configuration, where overrides can be saved in conf and flashed back into the ROM?

https://github.com/osresearch/heads/blob/master/initrd/etc/functions#L299-L344

MrChromebox, 2021-01-10:
if you set the proper boot device/partition and save it to flash, that's the default used on the next boot. The only time it should iterate thru other options is if the default fails to mount or if  /grub*/grub.cfg cannot be located after mounting

MrChromebox, 2021-01-10:
@tlaurion because the environment from which gui-init is run is not persisted when dropping to the recovery shell. Any config changes made that were not applied from the user's config at boot (such as gui-init's overriding of CONFIG_BOOT_DEV via detect_boot_device) would not persist.

jtmoree-github-com, 2021-01-10:
> if you set the proper boot device/partition and save it to flash, that's the default used on the next boot. The only time it should iterate thru other options is if the default fails to mount or if /grub*/grub.cfg cannot be located after mounting

So it is a bug.  I'll report it.  https://github.com/osresearch/heads/issues/959

MrChromebox, 2021-01-11:
> So it is a bug. I'll report it. #959

it's not going to be one I can reproduce here without significant input from you on how your partitions are set up and where your grub.cfg is located. Because you can see the logic is exactly as I described in the link tlaurion provided above

jtmoree-github-com, 2021-01-15:
Update.  The 'bug' described above seems to be a misunderstanding of requirements.  Closed that issue and will add notes to the docs I am working on.  Since the last post I have failed to get pureos to install using LUKS but have installed Kubuntu with LUKS.  I am still stuck in the loop of heads not configuring or booting and still not able to read all of the messages on the console when it fails and still slowly working my way through all of the individual commands for configuring and booting with heads in the recovery shell.  Maybe I'll change the title of this issue to 'recovery shell needs documentation'

jtmoree-github-com, 2021-01-16:
This thread (https://github.com/osresearch/heads/issues/639#issuecomment-570014587) says that the secrets from nvram are wiped before the recovery shell is launched.  Does this limit what the recovery shell can do?  Can the secrets be recovered using my other security infrastructure for validating that each step of the configuration and boot is working?
p.s.  I can boot Kubuntu using kexec-boot but heads fails to parse the kernels to pick one for booting.  then it wants to restart the process of owning the tpm and generating secrets.  That seems like the wrong action to take since I just did that 30 seconds ago.  This is where I always get stuck and why I am trying to run the entire boot process in the recovery shell.

MrChromebox, 2021-01-16:
> This thread (#639 (comment)) says that the secrets from nvram are wiped before the recovery shell is launched

correct, the TPM PCRs are cleared so they cannot be read / used to generate the secret from the recovery shell

> I can boot Kubuntu using kexec-boot but heads fails to parse the kernels to pick one for booting

fails how?

> then it wants to restart the process of owning the tpm and generating secrets

you mean when booting from the GUI? The normal process is to validate the boot state on each loop thru the gui-init script, which re-cycles when boot fails if not dumped to recovery shell.

jtmoree-github-com, 2021-01-16:
| fails how?

That's the problem.  The gui asks if I want to 'load a menu of boot options' or return to the main menu.   After choosing  Yes (menu) it then flashes the console by so quickly I cannot see the errors.  The next screen is the gui telling me that it could not generate totp code and that I should generate new secrets or reset the tpm.  "ERROR: PureBoot couldn't generate the TOTP code."

What I think is going on is that grub.cfg is missing which is leading to other missing items such as hotp/totp.  I have been troubleshooting by following the instructions from the GUI about secrets and TPM but the root cause is not being identified.

If the grub.cfg is missing could that be detected earlier in the configuration (such as when signing files in boot)?  That would help identify the problem earlier.  Are there other required files and devices that we should check for?  I can code a dependencies checker.

MrChromebox, 2021-01-16:
>  The next screen is the gui telling me that it could not generate totp code and that I should generate new secrets or reset the tpm. "ERROR: PureBoot couldn't generate the TOTP code."

is that after having successfully verified the firmware on boot (or having previously generated new secrets)?

> If the grub.cfg is missing could that be detected earlier in the configuration (such as when signing files in boot)? That would help identify the problem earlier. Are there other required files and devices that we should check for? I can code a dependencies checker.

grub.cfg is checked for as part of the automatic /boot device identification, but only if mounting the default or user-set CONFIG_BOOT_DEV fails. Otherwise there's no check until Heads attempts to boot, IIRC

jtmoree-github-com, 2021-01-18:
I'll list every action I take from power on.  maybe that will make things more clear.  I start with a system that has a gray background and end with one that has a red background on the next reboot.

1. poweron with key inserted
2. HTOP verification success displayed on screen
3. Automatic boot unless key press countdown
3. No Default Boot Option Configured ... load a menu of boot options?.... => YES
4. ERROR:  PureBoot coudn't generate the TOTP code...
5. Generate HTOP/TOTP secret
6. ...erase your old secret.... => YES
7. Unable to unseat HOTP secret; Press Enter.... => Enter
8. ERROR:  PureBoot coudn't generate the TOTP code...

jtmoree-github-com, 2021-01-18:
Can we add something like this to gui-init?

``` 
BOOTFILE="/boot/grub.cfg"

# check for CRITICAL file /boot/grub.cfg
if [ ! -f "${BOOTFILE}" ] ; then
    whiptail $BG_COLOR_WARNING --clear --title "WARNING:  Missing file" \
      --menu "WARNING:  Could not find file for boot options: '${BOOTFILE}'" 90 4 \
      'i' ' Ignore error and continue to main menu' \
      'x' ' Exit to recovery shell' \
      2>/tmp/whiptail || recovery "No '${BOOTFILE}'"
fi
```

I have tried this in my heads install and it does not work.  Seems like I'm not passing params to whiptail properly.

MrChromebox, 2021-01-18:
I think there's quite a few things we can do to improve gui-init, starting with removing the looping checks, but also including checking for grub.cfg at the start of the default boot process. I'll try to put together a PR this week