linuxboot/heads #939: qemu-linuxboot: x86_64-linux-musl-gcc cannot find Linux headers when building

synackd, 2020-12-24:
When attempting to build the `qemu-linuxboot` target on an Ubuntu 18.04 docker container, the following error occurs:

```
$ make BOARD=qemu-linuxboot
[...]
FV Space Information
SECFV [10%Full] 212992 total, 21616 used, 191376 free
FVMAIN_COMPACT [4%Full] 7634944 total, 354192 used, 7280752 free
DXEFV [10%Full] 10485760 total, 1069000 used, 9416760 free
PEIFV [20%Full] 917504 total, 185768 used, 731736 free

- Done -
Build end time: 23:50:12, Dec.23 2020
Build total time: 00:00:19

cp edk2/Build/OvmfX64/DEBUG_GCC5/FV/OVMF.fd boards/qemu/qemu.rom
( \
cd build/qemu ; \
/home/docker/workspace/heads/build/linuxboot-git/bin/extract-firmware \
        -o rom \
) < boards/qemu/qemu.rom \
> build/qemu/qemu.txt.tmp
mv build/qemu/qemu.txt.tmp build/qemu/qemu.txt
./bin/create-fv \
        -v \
        -o build/qemu/user.vol \
        --size 0x048000 \
        --compress 0 \
        FORCE
Can't open FORCE: No such file or directory at ./bin/create-fv line 64.
build/qemu/user.vol: 0x00000000 out of 00048000 bytes in FV
./bin/create-ffs -o build/qemu/DxeCore.ffs --name DxeCore --version 1.0 --type DXE_CORE --depex "" --guid "D6A2CB7F-6A18-4E2F-B43B-9920A733700A" edk2/Build/MdeModule/DEBUG_GCC5/X64/DxeCore.efi
/home/docker/workspace/heads/build/make-4.2.1/make -C dxe linuxboot.ffs
make[3]: Entering directory '/home/docker/workspace/heads/build/linuxboot-git/dxe'
/home/docker/workspace/heads/crossgcc/bin/x86_64-linux-musl-gcc -std=c99 -D__efi__ -DGNU_EFI_USE_MS_ABI -fshort-wchar -mno-red-zone -fno-stack-protector -m64 -fpic -O3 -W -Wall -I . -I efi/x86_64 -MMD -MF .linuxboot.o.d    -c -o linuxboot.o linuxboot.c
In file included from linuxboot.c:23:
/home/docker/workspace/heads/crossgcc/x86_64-linux-musl/include/asm/bootparam.h:35:10: fatal error: linux/types.h: No such file or directory
 #include <linux/types.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [<builtin>: linuxboot.o] Error 1
make[3]: Leaving directory '/home/docker/workspace/heads/build/linuxboot-git/dxe'
make[2]: *** [Makefile:144: dxe/linuxboot.ffs] Error 2
make[2]: Leaving directory '/home/docker/workspace/heads/build/linuxboot-git'
```

I assume that it is supposed to use the headers included with the downloaded/built kernel and not the ones on the system.

synackd, 2020-12-24:
The checkout is clean except for a small change made manually in `modules/linuxboot` that changes `$$CPUS` to `$$(CPUS)` on line 24 (addressed in #845).

cjia4, 2021-09-10:
Meet the same issue.  

>>>

ASH 7a6a3fe5a4045f9ff57727fca3ccd63599435d44 dirty qemu-linuxboot
2021-09-10 10:34:21+08:00 CPIO      build/qemu-linuxboot/tools.cpio
2021-09-10 10:34:21+08:00 UNCHANGED build/qemu-linuxboot/tools.cpio
a2b778ec3b9ae543c2d9f3173a58eec3d0a14320a0a63385fb2963ac687b081e  /home/abc/linuxboot/heads/build/qemu-linuxboot/tools.cpio
2021-09-10 10:34:22+08:00 HASHES    build/qemu-linuxboot/tools.cpio
2021-09-10 10:34:22+08:00 CPIO      build/qemu-linuxboot/heads.cpio
2021-09-10 10:34:22+08:00 UNCHANGED build/qemu-linuxboot/heads.cpio
2e06c89e72267ad15c11765dc129479e1062f07d49dbf88e7834ff5da57829a1  /home/abc/linuxboot/heads/build/qemu-linuxboot/heads.cpio
2021-09-10 10:34:22+08:00 HASHES    build/qemu-linuxboot/heads.cpio
2021-09-10 10:34:22+08:00 MAKE linuxboot
tail /home/abc/linuxboot/heads/build/log/linuxboot.log
-----
make[4]: Leaving directory '/home/abc/linuxboot/heads/build/linuxboot-git/edk2/Build/MdeModule/DEBUG_GCC5/X64/MdeModulePkg/Universal/RegularExpressionDxe/RegularExpressionDxe'

- Done -
Build end time: 10:38:32, Sep.10 2021
Build total time: 00:00:40

make[3]: Leaving directory '/home/abc/linuxboot/heads/build/linuxboot-git/edk2'
./bin/create-ffs -o build/qemu/DxeCore.ffs --name DxeCore --version 1.0 --type DXE_CORE --depex "" --guid "D6A2CB7F-6A18-4E2F-B43B-9920A733700A" edk2/Build/MdeModule/DEBUG_GCC5/X64/DxeCore.efi
/home/abc/linuxboot/heads/build/make-4.2.1/make -C dxe linuxboot.ffs
make[3]: Entering directory '/home/abc/linuxboot/heads/build/linuxboot-git/dxe'
/home/abc/linuxboot/heads/crossgcc/bin/x86_64-linux-musl-gcc -std=c99 -D__efi__ -DGNU_EFI_USE_MS_ABI -fshort-wchar -mno-red-zone -fno-stack-protector -m64 -fpic -O3 -W -Wall -I . -I efi/x86_64 -MMD -MF .linuxboot.o.d    -c -o linuxboot.o linuxboot.c
In file included from linuxboot.c:23:
/home/abc/linuxboot/heads/crossgcc/x86_64-linux-musl/include/asm/bootparam.h:35:10: fatal error: linux/types.h: No such file or directory
 #include <linux/types.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [<builtin>: linuxboot.o] Error 1
make[3]: Leaving directory '/home/abc/linuxboot/heads/build/linuxboot-git/dxe'
make[2]: *** [Makefile:144: dxe/linuxboot.ffs] Error 2
make[2]: Leaving directory '/home/abc/linuxboot/heads/build/linuxboot-git'
make[1]: *** [Makefile:455: /home/abc/linuxboot/heads/build/linuxboot-git/build/qemu/.build] Error 1
make[1]: Leaving directory '/home/abc/linuxboot/heads'
Makefile:668: recipe for target 'all' failed
make: *** [all] Error 2

cjia4, 2021-09-10:
There are install directory under heads root directory: <heads>/install/include/linux/types.h
Pointing 'include' to that directory will fix this issue.   Not sure why it is missing.  For someone who meet similar issue, following workaround could work:

1.Edit build/linuxboot-git/dxe/Makefile and add one more line
git diff dxe/Makefile 
diff --git a/dxe/Makefile b/dxe/Makefile
index 6d0c25b..a803f4f 100644
--- a/dxe/Makefile
+++ b/dxe/Makefile
@@ -61,6 +61,7 @@ CFLAGS += \
        -Wall \
        -I . \
        -I efi/x86_64 \
+       -I ../../../install/include/ \
        -MMD \
        -MF .$(notdir $@).d \

tlaurion, 2021-09-12:
Can somebody post a clean working PR adressong pinning of linuxboot to a commit ID and patches under patches as addressed in #845?

tlaurion, 2021-10-22:
@cjia4 @synackd ?

saper, 2023-03-27:
I realized today that host `#include <linux/xxx.h>` headers are being used. This might mean we might need to pre-install the Linux headers for the desired kernel version before the kernel is built (for the modules that need them).