linuxboot/heads #951: xx30-flash boards produces two roms

tlaurion, 2021-01-04:
Only x230-flash.rom and t430-flash.rom should be produced.

Thrilleratplay, 2021-01-04:
@tlaurion If someone has a unlocked IFD and is currently running coreboot or Skulls, the 12mb rom could be used to install Heads via internal Flash.  Unless the xx30 non-flash boards include a 12mb internally flashable rom.

tlaurion, 2021-01-04:
> @tlaurion If someone has a unlocked IFD and is currently running coreboot or Skulls, the 12mb rom could be used to install Heads via internal Flash. Unless the xx30 non-flash boards include a 12mb internally flashable rom.

@Thrilleratplay : Correct.
That would be as simple as calling flashrom without specifying the BIOS region, as specified into maximized BOARD flashrom instruction.

BUT! this is where those boards (xx30, x230-flash) are different (apart of tools encompassed). The FLASHROM command board configuration specifies to flash only the BIOS region. Which is consistent for users not having unlocked/modified IFD to take advantage of the freed ME space.

IF IFD is unlocked, a user could flash the whole SPI internally with maximized boards made roms.

Thrilleratplay, 2021-01-04:
@tlaurion After the discussion on Slack and realization that the `xx30` Heads BIOS spans two chips, the scenario I suggested is marginal and not worth the headaches leaving the 12mb `xx30-flash` rom cause.

tlaurion, 2021-01-04:
@Thrilleratplay @MrChromebox @Tonux599  ? Agreed with the rm of full rom in the board config? Have not came with anything better.

Thrilleratplay, 2021-01-04:
@tlaurion I can't think of a better solution either.  It is easy enough to reverse if needed.

Tonux599, 2021-01-04:
> @Thrilleratplay @MrChromebox @Tonux599 ? Agreed with the rm of full rom in the board config? Have not came with anything better.

Seems sensible if end user is only expected to use the 4mb rom and as stated avoids confusion.

tlaurion, 2021-01-04:
Will wait for cache from new cache system to be created and merge later.

PR #952 (2021-01-05): xx30-flash boards: produce top.rom and remove 12mb rom for clarity
Produces only a 4mb top rom following actual name standard as other boards.

Fixes #951 (not so elegant. If you have better idea, please advise)
closed 2021-01-05 by tlaurion