linuxboot/heads #959: default boot not saved

jtmoree-github-com, 2021-01-10:
PROBLEM
I have multiple partitions with linux kernels in them and multiple distros installed.  At some point heads stopped recording the default boot properly.  Maybe after I updated to the last release? pureboot-librem_15v3-Release-15.1.rom.gz

I have repeatedly booted the heads environment and navigated to the default boot and reset it, then told the system to save the settings to flash.   When the system boots again, the default boot in the gui is set correctly to the one I chose last time but if I drop to a recovery shell a different partition is mounted and specified in the environment variable CONFIG_BOOT_DEV.

STEPS TO SEE PROBLEM
*) power on
*) at the gui main menu explicitly set the boot device to /dev/sda2
`Options => Change Configuration Settings => Change the Boot Device`
*) now save the configuration
`Save the configuration to the running BIOS`
Warning: Setting BIOS Control at 0x8b to 0x89 failed.
`This will reflash your BIOS with the updated version.  Do you want to proceed?`
*) Choose Yes
`BIOS updated successfully....Enter to reboot`
*) OK/Enter
*) system restarts and shows red background.  "PureBoot couldn't generate the TOTP code."  Done this dozens of times.  It does not  work.
*) Exit to recovery shell 
# mount
/dev/sda1 /boot type ext4 (ro,relatime)
# export | grep CONFIG_BOOT_DEV
export CONFIG_BOOT_DEV=/dev/sda1
*) run gui-init again to see 'ERROR: Pureboot couldn't generate the TOTP code'
`Ignore error ... => Options => Change Configuration Settings => Change the Boot Device`
*) still set to the correct partition /dev/sda2

jtmoree-github-com, 2021-01-10:
BTW.  /dev/sda2 on my system is a stand alone fully functioning distro (pureos 9) with no encryption (/boot embedded).  I can boot it manually using kexec and kexec-boot.    

I can get the default boot in the Boot Device selection to 'stick' and 'show up' in the environment with some other partitions--just not /dev/sda2.  

I can't get pureboot working with any of them.  I can't get signing of files to work with any of them.  I can sometimes get HOTP/TOTP functions to work--other times not.  I can almost always reset the TPM.

MrChromebox, 2021-01-11:
does /grub*/grub.cfg exist on /dev/sda2? If not, then it's not compatible with Heads/Pureboot, and the firmware is working as designed

jtmoree-github-com, 2021-01-11:
Sounds like heads/pureboot requires a separate boot partition?  Did I miss that in the docs? 

I can symlink grub in /dev/sda2 but the sig process will not really work as designed.

MrChromebox, 2021-01-11:
> Sounds like heads/pureboot requires a separate boot partition? Did I miss that in the docs?

yes

jtmoree-github-com, 2021-01-11:
Where in the docs?  Just looked again.   If not, I'll add it as I work on them.

MrChromebox, 2021-01-11:
good question, IDK offhand. Heads has always used/required a dedicated, unencrypted /boot partition (and assumed a LUKS-encrypted root /) AFAIK

jtmoree-github-com, 2021-01-15:
I am digging through and reworking docs.  As part of that I will add notes on the requirement for a separate boot partition.  Considering this issue closed.

closed 2021-01-15 by jtmoree-github-com