linuxboot/heads #963: "fdisk -l" behavior changed under busybox bumped to 1.32

tlaurion, 2021-01-17:
As reported [here](https://github.com/osresearch/heads/pull/957#issuecomment-758269030)

Seems like the fix is to cut the output as in here https://github.com/osresearch/heads/pull/955/files#diff-0fd54c445ad40620392abb0e6defb6974df6798254e326594d6300cc86058e24R315

@MrChromebox ?

tlaurion, 2021-01-17:
```
user@x230-master:~/heads$ grep -Rn fdisk initrd/
initrd/bin/config-gui.sh:34:      fdisk -l | grep "Disk" | cut -f2 -d " " | cut -f1 -d ":" > /tmp/disklist.txt
initrd/bin/gui-init:434:      --msgbox "${BOARD_NAME}\n\nFW_VER: ${FW_VER}\nKernel: ${kernel}\n\nCPU: ${cpustr}\nRAM: ${memtotal} GB\n\n$(fdisk -l | grep -e '/dev/sd.:' -e '/dev/nvme.*:' | sed 's/B,.*/B/')" 16 60
initrd/etc/functions:315:	fdisk -l | grep "Disk" | cut -f2 -d " " | cut -f1 -d ":" > /tmp/disklist

```

MrChromebox, 2021-01-17:
@tlaurion right now, there are two separate issues with the output of fdisk due to busybox bump:
* output of disk sizes in System Info function of gui-init
* output of disks available for boot device selection in /etc/functions

PR #955 addresses the latter. The former is purely cosmetic, and as the output is providing the size in both sectors MB, would require a little more massaging to cut/reformat in GB as before

closed 2022-07-14 by tlaurion