linuxboot/heads #971: ASSERT error when running linuxboot.rom in QEMU

aaldhaher, 2021-02-03:
I get the following ASSERT error when trying to run qemu-linuxboot built using HEADS with kernel version 4.14.62:

```
[...]
+--------------------+
|                    |
| Starting LinuxBoot |
|                    |
+--------------------+
num tables=0005
49AA7D31FC1BCDB0
42146F0205AD34BA
FvLoader: adding firmware volume 0xFF500000
FvLoader: error rc=0000000A
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
InstallProtocolInterface: 665E3FF6-46CC-11D4-9A38-0090273FC14D 7B2E408
LinuxBoot: waiting for BDS callback
[...]
--------
Transfering control to BDS 7B2E408
Entry Point 7B2C280 (7B2C280)
--------
LinuxBoot: BDS time has arrived
LinuxBoot: connect pci root brdiges
efi_visit_handles 2F707EBB
handle_count=00000001
PCI Bus First Scanning
PciBus: Discovered PCI @ [00|00|00]

PciBus: Discovered PCI @ [00|01|00]
   BAR[0]: Type = PMem32; Alignment = 0xFFFFFF; Length = 0x1000000;     Offset = 0x10
   BAR[2]: Type =  Mem32; Alignment = 0xFFF;    Length = 0x1000;        Offset = 0x18

PciBus: Discovered PCI @ [00|02|00]
ASSERT [PciHostBridgeDxe] <workpath>/linuxboot/edk2/MdePkg/Library/BasePciCf8Lib/PciCf8Lib.c(1114): ((Address) & (~0xffff0ff | (3))) == 0
```

It also occurs when running `make BOARD=qemu` in the LinuxBoot project itself and using the bzImage and initrd artifacts from the HEADS build.

It looks like the assertion is made during the call to `gBS->ConnectController()` in `efi_connect_controllers()` on line 170 of dxe/linuxboot.c

Has anyone else run into this problem before?

synackd, 2021-02-03:
Looks like you copied and pasted this from [my Linuxboot issue](https://github.com/linuxboot/linuxboot/issues/46). This isn't an issue with HEADS, but with Linuxboot.

aaldhaher, 2021-02-03:
Sorry about that; I wasn't sure which side of the coin the problem lies on. But it sounds like you've had success rolling back to an older commit in Linuxboot. Thanks for the tip!

tlaurion, 2021-02-04:
@aaldhaher would you mind specifying the latest commit that builds with heads so we can revert until the upstream problem is figured out? Not testing linuxboot at all here, personally.

aaldhaher, 2021-02-04:
@synackd was kind enough to point out that commit https://github.com/linuxboot/linuxboot/commit/f0ea0af8609e235631f1fcb65a4f7eb69f28ea26 seems to work with the kernel+initrd for the qemu-linuxboot board in HEADS.