linuxboot/heads #979: x230 Bottom Chip

greasy-crow, 2021-02-25:
I am currently in the middle of building Heads for my thinkpad x230. I have experience with coreboot and skulls already, but they did not play well with qubes so I decided to give heads a go. As of now, I have built the x230-flash rom, and went through the process of cleaning the ME, but I have not flashed the clean rom to the bottom chip yet. I am a little confused on what I am supposed to do with the 12mb rom produced in the building process. The instructions on the Heads wiki say to flash the full 12mb rom from the recovery shell, but how do I access the rom once I have booted into recovery? I am planning on using a raspberry pi for flashing, as I did with coreboot and skulls. If somebody could sort of walk me through this last step, it would be greatly appreciated.

tlaurion, 2021-02-26:
After having cleaned me and flashing it back to bottom spi chip, you should flash the 4MB x230-flash.rom to top spi flash.

Then put the 12MB x230 board compiled rom onto a USB stick and flash it per on screen instructions.

Other board which does it all for you is to build x230-maximized, and flash top and bottom spi chips with properly named rom images completely externally. 

Hope that helps.

I would also like to know which part of osresearch.net instructions were not clear enough, directly from heads-wiki github repo.

greasy-crow, 2021-02-26:
@tlaurion Thank you, that really clears things up. I was following the instructions at https://osresearch.net/x230-flashing/ and the only part I had issues with was the last paragraph:

> If all goes well, you should see the keyboard LED flash, and within a second the Heads recovery splash screen will appear. It currently drops you immediately into the shell, to allow you to flash the full 12MB x230.bin Heads rom (or build/x230/coreboot.rom if you’ve built it locally). If it doesn’t work, well, sorry about that. Please let me know what the symptoms are or what happened during the flashing.

It does not say anything about a USB drive, and I had not gotten to that step of the installation so I was a little confused. The rest was very clear though.

tlaurion, 2021-02-26:
@greasy-crow don't hesitate to reopen this issue if you feel the documentatio nis not precise enough from https://github.com/osresearch/heads-wiki/commit/ef46a10c28f25a3260f24646a4bad85772d1659f changes.

closed 2021-02-26 by tlaurion