linuxboot/heads #980: GPG and USB boot issues after flashing Heads on x230

greasy-crow, 2021-02-27:
After flashing heads on my thinkpad x230, I am unable to boot from a USB, install an ISO, or add pgp keys. 

Initially after the install, I tried to generate GPG keys on a Yubikey 5 security key. I successful generated my keys by following the prompts on the GPG Management Menu, however, no USB devices or keys were recognized when I went back to the main menu to add the keys, and subsequent attempts produced the same result. 

I am also unable to install Qubes from a USB drive. I have attempted to use both a bootable installer as well as an ext4 formatted USB containing the Qubes ISO and signature, but neither of those methods were successful. When I attempt to run the bootable installer, the boot process gets stuck at "kexec_core: Starting new Kernel," and when I try to boot from the USB drive containing the ISO and signature,  the boot process halts after attempting to load the kernel, and returns to the recovery shell. I am not able to export a log, but I can type out the errors at a later time if necessary. I tried to boot from a Tails USB as well, but the process gets stuck at "Starting Gnome Display Manager."

One other minor issue that I've encountered is that the keyboard will not work after rebooting from the heads recovery shell. There is an open issue for it, #765, although it does not appear that any resolution was reached. This is an issue that I can definitely live with though.

To my knowledge, I did not make any mistakes while building heads, and I tried to reflash it after I first encountered these errors, but they still persist.

greasy-crow, 2021-03-01:
Update: I realized that I had some missing dependencies when I first started the building process, so I installed the dependencies and rebuilt the 4MB rom, the 8MB rom with the stripped ME, and the full 12MB rom. 

After re-flashing my bios with the newly built roms, I was able to add my GPG key by exporting it to the same USB that I had my 12MB rom stored on. I have setup TOTP as well. 

The keyboard issue has also been resolved: I discovered that if I unplug my USB devices before rebooting from the recovery shell, the keyboard on the x230 will work once it starts back up. 

The installation and USB boot issues are unfortunately still occurring. I have tried to install Qubes using a bootable USB, and also boot from a Tails live USB, but each time the boot process gets stuck at "kexec_core: Starting new kernel." I have also attempted to install Qubes using USB drive containing the Qubes ISO and the gpg signature, but when I do that, the ISO verification fails. The error message it gives is as follows:
`gpg: BAD signature from "Qubes OS Relase 4 Signing Key" [ultimate]
ISO signature failed
!!!!! Something failed during USB boot`

It then returns to the recovery shell. I have verified the  ISO and the release signature on my other computer, and it is valid, so I'm not sure why the verification is failing.

I would try skipping the installation process and installing Qubes to an SSD using another computer and then signing the boot files with my gpg key on heads, but I am worried about future errors and am unsure whether it would boot anyway. At this point I don't know whether this is user error or an issue with heads, but either way, some help would be greatly appreciated.

tlaurion, 2021-12-04:
#1015 merged. Please reopen if still present

closed 2021-12-04 by tlaurion