linuxboot/heads #983: Fedora 33 IASL build failure

osresearch, 2021-03-03:
```
/build/make-4.2.1/make 
/home/hudson/heads/./build/make-4.2.1/make -C "/home/hudson/heads/build/coreboot-4.8.1" CPUS=128 iasl
make[1]: Entering directory '/home/hudson/heads/build/coreboot-4.8.1'
Welcome to the coreboot cross toolchain builder v1.52 (June 11th, 2018)

Building toolchain using 128 thread(s).

Downloading and verifing tarballs ...
 * acpica-unix2-20180531.tar.gz (cached)... hash verified ("17717140438d506533b4a56e34350749d7b84d6c")
Downloaded tarballs ... ok
Unpacking and patching ...
Unpacked and patched ... ok
Building packages ...
Building IASL v20180531 for host ... failed. Check 'build-IASL/build.log'.
make[2]: *** [Makefile:36: build_iasl] Error 1
make[1]: *** [util/crossgcc/Makefile.inc:52: iasl] Error 2
make[1]: Leaving directory '/home/hudson/heads/build/coreboot-4.8.1'
make: *** [modules/coreboot:79: "/home/hudson/heads/build/coreboot-4.8.1/util/crossgcc/xgcc/bin/iasl"] Error 2
```

and the IASL build log:

```
make[1]: Entering directory '/home/hudson/heads/build/coreboot-4.8.1/util/crossgcc/acpica-unix2-20180531/generate/unix/iasl'
- Link obj/iasl
/usr/bin/ld: obj/aslcompilerparse.o:(.bss+0x8): multiple definition of `AslCompilerlval'; obj/aslcompilerlex.o:(.bss+0x0): first defined here
/usr/bin/ld: obj/prparserlex.o:(.bss+0x0): multiple definition of `LexBuffer'; obj/dtparserlex.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [../Makefile.rules:15: obj/iasl] Error 1
make[1]: Leaving directory '/home/hudson/heads/build/coreboot-4.8.1/util/crossgcc/acpica-unix2-20180531/generate/unix/iasl'
make: *** [Makefile.common:7: iasl] Error 2
cp: cannot stat 'bin/iasl': No such file or directory
```

osresearch, 2021-03-03:
Appears to have been fixed in a more recent version (26 March 2020): https://acpica.org/sites/acpica/files/changes_50.txt

> iASL: table compiler: make LexBuffer static to avoid linking errors in newer compilers. Patch based on suggestions by David Seifert and Benjamin Berg.

itworksfinebro, 2021-03-15:
I have experienced this issue over 4 different platforms, fedoras and debians as well.
Can confirm that substituting the following in the `build/coreboot-4.8.1/util/crossgcc/buildgcc` file fixes the issue and allows for compilation:
```
< IASL_VERSION=20210105
---
> IASL_VERSION=20180531
75c75
< IASL_ARCHIVE="http://acpica.org/sites/acpica/files/acpica-unix-${IASL_VERSION}.tar.gz"
---
> IASL_ARCHIVE="http://acpica.org/sites/acpica/files/acpica-unix2-${IASL_VERSION}.tar.gz"
100c100
< IASL_DIR="acpica-unix-${IASL_VERSION}"
---
> IASL_DIR="acpica-unix2-${IASL_VERSION}"
```

The repo change to unix from unix2, highlights the changes they have done to their dual license policy.

closed 2022-02-05 by tlaurion

tlaurion, 2022-02-06:
Fixed upstream, where debian-11 is including further fixes to make sure we use the same make gawk and output what is used and propagated in Heads builds. Please tag me to reopen.