linuxboot/heads #985: Documentation fails to explain SRWD issue: hardware write protect is currently (mostly) not possible

ln2max, 2021-03-03:
The Heads documentation [suggests](https://osresearch.net/Heads-threat-model/#system-firmware):
>Finally, once Coreboot has been flashed into the ROM, the write protect pins on the ROMs can be shorted to ground as an extra layer of protection. This prevents any software re-writes of the ROM, even from the Management Engine or other devices on the SPI bus.

However, this is not as easy as it sounds. Most Thinkpad chips [require extra SRWD bits to be set when flashing](https://github.com/flashrom/flashrom/issues/185) in order for the hardware write protect pins to have any effect. Flashrom doesn't currently support setting these bits, see the [linked issue](https://github.com/flashrom/flashrom/issues/185) for a more in-depth description of what is needed (and a patch for Flashrom).

tlaurion, 2021-06-17:
@ln2max thanks

jtf7, 2024-08-18:
### Hardware Write Protection (KGPE-D16)
A number of users have had issues getting this working so here are some tips and using hardware write protection

_**Hardware write protection is only available on eeprom chips that explicitly support it and typically requires setting specific fuse settings appropriately to enable this function.**_   This information is found in the manufacturer’s documentation for the specific part # of chip you intend to use.

### Winbond W25Q128FVAIG
The Winbond W25Q128FVAIG is a 16MB eeprom chip capable of hardware write protection and is compatible with the KGPE-D16. This chip is physically engraved with the abbreviated text 25Q128FVIG. 

**_Fuse Settings_**
Fuse settings required to activate the hardware write protection feature for this specific part number of Winbond chip are as listed below: 

[S18=WPS=0]
[S14=CMP=0]

[S9=QE=0]
[S8=SRP1=0]
[S7=SRP0=1]

[S6=SEC=0]
[S5=TB=0]
[S4=BP2=1]
[S3=BP1=1]
[S2=BP0=1]

Experts can tweak these settings to fine tune the chip’s configuration however, the settings listed above have worked well for me and should work well for most users.

Connecting pin 3 (/WP) and pin 4 (GND) together with these fuse settings will enable hardware write protection. It may be required to power off the system before connecting/disconnecting these pins in order for the correct write state to be recognized.

Many users have a similar version of Winbond chip with part number W25Q128FVAIQ however, it should be noted that this alternate Q version chip is not capable of hardware write protection. 

**_Additional Fuse Settings: Security Register Lock Bits_**
You may also want to set S11, S12, and S13 as listed below in order to permanently lock the security register as read-only. This is where manufacturer’s info, security related information, and hard coded data for the chip is stored. This is a separate location on the chip from where your ROM image is stored. The vibe I got from reading the Winbond documentation is that this could be a great place for backdoor related shenanigans and it’s probably a good idea to set these fuses in order to lock as read-only. That being said, I’m not entirely sure. I set mine as listed below and it worked for me however, as always, use at your own risk.

[S13=LB3=1]
[S12=LB2=1]
[S11=LB1=1]

tlaurion, 2024-08-18:
@ln2max @jtf7 are all those bits set properly with patches having landed to flashrom?

Docs at https://docs.dasharo.com/variants/asus_kgpe_d16/spi-wp/#flashrom-write-protection-cli

jtf7, 2024-08-18:
> @ln2max @jtf7 are all those bits set properly with patches having landed to flashrom upstream? Should have landed under 1.3.0 and ones missing there, present in 1.4.0 which was recently released.
> 
> Docs at https://docs.dasharo.com/variants/asus_kgpe_d16/spi-wp/#flashrom-write-protection-cli
> 
> CC @SergiiDmytruk [flashrom/flashrom#185 (comment)](https://github.com/flashrom/flashrom/issues/185#issuecomment-2106390030)
> 
> If so, it would be a good time to open an issue under heads-wiki, close [flashrom/flashrom#185](https://github.com/flashrom/flashrom/issues/185) and close this issue leaving documentation to be drafted and merged referencing original docs, not redoing them, or, point upstream what is missing for flashrom to be feature complete on SPI WP.

The bits I provided should all be correct. When I did this, I went through the Windbond 25Q128FV datasheet (Revision I) extensively and verified multiple times due to the importance I see in having HW write protection working properly. I hired someone to do the custom flashing and so I'm not familiar with the flashrom patches or whats going on there although I'll take a look and try to comprehend.

In the Dasharo link you provided they deliberately only write protect a portion of the chip, the settings I provide HW write protect all the memory blocks. @tlaurion --Can you confirm first that it isn't necessary for the boot functionality of Heads to have part of the memory in read-write mode and that keeping it fully locked as read-only the majority of the time is ok?

tlaurion, 2024-08-18:
> In the Dasharo link you provided they deliberately only write protect a portion of the chip, the settings I provide HW write protect all the memory blocks. 

This is unfortunately needed; how would RAM init cache be written back to SPI? How would you update Heads? 
 
>@tlaurion: Can you confirm that it isn't necessary for the boot functionality of Heads to have part of the memory in read-write mode and that keeping it fully locked as read-only the majority of the time is ok? The way they wrote that article leads me to believe they had some legitimate reason for this.

Absolutely. 
- Heads is coreboot measured boot https://doc.coreboot.org/security/vboot/measured_boot.html +
- Heads seals/unseals two different secrets based on measured boot (bootblock up to payload: Heads). Draft for PR: https://github.com/linuxboot/heads-wiki/issues/116 and https://github.com/linuxboot/heads-wiki/issues/62
  - One (always enforced) for firmware integrity attestation (HOTP/TPMTOTP) in TPM nvram (sealing of basic measurements)
  - A second when one setup a TPM Disk Unlock Key in TPM nvram (sealing of advanced measurements)
- Bootblock is SPI root of Trust (RoT in firmware, not CPU/Chipset as bootguard) : https://github.com/linuxboot/heads/pull/1568#issuecomment-2284151446

TLDR: only the bootblock should be write protected under Heads, so that TPM TCPA/TPM event log (measured boot enforced by coreboot) can be meaningful/helpful in tamper evidence. Setting write protection for any other region would prevent upgrade, which might be useful for your TV or any other device you wish to protect against upgrades and tampering, otherwise you simply prevent the SPI from being effectively writeable, if that is your desire, but this is normally NOT desired, hence why this was pushed forward for joint NlNet application and funded work under https://github.com/linuxboot/heads/issues/1741.

From those reads, anything missing for documentation?

Note that for KGPE-D16, which was directly targeted for funding and grant PoW, public direct outcomes were:
- https://shop.3mdeb.com/shop/adapters/asus-kgpe-d16-flash-chip-adapter/
- https://docs.dasharo.com/variants/asus_kgpe_d16/spi-wp/#flashrom-write-protection-cli

jtf7, 2024-08-18:
> Setting write protection for any other region would prevent upgrade, which might be useful for your TV or any other device you wish to protect against upgrades and tampering, otherwise you simply [...]

I'm not sure I'm following.  I will try to comprehend better when I'm a little more refreshed.

I was testing other things awhile back and drifted away from using the Heads TPM and Nitrokey features to boot. Am working on reinstalling and getting everything set up properly now. Then planing to do a Qubes HCL and help document to lower the learning curve for newbies.

For KGPE-D16:  My original plan was to leave the pins 3+4 jumpered to keep the Winbond chip HW write-protected for the vast majority of the time and then in those rare times I need to update Dom0 or firmware or change some Heads setting, I would disconnect the jumper in order to allow writes and make changes. After changes are made, I would rejumper pins 3+4 to HW protect the entire chip again.  Are you saying that doing this will prevent Heads from measuring the TPM and operating as intended because it is necessary for Heads to write something to Winbond in order to accomplish this??

tlaurion, 2024-08-19:
> ### Hardware Write Protection (KGPE-D16)
> A number of users have had issues getting this working so here are some tips and using hardware write protection
>
> _**Hardware write protection is only [...]

@jtf7 my question around this was more in the real of: are those needed if using flashrom and wp whole SPI?
Do you still need to apply those bits manually or flashrom takes care of it correctly?
CC @ln2max

SergiiDmytruk, 2024-08-19:
> Do you still need to apply those bits manually or flashrom takes care of it correctly?

This is what those flashrom changes do (except for security/OTP registers which aren't related to WP and I don't know of software that even uses them).  I guess @jtf7 just applied the same configuration without flashrom (must be via some other tool).

tlaurion, 2024-08-19:
> This is what those flashrom changes do (except for security/OTP registers which aren't related to WP and I don't know of software that even uses them). I guess @jtf7 just applied the same [...]

@SergiiDmytruk OTP in the sense that writing to those registers make the SPI chip WP the regions specified only once and cannot be undone?

SergiiDmytruk, 2024-08-19:
> @SergiiDmytruk OTP in the sense that writing to those registers make the SPI chip WP the regions specified only once and cannot be undone?

No, it doesn't interact with WP.  Security registers exist separately from the main flash memory and provide small (in this case it's 3 registers with 256 bytes in each) amount of memory that can be made read-only with those LB1-LB3 bits which are OTP.  So you write something in that register (like a unique ID), set corresponding LB bit and that register can never be changed again.

jtf7, 2024-08-29:
> does that clarify my previous post and referenced content?

Yes, thank you for taking the time to elaborate so thoroughly. All of the technical explanations I've read over the years from yourself and others such as the Qubes devs have been very helpful in gaining a better understanding of firmware and software security.

jtf7, 2024-08-29:
> Security registers exist separately from the main flash memory and provide small (in this case it's 3 registers with 256 bytes in each) amount of memory that can be made read-only with those LB1-LB3 [...]

It's probably best to lock these bits permanently if you are able to, though to be clear this recommendation is more based on my intuition, gut feelings, and street smarts of my experience and knowledge of backdoors, in-general best engineering practices, and technical workings of this Winbond chip as opposed to hard documented rules and facts. 

The Winbond SPI firmware and unique identifiers that could potentially tag a chip are stored in this section. We know persistently infected firmware has become more and more of an issue and this section of the SPI chip would be a great place to infect if you were someone looking to infect the device persistently. Permanently locking this section is probably a good idea if practical, especially since its not like the end user will ever have a legitimate need to update this section. And even if there was a reason the end user needed to update in the future, the chips are only a few dollars to replace.

jtf7, 2024-08-29:
> > **_Additional Fuse Settings: Security Register Lock Bits_** You may also want to set S11, S12, and S13 as listed below in order to permanently lock the security register as read-only. This is where manufacturer’s info, security related information, and hard coded data for the chip is stored. This is a separate location on the chip from where your ROM image is stored. The vibe I got from reading the Winbond documentation is that this could be a great place for backdoor related shenanigans and it’s probably a good idea to set these fuses in order to lock as read-only. That being said, I’m not entirely sure. I set mine as listed below and it worked for me however, as always, use at your own risk.
> > [S13=LB3=1] [S12=LB2=1] [S11=LB1=1]
> 
> @jtf7 my question around this was more in the realms of: are those needed if using flashrom and wp whole SPI with flashrom 1.3+?

This is a separate section of the chip. It's geared more toward the SPI chips admin and operation than user data. I would consider this section to be critical to security despite not necessarily being used by the end user.

I've added this below, does this explanation help?

> It's probably best to lock these bits permanently if you are able to, though to be clear this recommendation is more based on my intuition, gut feelings, and street smarts of my experience and [...]

jtf7, 2024-08-29:
> my question around this was more in the realms of: are those needed if using flashrom and wp whole SPI with flashrom 1.3+?

A couple of the fuse settings I specified are listed as "x" in the documentation and could technically be set to either 1 or 0. Despite this, I decided to explicitly define the fuses for simplicity and since it's possible there may be some "gotchas" that could apply when using alternate settings. (Using alternate settings for these pins could possibly change other settings, you'd need to read through the documents in full if for some reason you wanted to do this.)