linuxboot/heads #988: extract.sh oversight

shamen123, 2021-04-10:
[https://github.com/osresearch/heads/blob/master/blobs/xx30/extract.sh](https://github.com/osresearch/heads/blob/master/blobs/xx30/extract.sh)

>echo "###Unlocking $bioscopy IFD..."
>$IFDTOOL -u $bioscopy
>echo "###Extracting regions from ROM..."
>$IFDTOOL -x $bioscopy
>echo "###Copying GBE region under $BLOBDIR/gbe.bin..."
>cp "$extractdir/flashregion_3_gbe.bin" "$BLOBDIR/gbe.bin"
>echo "###Applying me_cleaner to neuter+deactivate+maximize reduction of ME on $bioscopy, outputting minimized ME under $BLOBDIR/me.bin and adapting BIOS+ME regions under $BLOBDIR/ifd.bin... "
>$MECLEAN -r -t -d -O /tmp/unneeded.bin -D "$BLOBDIR/ifd.bin" -M "$BLOBDIR/me.bin" "$bioscopy"

@tlaurion The unlocked version after -u would be $bioscopy.new 

While on this topic - where in your view is the best place in a board build to automate any blob building scripting?

tlaurion, 2021-04-11:
@shamen123 you're right. CircleCI builds for qemu first, which makes ifdtool and me_cleaner in searched paths found.

I remember a discussion that happened and guideline to build only coreboot buildstack but notthing to have coreboot build only ifdtool and dowload me_cleaner, so there is a catch 22 here as exposed under #989.

Basically, the current process requires to build the board two times. One will fail at the moment of including expected blobs. Then calling the extract.sh script (since then, coreboot ifdtool and me_cleaenr downloaded, so both scripts are in path) will succeed. Sorry about that, will try to see a solution, but that would be the workaround for now for #989 .

shamen123, 2021-04-11:
@tlaurion i think im good for now I reworked download_clean_me.sh into an overall solution not reliant on the coreboot build path for p8h61.  Ill see if i can do the same for xx30 and do a pull req. 

Re this issue itself, I think can be closed. $IFDTOOL -u $bioscopy is actually redundant in the code (as unlocking happens at end of full build anyway). So it doesnt matter that me_cleaner uses $bioscopy and not the unlocked $bioscopy.new

basically this line $IFDTOOL -u $bioscopy makes a file that is not referenced nor used.

closed 2021-04-11 by shamen123

tlaurion, 2021-04-11:
Might be an error still, or code should be modified after verification. Eg:
https://github.com/osresearch/heads/blob/afa6753a3038eb111c31c314f0f84f1136f540dd/blobs/x220/extract.sh#L60

>as unlocking happens at end of full build anyway
Which is coincidental. coreboot unlocking prior of integrating blob ifd? To be verified.

tlaurion, 2021-04-11:
@shamen123 : I have verified. This is an error, since extract.sh is supposed to unlock.
`$IFDTOOL -x $bioscopy.new` is supposed to be there and do the work.

When calling https://github.com/osresearch/heads/blob/master/blobs/xx20/download_parse_me.sh or https://github.com/osresearch/heads/blob/master/blobs/xx30/download_clean_me.sh , the ifd.bin in tree is used for the finalized oreboot build.

So here, my intuition tells me (unconfirmed) that calling https://github.com/osresearch/heads/blob/master/blobs/xx30/extract.sh#L60 or https://github.com/osresearch/heads/blob/master/blobs/xx20/download_parse_me.sh (which never extracts ifd.bin, just reuses what is in tree) are not working.

To be tested.

tlaurion, 2021-04-11:
@shamen123 reopened https://github.com/osresearch/heads/issues/933