linuxboot/heads #994: not working because of iasl/multiple definition of Asl Compiler Val; Counter Forensics Feature

wyvern-fairlight, 2021-05-10:
hi, i was going to contribute my first feature, a category of increasingly advanced "counter forensics" techniques, which can optionally be enabled, as subentries, of the Options >> menu. i'm being stopped by something called iasl preceeded by a duplicate reference warning/error in the buildchain. kindly support on getting up to speed with the compile chain / elaborate on why clandestine features would not be desired.
[image]
[image]

tlaurion, 2021-05-25:
@wyvern-fairlight @Buom01 have you fixed this issue? Seems like host dependency **_iasl_** was not found on system per required host dependencies needed on debian-10:
`apt install -y build-essential zlib1g-dev uuid-dev libdigest-sha-perl libelf-dev bc bzip2 bison flex git gnupg iasl m4 nasm patch python python2 python3 wget gnat cpio ccache pkg-config cmake libusb-1.0-0-dev autoconf texinfo ncurses-dev doxygen graphviz udev libudev1 libudev-dev automake libtool rsync innoextract`

Please close this issue if resolved.

tlaurion, 2021-06-15:
The host dependency was not installed

closed 2021-06-15 by tlaurion

AldenIsMe, 2021-06-21:
> @wyvern-fairlight @Buom01 have you fixed this issue? Seems like host dependency **_iasl_** was not found on system per required host dependencies needed on debian-10:
> `apt install -y build-essential [...]

Still having this issue, I ran the command you said, it installed a few extra depends and still comes to same error, I'm trying to compile on Kali Linux WSL.