linuxboot/heads #995: Better document limitations of platforms under #692 and board configs

tlaurion, 2021-05-13:
Example: [no microcode updates for Sandy/older boards](https://software.intel.com/security-software-guidance/processors-affected-transient-execution-attack-mitigation-product-cpu-model) and their consequences, without going into unneeded details.

Find a way to explain succinctly side channels attacks, software mitigation, microcode non-existing/existing mitigations, QubesOS `smt=off` disablement (hyperthreading) and expected behaviors of such CPU/Memory intensive gadgets impacts.

Mitigations/Opsec in users behaviors and user choices need to make between blob absence, performance, security, owner controllable hardware, microcode availability etc. In the goal of enforcing proper behaviors to mitigate the risks (QubesOS DispVM usage in case of untrusted, short term activities and better understanding/enforcing the concept of "No need to protect an information that is not in memory".

Any idea on wording, ways of covering this, welcome!

natterangell, 2021-06-14:
Hi,

Since you haven't received a lot of suggestions, here's a start. Not sure if this explains the risks in enough details:

With the disclosure of the Spectre and Meltdown vulnerabilities in January 2018, it became apparent that most processors manufactured since the late 1990s can potentially be compromised by attacks made possible because of [transient execution CPU vulnerabilities](https://en.wikipedia.org/wiki/Transient_execution_CPU_vulnerability). Modern CPUs utilise speculative execution in order to increase performance. A branch misprediction may leave observable side effects that may reveal private data to attackers using a timing attack. Future not-yet-identified vulnerabilities of this kind is likely. For users of Qubes OS, this class of vulnerabilities can additionally compromise the enforced isolation of virtual machines, and it is prudent to take the risks associated with these vulnerabilities into account when deciding on a platform on which to run Heads and Qubes OS.

Mitigation of these vulnerabilities is achieved through a combination of microcode updates for the CPU and software mitigation techniques. For linux systems, software mitigation includes the implementation of retpolines and kernel page-table isolation. However, some of the vulnerabilities of this class cannot be effectively mitigated without updated CPU microcode. 

Heads is built as a payload for Coreboot, which includes the latest available microcode updates for CPUs on supported platforms. However, Intel has stated publicly they will not release further microcode updates for products they consider discontinued and unsupported. This includes processors from the Sandy Bridge and Ivy Bridge line of CPUs. Notably this means all ThinkPad models supported by Heads will most likely not receive further microcode updates to protect against future vulnerabilities of this class. A list of current vulnerabilities and whether microcode updates are available is provided by [Intel](https://software.intel.com/content/www/us/en/develop/topics/software-security-guidance/processors-affected-consolidated-product-cpu-model.html).

Operating systems (e.g. Qubes OS) can mitigate some of these vulnerabilities by other means. An example is the SRBDS vulnerability, where the Xen hypervisor can apply a [workaround](https://seclists.org/oss-sec/2020/q2/182) to mitigate the vulnerability for Ivy Bridge CPUs, which will not receive any microcode update. It is uncertain whether such mitigations are possible in the future without microcode updates supplied from the CPU manufacturer.

The impact on performance as a result of software and microcode mitigations vary between systems and the type of workloads involved. Compared to a system not applying any mitigation measures, a conservative estimate is that the percentage of lost performance is in the low double digit range. Furthermore, as part of mitigating the L1TF vulnerability, Qubes OS [disables](https://www.qubes-os.org/news/2018/09/02/qsb-43/) HyperThreading by default. This has a notable impact on performance in itself, effectively halving the number of cores visible to the system, but is implemented to minimise vulnerability.

The ThinkPad boards supported by Heads are mostly free of binary blobs, with the exception of the Intel Management Engine (which can be "neutered" or minimised) and the ethernet blob (which can be generated). CPU microcode can also technically be considered a binary blob.

Newer hardware platforms like the Librem line of computers, while great care has been taken by to minimise blobs, still contain the closed-source Intel FSP (Firmware Support Package, needed to initialise memory/silicon). However, the CPUs used in these computers are newer, and thus will perceivably receive microcode updates for some time. They generally also allow for more RAM and arguably more modern hardware in general.

When choosing a platform for Heads/Qubes OS, the user must make an informed decision on whether the presence of binary blobs and potential security benefits in future microcode updates outweigh the desirability of maximum control over the firmware on the hardware platform. In addition, the user must take into account the limitations on hardware performance by the various supported boards, and whether the potential increase in performance afforded by a more recent CPU and additional RAM is desirable to compensate for performance reductions as a result of mitigating these vulnerabilities.

tlaurion, 2021-06-15:
@natterangell Awesome! Now, where to put that :)

natterangell, 2021-06-15:
Maybe there is still some clarity missing concerning the status of AMD and the server/workstation boards supported by Heads(?) I am not familiar with AMD policy on microcode updates. However, from what I gather AMD is way less exposed to these vulnerabilities (as can be seen on the wikipedia page referenced), and as far as I can tell KGPE-D16 compatible CPUs do not have the PSP (ME "equivalent") either, so maybe there is no point spilling ink on that. Not sure if AGESA is needed.

Maybe this sort of information belongs in the wiki at the bottom of the "threat model" (after "cover, concealment and compartmentalization").

tlaurion, 2021-06-16:
@natterangell : kgpe-d16 is last coreboot port (apart from the g505s laptop) that doesn;t need AGESA.
PRs welcome!

natterangell, 2021-06-17:
PR submitted.

tlaurion, 2021-06-23:
Merged! Thanks!

closed 2021-06-23 by tlaurion