linuxboot/heads #997: Build x230-hotp-maximized against coreboot 4.13

eganonoa, 2021-06-16:
Building on Debian 10, I am trying to build the x230-hotp-maximized board against coreboot 4.13. Everything proceeds well until I hit the following error.

```
make[1]: *** No rule to make target '../../blobs/xx30/me.bin', needed by 'add_intel_firmware'.
 Stop.
```
I've seen people saying that they have successfully built against coreboot 4.13, and I can't see what I should be doing.

tlaurion, 2021-06-16:
@eganonoa you have to run the script that downloads neuter and trim ME which is under https://github.com/osresearch/heads/blob/master/blobs/xx30/download_clean_me.sh. [README](https://github.com/osresearch/heads/blob/master/blobs/xx30/README).
Would giving a pointer to that prereuisite in [board configs](https://github.com/osresearch/heads/blob/master/boards/x230-maximized/x230-maximized.config#L4) be enough?

eganonoa, 2021-06-16:
> @eganonoa you have to run the script that downloads neuter and trim ME which is under https://github.com/osresearch/heads/blob/master/blobs/xx30/download_clean_me.sh.
> 
> [README](https://github.com/osresearch/heads/blob/master/blobs/xx30/README) explains the process.
> 
> Would giving a pointer to that prereuisite in [board configs](https://github.com/osresearch/heads/blob/master/boards/x230-maximized/x230-maximized.config#L4) be enough?

Thanks. Sorted. Not sure how I missed that. Is it specific to building off of coreboot 4.13? I'm managing a ton of these across my org and can't remember ever running this script. But maybe my memory is failing me.

Yes, I think it would be useful in the board configs. That is the first place I generally look.

I should say, while now is absolutely not the time for me, if its helpful I'd be keen to help out with documentation (x230, t430, x220 at least) and could probably start doing so sometime in August.

closed 2021-06-16 by eganonoa