linuxboot/heads #999: TOTP error after reboot

akfhasodh, 2021-06-23:
Flashed heads-t420-maximized-v0.2.0-1039-g9339394.rom, gpg key works. The totp code works until reboot. After rebooting, it shows "ERROR: Heads couldnt generate the TOTP code."
Everything else seems to be fine, time is synced properly.

tlaurion, 2021-06-23:
@akfhasodh have you used the GUI to seal secret under TPM (Qr code generated)/HOTP inside the smartcard (requiring GPG Admin PIN)?

Instructions are not that clear as of right now, but if you did so from the recovery shell, the measurements will be invalid, since going into recovery shell changes the value (PCRs final outcome) which is sealed in both places. The reason behind this is that going into recovery shell [changes the value in PCR4](https://osresearch.net/Keys/#tpm-pcrs).

You can see the result of  the documentation [PR](https://github.com/osresearch/heads-wiki/pull/71) (Pull request) here:
https://jtmoree-github-com.github.io/heads-wiki/

(Note that this is under review).

If anything unclear, please participate under https://github.com/osresearch/heads-wiki/pull/71

natterangell, 2021-06-23:
@akfhasodh, do you mean that it worked the very first time after you initially set it up, but not afterwards?

If so, I suspect it will work if you do an OEM factory reset within the heads menu and then set it up again. I think I ran into the same error at some point.

jtmoree-github-com, 2021-07-07:
Make sure you have grub.cfg.  Heads doesn't have an error message when it is missing but you will get other errors which are seemingly unrelated.

akfhasodh, 2021-07-07:
where would that be located?

akfhasodh, 2021-07-07:
> Make sure you have grub.cfg. Heads doesn't have an error message when it is missing but you will get other errors which are seemingly unrelated.

where would that be located?

akfhasodh, 2021-07-07:
> @akfhasodh have you used the GUI to seal secret under TPM (Qr code generated)/HOTP inside the smartcard (requiring GPG Admin PIN)?
> 
> Instructions are not that clear as of right now, but if you did so from the recovery shell, the measurements will be invalid (which would explain the error you are getting), since going into recovery shell changes the value (PCRs final outcome) which is sealed in both places. The reason behind this is that going into recovery shell [changes the value in PCR4](https://osresearch.net/Keys/#tpm-pcrs).
> 
> You can see the result of the documentation [PR](https://github.com/osresearch/heads-wiki/pull/71) (Pull request) here:
> https://jtmoree-github-com.github.io/heads-wiki/
> 
> (Note that [osresearch/heads-wiki#71](https://github.com/osresearch/heads-wiki/pull/71) is under review).
> 
> If anything unclear, please participate under [osresearch/heads-wiki#71](https://github.com/osresearch/heads-wiki/pull/71)

Yes, I used the GUI

akfhasodh, 2021-07-07:
FYI, I am currently running without a hdd or ssd with an os on it, dont know if thats required

tlaurion, 2021-07-08:
TPM rollback counter is stored on /boot, where /boot is still required to be mounted everywhere under Heads as of now last time I checked. Let's see.....

Taking for granted you use[ gui-init ](https://github.com/osresearch/heads/blob/055165d61a3c312e7e03998ea832447452a86d71/initrd/bin/gui-init) as boot script per default t420-hotp-maximized board config:
1- init is ran, all depends of /etc/functions in all scripts per imports. gpg distrib public keys and user keys are extracted from rom.
2- gui-init is selected per definition under board config
3- gui-init starts [here](https://github.com/osresearch/heads/blob/055165d61a3c312e7e03998ea832447452a86d71/initrd/bin/gui-init#L170) after first_pass being set to true 
4- Since you have public key injected in rom, you bypass [key generation](https://github.com/osresearch/heads/blob/055165d61a3c312e7e03998ea832447452a86d71/initrd/bin/gui-init#L176)
5- Since totp_confirm is empty (no user input fro menu yet), totp_confirm is undefined (-z matches here). [Verification of HOTP](https://github.com/osresearch/heads/blob/055165d61a3c312e7e03998ea832447452a86d71/initrd/bin/gui-init#L216)
6- HOTP is success, we are still on first_pass but there is no AUTOBOOT_TIMEOUT defined under board config. [gui-init skips this condition](https://github.com/osresearch/heads/blob/055165d61a3c312e7e03998ea832447452a86d71/initrd/bin/gui-init#L176)

@akfhasodh So we still are in first_pass=true since it is unset here.
Not sure what happens for you at this point. What is the first screen you are shown with?

jtmoree-github-com, 2021-07-09:
> > Make sure you have grub.cfg. Heads doesn't have an error message when it is missing but you will get other errors which are seemingly unrelated.
> 
> where would that be located?

standard location in /boot/grub/grub.cfg

tlaurion, 2021-07-09:
> FYI, I am currently running without an hdd or ssd with an os on it, dont know if thats required

@jtmoree-github-com : @akfhasodh doesn't have a /boot following that prior statement.

Rebooting test laptop with latest x230-hotp-maximized flashed to see behavior without SSD plugged in, which should replicate @akfhasodh use case. I also guess the goal here is to boot from USB here.

tlaurion, 2021-07-09:
@akfhasodh 

Without SSD/HDD, I am brought to a red screen saying
```
No bootable OS was found on the default /boot device /dev/sda1
How would you like to proceed?
Select a new boot device
Boot from USB
Continue to main menu
Exit to recovery shell
```

If I choose to continue to main menu, next screen shows a good TOTP verified over phone and HOTP: Error checking code.
That is, because HOTP saves its counter on /boot.

If I try to Generate a new HOTP from Options menu, it fails saying mount: mounting /dev/sda1 on /boot failed: No such file or directory.

If I try to Reset TPM: same thing while error is not shown on screen.

Conclusion: As of right now, a SSD/HDD /boot partition is required to store counters and all scripts are looking for /boot to be mounted. You can still boot from USB, but firmware remote attestation is not possible without a mountable /boot device, Heads being aimed to protect the boot chain in the use case of persistent data storage.

Which [confirms](https://github.com/osresearch/heads/issues/999#issuecomment-876455334):
> TPM rollback counter is stored on /boot, even if unrequired per board config, where /boot is still required to be mounted everywhere under Heads as of now, last time I checked. 

- Interesting enough TPMTOTP unsealing doesn't require /boot. _Sealing does._ (tested on first boot because I simply removed SSD drive and TOTP was displayed aven if Heads was complaining about /boot not being mounted on console.)
- HOTP sealing/unsealing requires /boot as of now.

tlaurion, 2021-07-09:
@osresearch @MrChromebox 

- TPM reset could check for mounted /boot and if not found, [check if rollback protection is forced](https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/bin/gui-init#L358-L373)
- [seal-hotpkey](https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/bin/seal-hotpkey) could do the same. 

Maybe cleaner way is to have only one mount_boot function(its duplicated everywhere) and do the validation/mount there for /boot if rollback is required per config instead of duplicating code logic. /boot not being found should show boot options (set new /boot drive etc) while setting new TOTP/HOTP forcing rollback is a bug, not a feature.

akfhasodh, 2021-07-15:
After setting up a qubes drive and signing the files in boot, I have been able to successfully store both totp and hotp codes in heads. Thank you for the help!

closed 2021-07-15 by akfhasodh