linuxboot/heads #1003: Build failure of crossgcc build-i386-elf-BINUTILS on Fedora 33

RGB-glitch, 2021-07-09:
`Building BINUTILS v2.30 for target ... failed. Check 'build-i386-elf-BINUTILS/build.log'.
make[4]: *** [Makefile:26: build_gcc] Error 1
make[3]: *** [Makefile:48: build-i386] Error 2
make[2]: *** [util/crossgcc/Makefile.inc:46: crossgcc-i386] Error 2
make[2]: Leaving directory '/home/user/CoreBoot/heads/build/coreboot-4.8.1'
make[1]: *** [modules/coreboot:75: "/home/user/CoreBoot/heads/build/coreboot-4.8.1/.xcompile"] Error 2
make[1]: Leaving directory '/home/user/CoreBoot/heads'
make: *** [Makefile:648: all] Error 2
`
and build.log : 
`/build/coreboot-4.8.1/util/crossgcc/build-GMP/config.log:conftest.s:2: Error: no such instruction: `somelabel'
./build/coreboot-4.8.1/util/crossgcc/build-GMP/config.log:conftest.s:2: Error: no such instruction: `somelabel'
./build/coreboot-4.8.1/util/crossgcc/build-i386-elf-BINUTILS/build.log:make[4]: *** [Makefile:918: descriptors.o] Error 1
./build/coreboot-4.8.1/util/crossgcc/build-i386-elf-BINUTILS/build.log:make[3]: *** [Makefile:941: all-recursive] Error 1
./build/coreboot-4.8.1/util/crossgcc/build-i386-elf-BINUTILS/build.log:make[2]: *** [Makefile:692: all] Error 2
./build/coreboot-4.8.1/util/crossgcc/build-i386-elf-BINUTILS/build.log:make[1]: *** [Makefile:6044: all-gold] Error 2
./build/coreboot-4.8.1/util/crossgcc/build-i386-elf-BINUTILS/build.log:make: *** [Makefile:849: all] Error 2
./build/coreboot-4.8.1/util/crossgcc/build-i386-elf-BINUTILS/build.log:make[3]: *** [Makefile:918: descriptors.o] Error 1
./build/coreboot-4.8.1/util/crossgcc/build-i386-elf-BINUTILS/build.log:make[2]: *** [Makefile:941: install-recursive] Error 1
./build/coreboot-4.8.1/util/crossgcc/build-i386-elf-BINUTILS/build.log:make[1]: *** [Makefile:6111: install-gold] Error 2
./build/coreboot-4.8.1/util/crossgcc/build-i386-elf-BINUTILS/build.log:make: *** [Makefile:2225: install] Error 2
`

Applied the temporary fix from #983 in order to build IASL, not sure if this could be related in any way. Thanks

tlaurion, 2021-07-12:
@RGB-glitch The best and easy way to build Heads is over debian-10 for the moment. This is what [CircleCI builds is using](https://github.com/osresearch/heads/blob/214703f92dff45032cb5c82ada777eb42a297f32/.circleci/config.yml#L37) with relevant [dependencies](https://github.com/osresearch/heads/blob/214703f92dff45032cb5c82ada777eb42a297f32/.circleci/config.yml#L43) being deployed prior of applying boards building recipes.

Build tools evolve, with major jumps between new releases of OSes. Building old version of coreboot is becoming a burden on newer versions of OS builld tools, the next step is probably to have a ready to use docker image (non-reproducible) to build Heads reproducibly, unless pinned a reproducible buildstack (guix-toolstack versions of the buildstack are used) or pinned coreboot-sdk docker image is used to build specific coreboot.

Are you saying that applying https://github.com/osresearch/heads/issues/983#issuecomment-799041405 fixed your problem altogether?

RGB-glitch, 2021-07-13:
Hi @tlaurion, I will follow your recommendation, wasn't aware of that but it does make sense. Thanks a lot for your answer. No the problem is still there, however the process was able to go a little further by using #983 recommendation (made building IASL possible)

closed 2021-09-07 by tlaurion