linuxboot/heads #1006: Support diskless platforms while having woking TOTP/HOTP

tlaurion, 2021-07-15:
@osresearch @MrChromebox 

- TPM reset could check for mounted /boot and if not found, [check if rollback protection is forced](https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/bin/gui-init#L358-L373)
- [seal-hotpkey](https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/bin/seal-hotpkey) could do the same. 

Maybe cleaner way is to have only one mount_boot function(its duplicated everywhere) and do the validation/mount there for /boot if rollback is required per config instead of duplicating code logic. /boot not being found should show boot options (set new /boot drive etc) while setting new TOTP/HOTP forcing rollback is a bug, not a feature.

The path for resolution:
- Wrap mount_boot at sealing/unsealing with checks for rollback protection being enforced in board config (no board config enforces this by default. Maybe they all should since its the current behavior?)

_Originally posted by @tlaurion in https://github.com/osresearch/heads/issues/999#issuecomment-877314233_

tlaurion, 2021-07-15:
Context: https://github.com/osresearch/heads/issues/999#issuecomment-877240716

MrChromebox, 2021-07-21:
@tlaurion reading thru the comments here and in 999, I'm not sure what the benefit would be to enforcing rollback protection when /boot isn't found / can't be mounted, as you suggest above. If RP exists to protect against a disk swap with an older kernel, then enforcing RP only when /boot can be mounted would seem sufficient. Not to mention, but the swapped disk would almost certainly fail the hash/signature check.

so I'd propose:
* Wrap mount_boot at sealing/unsealing with checks for rollback protection being enforced only if /boot exists and has been mounted

Going further though, we need to consider what Heads can/can't protect when used in a diskless state. And we want to handle booting an ISO from USB vs an installed OS on USB-attached media.