linuxboot/heads #1010: Add fepitre-bot public key inside distribution trusted keys to test QubesOS 4.1

tlaurion, 2021-07-23:
Key: https://keys.openpgp.org/search?q=1C8714D640F30457EC953050656946BA873DDEC1
To test those builds: https://forum.qubes-os.org/t/qubesos-4-1-alpha-signed-weekly-builds/3601/4

Heads users ask it (and would be useful to troubleshoot #789 on sandy/ivy bridges based laptops).

tlaurion, 2021-07-23:
@fepitre :

```
user@x230-master:~/heads$ gpg --import initrd/etc/distro/keys/qubes-testing.key 
gpg: key 656946BA873DDEC1: new key but contains no user ID - skipped
gpg: Total number processed: 1
gpg:           w/o user IDs: 1
user@x230-master:~/heads$ cat initrd/etc/distro/keys/qubes-testing.key
-----BEGIN PGP PUBLIC KEY BLOCK-----

xsFNBFydTZsBEACv4+MTBz0dK6lIGAYGsApL8iIad7MTrHnWtmrB4UP3mNAPDp8W
zY+HHZQatWTieHL3eT76/UvaVR0bRoeWnxwMxVofxUDRC/nGFWZx7W6un/yb2D30
zGMoRFQegveAR9TelG1KGtWjV8hq7FZ+JtSggkGCOR9+dTUMCgKkeqIHzRHzLVab
MxTDEJplfDDDU+OmxM211j4Oj67WBP19RmrkCYB3FVfgcQ18f8r4NU+BHN3V8egr
5hH3JZboErezPe9Nxoql3A9cHN/zkkJPIHX/pGir5lPkiXrpeMme+d2kjJEtFbXe
ipCi352ILP5qOqqZjcoTCylq7EolgrDXNwWpNPcODI2+zOGW0WKv7wq7yy3rLvDA
HSkG7GPPn1nnJRTpZH8W8jZhWIBV5GpCDSbk0rHsK7qTbUlFkrklonY4jnsTUc2q
jQY38uaJRCLDgvK5TjwYT4IpKxIywSZlxX+8slHzSdbzZ2CPey0tRS0OWcib3/8E
a3k3Ynj99aXpAexoOceJS/CTW33W8cP5GKjDn97aRkKV2BAezXB1xlzfMlG9O2rH
5eZsa/I6ejIjmmsrclYS7xnvBCc8NWTu+gQG+545z7nkwZwFsSrt5efgx7MLtneo
yV/lv7hoZresI9xdezRhq3WF609H2vSr8qqkPSDUGvhuM1Sk4PVzAgxWDwARAQAB
zsFNBFydTZsBEACq+EY5bbvbxp7+heRhKLrEPu9coGXuWOQEmZflzoSFKGKIlMWn
s7JqF0uLVXGd4HSk+c6RArEqQb/rnyAizycxvyhItMR1Y0tXELKXSX8bYaQYIt0Q
APaZc/jcOrQg0XDtyVeWk3pC93MadGruhKI+URwAS9YeJxz3iaUBycHosURRNv53
uCegDtjXGWQq4gT00eQxMYigH62hxL5lWgosV6iDUn8RpZFGu3n2tsRFknd7VbU3
jz8Kid2RnO9pjt9DpbyP0z96jQevd+nTDhVteoN3v9X5w9GMZk3+OT8lYwQIDZtA
+eiUNoIOByRsH0RuXoPsQEYretUa3rzrZCiDQ9bW609iSoqHzOImJSGe2vty1UZA
zbHvwzaS6jW2fNPjzA+IVUdcqkMq/VwjZRDnEOACDo6QBkCGD4gz9uMEdvu/iODh
VOI7D7oi4Bmx12+BbJyLKtFgfHQRbCaFbk7IYa0BfGNjFZyjGLltfj9QEK85LQz5
xybltGEfy8x/eFJ46M8KKEUBGr1Bw+lbT6oAdzQ0krbg50P7+KWNyaXNsK6uLwXd
sNEtvnBN+alcYBYgOROeVYCv4wgVDbNahXjiMdITZAAR7pqWKrFkN8sTbzgMx64L
f7FzK6MOMh10bQYl5XYj6Pn8cJXGkpvSWyaYYfTlX5rUUK+FzMBynoPRAQARAQAB
wsFfBBgBAgAJBQJcnU2bAhsMAAoJEGVpRrqHPd7BoiAP/iZCKzlI+PA+UQtb6RrY
wDGvj9WavaA/diBECYrHefsc4mHVh+I9S9MTmCpBpiVm/fUB02A3cTMSo5xA6gUE
DSLGwn+GFbRkhW8tF7JYQrqesxDNxl75miEILR8lhnfQ8dViPRM/xhB73J9VI0tG
WhqKVQlViR84eNCndmkCvqsduFbeHo3hkthydmcEoTn9UUWsXeAxXMJoa8sPxK+p
raGZqLOqS6nzw33LeAbyIBtnR86/6OHJzGAojl0xsOBiYm27kDOnxkUulj9noqd+
e47UGikC+bMjkYTgPPAEZijJXdzKsq98x+I7AI2VwU4vg44VVpUVs/878gZ8di07
nLK28Z0i7Oq3wR9GIAwOEkjJPbHmx7O2ThloMbIu2x5PSd5YEGwUn3O58RPKNnb1
+9inAE2IywLmReE484zfFDlgFdpqSuooIFDC41Y+0umV+ZhSpAkXy0PFAWGomhKf
OJoaM4in0m3XU3QorQWZR673gLqYiSJj6UnNApEa1rxRoFb3IzkpiL2H2CHyDL14
4qxAZknMXZ0YBl6wHoaJwRNeoR2uQn6g5I2YuRiHDAKN68f3ueLp5gudzXymM23x
28dzNG6b5z9hRxdudNZSSlcfRz9ikorY/b/7ep90kpKSMXA9BA5yF4uttdkSOtFb
m2p03apO69tWotzbpqJyvOgB
=8gwt
-----END PGP PUBLIC KEY BLOCK-----
user@x230-master:~/heads$ gpg --version
gpg (GnuPG) 2.2.12
libgcrypt 1.8.4
```

So basically, Heads code for importing public distro key does not behave any differently then above, hence [that public key is processed but not imported since lacking a UID](https://github.com/osresearch/heads/blob/005a19eeda7267d8a731fffe3a91ee0696a0df9a/initrd/bin/key-init#L10-L14). 
Consequently, verifying detached signature of that ISO fails since that public key is not imported (and unknown).

HW42, 2021-07-23:
AFAIK GnuPG doesn't support keys without User ID. The problem is that you downloaded it from keys.openpgp.org, which distributes User IDs only after explicit confirmation by the owner of the email address in the User ID.

Just download the key from https://qubes.notset.fr/repo/notset/RPM-GPG-KEY-notset this should include the User ID for that public key.

closed 2021-07-29 by tlaurion