linuxboot/heads #1017: Coreboot build ignores CONFIG_ROM_SIZE

Siproqu, 2021-08-17:
I currently write the coreboot configuration for the t430s-maximized:
```
CONFIG_MEASURED_BOOT=y
CONFIG_VENDOR_LENOVO=y
CONFIG_ROM_SIZE=0x01000000
CONFIG_CBFS_SIZE=0xBE0000
CONFIG_HAVE_IFD_BIN=y
CONFIG_HAVE_ME_BIN=y
CONFIG_HAVE_GBE_BIN=y
CONFIG_IFD_BIN_PATH="../../blobs/t430s/ifd.bin"
CONFIG_ME_BIN_PATH="../../blobs/t430s/me.bin"
CONFIG_GBE_BIN_PATH="../../blobs/t430s/gbe.bin"
CONFIG_BOARD_LENOVO_THINKPAD_T430S=y
CONFIG_NO_POST=y
CONFIG_UART_PCI_ADDR=0
# CONFIG_CONSOLE_SERIAL is not set
CONFIG_CONSOLE_CBMEM_BUFFER_SIZE=0x80000
CONFIG_DEFAULT_CONSOLE_LOGLEVEL_5=y
CONFIG_PAYLOAD_LINUX=y
CONFIG_PAYLOAD_FILE="../../build/t430s-maximized/bzImage"
CONFIG_LINUX_COMMAND_LINE="intel_iommu=igfx_off quiet"
CONFIG_LINUX_INITRD="../../build/t430s-maximized/initrd.cpio.xz"
CONFIG_USE_OPTION_TABLE=y
CONFIG_STATIC_OPTION_TABLE=y
```

Unfortunately the build fails with:

```
...
#
# configuration written to /home/user/repo/siproqu/heads/build/coreboot-4.13/t430s-maximized/.config
#
    CC         bootblock/arch/x86/boot.o
    CC         bootblock/arch/x86/bootblock_crt0.o
    CC         bootblock/arch/x86/cf9_reset.o
    CC         bootblock/arch/x86/cpu_common.o
    CC         bootblock/arch/x86/id.o
    CC         bootblock/arch/x86/memcpy.o
    FMAP       t430s-maximized/util/cbfstool/fmaptool -h t430s-maximized/fmap_config.h t430s-maximized/fmap.fmd t430s-maximized/fmap.fmap
syntax error
FATAL: Failed while processing provided descriptor
make[1]: *** [Makefile.inc:1071: t430s-maximized/fmap.fmap] Error 4
make[1]: Leaving directory '/home/user/repo/siproqu/heads/build/coreboot-4.13'
make: *** [Makefile:455: /home/user/repo/siproqu/heads/build/coreboot-4.13/t430s-maximized/.build] Error 1
```

My guess is the flash map descriptor could not be generated correctly because in `build/coreboot-4.13/t430s-maximized/.config` `CONFIG_ROM_SIZE` is incorrect.

It is `CONFIG_ROM_SIZE=0x00400000` instead of `CONFIG_ROM_SIZE=0x01000000` (as in the `coreboot-t430s-maximized.config`)

Can anyone give me a hint why `CONFIG_ROM_SIZE` is overwritten?

tlaurion, 2021-08-17:
@Siproqu MEASURED_BOOT is replaced with TPM_MEASURED_BOOT while some boards names have changed in coreboot when passing to 4.13

Can you copy coreboot_t430_maximized.config and start from there? Also run the blobs script to make sure the ifd me and gbe are valid and report back?

MrChromebox, 2021-08-18:
@Siproqu why are you changing CONFIG_ROM_SIZE? The flash chip size is already set correctly to 16MiB in the t430s mainboard Kconfig

Siproqu, 2021-08-18:
> Can you copy coreboot_t430_maximized.config and start from there?

The config file was a copy of the T430 one.

> Also run the blobs script to make sure the ifd me and gbe are valid and report back?

I extracted the blobs with the extract script from my dumped flash. Also downloaded the cleaned me with the download script.
Was I supposed to edit the ifd.bin?

But the configuration file of the coreboot build does not have the correct variables set.
`CONFIG_IFD_BIN_PATH=` and most other variables from `coreboot-t430s-maximized.config` are nowhere in the `.config` file of coreboot.

If I make a config file with coreboots `make menuconfig` and copy it to `coreboot-t430s-maximized.config`, it sets the variables correctly, but fails with the same error.

If I didn't do a mistake there seem to be two problems:

1. The configuration of Heads is not applied if some variables are missing.
2. The Flash map descriptor is incorrect.

I will write an fmd myself and report back.

tlaurion, 2021-08-19:
Testing build of the following, which is clone of t430 from #1015 (have not took your changes @Siproqu  from #723 )

```
diff --git a/boards/t430s-maximized/t430s-maximized.config b/boards/t430s-maximized/t430s-maximized.config
new file mode 100644
index 0000000..116691c
--- /dev/null
+++ b/boards/t430s-maximized/t430s-maximized.config
@@ -0,0 +1,89 @@
+# Configuration for a T430s running Qubes 4.1 and other Linux Based OSes (through kexec)
+#
+# Includes 
+# - Deactivated+neutered ME and expanded consequent IFD BIOS regions 
+# - Forged 00:DE:AD:C0:FF:EE MAC address  (if not extracting gbe.bin from backup with blobs/xx30/extract.sh)
+#   - Note that this MAC address can be modified under build/coreboot-VER/util/bincfg/gbe-82579LM.set
+#
+# - DOES NOT INCLUDE Nitrokey/Librem Key HOTP Security dongle remote attestation (in addition to TOTP remote attestation through Qr Code)
+export CONFIG_COREBOOT=y
+export CONFIG_COREBOOT_VERSION=4.13
+export CONFIG_LINUX_VERSION=4.14.62
+
+CONFIG_COREBOOT_CONFIG=config/coreboot-t430s-maximized.config
+CONFIG_LINUX_CONFIG=config/linux-x230.config
+
+#Additional hardware support
+CONFIG_LINUX_USB=y
+CONFIG_LINUX_E1000E=y
+
+CONFIG_CRYPTSETUP2=y
+CONFIG_FLASHROM=y
+CONFIG_FLASHTOOLS=y
+CONFIG_GPG2=y
+CONFIG_KEXEC=y
+CONFIG_UTIL_LINUX=y
+CONFIG_LVM2=y
+CONFIG_MBEDTLS=y
+CONFIG_PCIUTILS=y
+
+#Remote attestation support
+#TPM based requirements
+export CONFIG_TPM=y
+CONFIG_POPT=y
+CONFIG_QRENCODE=y
+CONFIG_TPMTOTP=y
+#HOTP based remote attestation for supported USB Security dongle
+#With/Without TPM support
+#CONFIG_HOTPKEY=y
+
+#Nitrokey Storage admin tool
+CONFIG_NKSTORECLI=n
+
+#GUI Support
+#Console based Whiptail support(Console based, no FB):
+#CONFIG_SLANG=y
+#CONFIG_NEWT=y
+#FBWhiptail based (Graphical):
+CONFIG_CAIRO=y
+CONFIG_FBWHIPTAIL=y
+
+#Additional tools:
+#SSH server (requires ethernet drivers, eg: CONFIG_LINUX_E1000E)
+CONFIG_DROPBEAR=y
+
+export CONFIG_BOOTSCRIPT=/bin/gui-init
+export CONFIG_BOOT_REQ_HASH=n
+export CONFIG_BOOT_REQ_ROLLBACK=n
+export CONFIG_BOOT_KERNEL_ADD="intel_iommu=on intel_iommu=igfx_off"
+export CONFIG_BOOT_KERNEL_REMOVE="quiet"
+export CONFIG_BOOT_DEV="/dev/sda1"
+export CONFIG_BOARD_NAME="Thinkpad T430-maximized"
+export CONFIG_FLASHROM_OPTIONS="--force --noverify-all -p internal"
+
+# xx30-*-maximized boards require of you initially call one of the
+#  following to have gbe.bin ifd.bin and me.bin
+#  - blobs/xx30/download_clean_me.sh
+#     To download Lenovo original ME binary, neuter+deactivate ME, produce
+#      reduced IFD ME region and expanded BIOS IFD region.
+#  - blobs/xx30/extract.sh
+#     To extract from backuped 8M (bottom SPI) ME binary, GBE and IFD blobs.
+#
+# This board has two SPI flash chips, an 8 MB that holds the IFD,
+# the ME image and part of the coreboot image, and a 4 MB one that
+# has the rest of the coreboot and the reset vector.
+#   
+# As a consequence, this replaces the need of having to flash t430-flash 
+#  and expands available CBFS region (11.5Mb available CBFS space)
+#
+# When flashing via an external programmer it is easiest to have
+# two separate files for these pieces.
+all: $(build)/$(BOARD)/heads-$(BOARD)-$(HEADS_GIT_VERSION)-bottom.rom
+$(build)/$(BOARD)/heads-$(BOARD)-$(HEADS_GIT_VERSION)-bottom.rom: $(build)/$(BOARD)/$(CB_OUTPUT_FILE)
+	$(call do,DD 8MB,$@,dd of=$@ if=$< bs=65536 count=128 skip=0 status=none)
+	@sha256sum $@
+
+all: $(build)/$(BOARD)/heads-$(BOARD)-$(HEADS_GIT_VERSION)-top.rom
+$(build)/$(BOARD)/heads-$(BOARD)-$(HEADS_GIT_VERSION)-top.rom: $(build)/$(BOARD)/$(CB_OUTPUT_FILE)
+	$(call do,DD 4MB,$@,dd of=$@ if=$< bs=65536 count=64 skip=128 status=none)
+	@sha256sum $@
diff --git a/config/coreboot-t430s-maximized.config b/config/coreboot-t430s-maximized.config
new file mode 100644
index 0000000..ede34ce
--- /dev/null
+++ b/config/coreboot-t430s-maximized.config
@@ -0,0 +1,22 @@
+CONFIG_USE_OPTION_TABLE=y
+CONFIG_STATIC_OPTION_TABLE=y
+# CONFIG_USE_BLOBS is not set
+CONFIG_VENDOR_LENOVO=y
+CONFIG_NO_POST=y
+CONFIG_CBFS_SIZE=0xB80000
+CONFIG_IFD_BIN_PATH="../../blobs/xx30/ifd.bin"
+CONFIG_ME_BIN_PATH="../../blobs/xx30/me.bin"
+CONFIG_GBE_BIN_PATH="../../blobs/xx30/gbe.bin"
+CONFIG_HAVE_IFD_BIN=y
+CONFIG_BOARD_LENOVO_T430S=y
+CONFIG_LINUX_COMMAND_LINE="intel_iommu=igfx_off quiet"
+CONFIG_UART_PCI_ADDR=0
+CONFIG_HAVE_ME_BIN=y
+CONFIG_HAVE_GBE_BIN=y
+CONFIG_NO_GFX_INIT=y
+CONFIG_DRIVERS_PS2_KEYBOARD=y
+CONFIG_TPM_MEASURED_BOOT=y
+CONFIG_CONSOLE_CBMEM_BUFFER_SIZE=0x80000
+CONFIG_PAYLOAD_LINUX=y
+CONFIG_PAYLOAD_FILE="../../build/t430s-maximized/bzImage"
+CONFIG_LINUX_INITRD="../../build/t430s-maximized/initrd.cpio.xz"
```

tlaurion, 2021-08-19:
@Siproqu note that board name changed to `CONFIG_BOARD_LENOVO_T430S`

tlaurion, 2021-08-19:
@Siproqu : the previous board config + coreboot config builds

```
2021-08-18 20:14:16-04:00 MAKE coreboot
2021-08-18 20:15:51-04:00 DONE coreboot
# Use coreboot.rom, because custom output files might not be processed by cbfstool
"/home/user/heads/build/coreboot-4.13/t430s-maximized/cbfstool" "/home/user/heads/build/coreboot-4.13/t430s-maximized/coreboot.rom" print
FMAP REGION: COREBOOT
Name                           Offset     Type           Size   Comp
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           86796 none
cpu_microcode_blob.bin         0x15400    microcode       26624 none
fallback/ramstage              0x1bc80    stage           99607 none
config                         0x34200    raw               840 none
revision                       0x34580    raw               697 none
fallback/dsdt.aml              0x34880    raw             14615 none
cmos.default                   0x38200    cmos_default      256 none
vbt.bin                        0x38340    raw              1409 LZMA (4459 decompressed)
cmos_layout.bin                0x38900    cmos_layout      1932 none
fallback/postcar               0x39100    stage           27288 none
fallback/payload               0x3fc00    simple elf    7288775 none
(empty)                        0x733400   null          4376984 none
bootblock                      0xb5fdc0   bootblock       65536 none
2021-08-18 20:15:51-04:00 INSTALL   build/coreboot-4.13/t430s-maximized/coreboot.rom => build/t430s-maximized/heads-t430s-maximized-v0.2.0-1058-gb9468f5-dirty.rom
4306937c0f1eadccb464e7fde485c925bbf9825e654f59017fc84e399d8b56d8  build/t430s-maximized/heads-t430s-maximized-v0.2.0-1058-gb9468f5-dirty.rom
2021-08-18 20:15:51-04:00 DD 8MB build/t430s-maximized/heads-t430s-maximized-v0.2.0-1058-gb9468f5-dirty-bottom.rom
29603efea40e7267063ddba6c9db7f595d90b531dc8f04c06c37627578fa8be3  /home/user/heads/build/t430s-maximized/heads-t430s-maximized-v0.2.0-1058-gb9468f5-dirty-bottom.rom
2021-08-18 20:15:51-04:00 DD 4MB build/t430s-maximized/heads-t430s-maximized-v0.2.0-1058-gb9468f5-dirty-top.rom
89e4e2bbcfd54691d1880ba0f1fe9a853f82f4626ea397cd92b8bc587634dd52  /home/user/heads/build/t430s-maximized/heads-t430s-maximized-v0.2.0-1058-gb9468f5-dirty-top.rom
```

But as said in previous comment, nothing of #723 was taken in.

tlaurion, 2021-08-19:
@Siproqu as stated https://github.com/osresearch/heads/pull/723/files#r516916422 the above config is the result of `make savedefconfig` to only have differences from default configs in Heads tree.

Siproqu, 2021-08-19:
@tlaurion Thank you very much for your help! It compiled just fine. Will flash it later this day.
Besides, I am sorry for wasting your time, just because I was inattentive.

> Where it was reviewed but you never answered questions there.
Also sorry for that, it simply fall through the cracks.

tlaurion, 2021-08-19:
@Siproqu Let us know how it goes. We could close #723 at the same time and merge those changes to #1015. If you have some additional commits, please do a PR and I will take your signed commits and add them to #1015 where relevant.

Siproqu, 2021-08-20:
Unfortunately I couldn't flashed the image yesterday. The very tiny legs of the chip were too wear down, the clip couldn't grip it and flashrom could not read the chip.
Since I won't have access to proper soldering equipment in the next weeks to replace it, I will maybe try to flash it via the internal programmer. So I will wait with that until I have soldering equipment available.

tlaurion, 2025-02-13:
Abandoned alongside https://github.com/linuxboot/heads/pull/1019. Closing

closed 2025-02-13 by tlaurion

tlaurion, 2025-11-22:
> Heya, I've recently picked up a T430s and am planning to create and maintain a heads config for it, when ready and everything is smoothed out, I'll submit a PR to upstream.
> 
> I'll also be willing to maintain it for the foreseeable as this will be my daily-driver.

Also see https://github.com/linuxboot/heads-wiki/pull/77
Awesome @CanadianBaconBoi, almost failed to see this comment in closed issue. 

see https://osresearch.net/community/

CanadianBaconBoi, 2025-11-22:
> Also see [linuxboot/heads-wiki#77](https://github.com/linuxboot/heads-wiki/pull/77) Awesome [@CanadianBaconBoi](https://github.com/CanadianBaconBoi), almost failed to see this comment in closed issue.
> 
> see https://osresearch.net/community/

Thanks much for the resource, and I'll be heading over onto the matrix.