linuxboot/heads #1018: POWER9 (Talos II and Talos II Lite) coreboot beta testers needed

pietrushnic, 2021-08-23:
Hi all, we reaching stage where beta testers of coreboot-based Dasharo Trustworthy computing will be crucial to further project progress. Why we want to replace [hostboot](https://github.com/open-power/hostboot) with [coreboot]()? Some reason were provided in excellent blog post by @krystian-hebel, which you can read [here](https://blog.3mdeb.com/2020/2020-12-31-coreboot_for_power/).

If you have Talos II, Talos II Lite or can use QEMU, and would like to help contribute feel free to join our [mailing list](https://newsletter.3mdeb.com/subscription/w2Y2G4Rrj), where we will post releases and communicate how to move forward. There is also OpenPOWER coreboot community call hold bi-weekly. We would appreciate community support.

@tlaurion feel free to mention here anyone interested or add cross references.

Just to cross-reference:
* https://github.com/osresearch/heads/issues/692
* https://github.com/osresearch/heads/issues/720
* https://github.com/QubesOS/qubes-issues/issues/4318

tlaurion, 2021-09-22:
@madscientist159 would be [willing](https://github.com/QubesOS/qubes-issues/issues/4318#issuecomment-925105667) to test and report!

Who else is owner of a Talos II and ready to collaborate?

tlaurion, 2022-02-01:
I have received board and will make space to install board and power supply in the next coming days.

Blocker is TPM support as of now.

@pietrushnic

pietrushnic, 2022-02-01:
@tlaurion yes, completly agree. As soon as we will gather technical feedback we will try to do something with that. Do you have required TPM hardware for further testing, when TPM would be unbloked?

tlaurion, 2022-02-04:
@pietrushnic nope, since I do not know which TPM module it should be, and will probably involve testing TPM2 toolstack which some are playing around with as of now without much input as of now, but where @hardened-vault branched a functional implementation including wrapper so that both TPM 1 and 2 can coexist.

Have only an Asus TPM 1.2 20-1 pin currently connected in KGPE-D16.