linuxboot/heads #1020: x230-hotp-maximized (and t430-hotp-maximized) builds failing

nicprov, 2021-08-24:
For reference, I have successfully built the x230, x230-flash and x230-hotp-verification roms, but the hotp-maximized ones seem to fail. I'm not sure if I did something wrong, or if there's a bug.

Here's the command I ran:
`make BOARD=t430-hotp-maximized CPU=2`

Here are the logs from the build (for x230-hotp-maximized, but the error is the same for t430-hotp-maximized):
`
[...]
perl: warning: Falling back to the standard locale ("C").
addd502bdcb73d17480d8722d8a6234620f97c90a0f116aa703d587820d78c30  build/t430-hotp-maximized/initrd.cpio.xz
2021-08-24 02:08:59+00:00 MAKE coreboot
tail /root/heads/build/log/coreboot.log
-----
    HOSTCC     cbfstool/LzmaEnc.o
    HOSTCC     cbfstool/cbfstool (link)
Created CBFS (capacity = 11927512 bytes)
    CBFS       fallback/romstage
    CBFS       cpu_microcode_blob.bin
    CBFS       fallback/ramstage
    CBFS       config
    CBFS       revision
    CBFS       cmos.default
    CBFS       cmos_layout.bin
    CBFS       fallback/dsdt.aml
    CBFS       fallback/payload
    Compile IFDTOOL
    HOSTCC     util/ifdfake/ifdfake
    DD         Adding Intel Firmware Descriptor
    IFDTOOL    me.bin -> coreboot.pre
Could not open file: No such file or directory
File t430-hotp-maximized/coreboot.pre is 12582912 bytes
make[1]: *** [src/southbridge/intel/common/firmware/Makefile.inc:53: add_intel_firmware] Error 1
make[1]: Leaving directory '/root/heads/build/coreboot-4.8.1'
make: *** [Makefile:455: /root/heads/build/coreboot-4.8.1/t430-hotp-maximized/.build] Error 1
`

tlaurion, 2021-08-24:
@nicprov you need to call desired script from blobs/xx30 prior of building, so the me is downloaded, extracted and cleaned.
Or extract the blobs from a local backup. 

Is there something else that should be added under the board config to clarify this better?

nicprov, 2021-08-24:
Thanks for the quick reply! 

I completely missed the comment at the bottom of the config file (xx30-*-maximized boards require of you initially call one of the...). If you added a note at the top of the config file, or added a note in the general README or something, maybe it could help avoid some confusion!

On a side note, what exactly is the difference between x230-hotp-verification and x230-hotp-maximized?

tlaurion, 2021-08-24:
@nicprov [This comment](https://github.com/osresearch/heads/pull/1015#issuecomment-894523907) should answer your question, where if clear enough will be used as a base for maximized documentation.

nicprov, 2021-08-24:
@tlaurion Thanks, that answers my question! Really appreciate the work you're doing here!!

closed 2021-08-24 by nicprov