linuxboot/heads #1021: gpg: waiting for lock

Niklas-Nitrokey, 2021-08-30:
About every 5 times when starting my Lenovo T430 the message "gpg: waiting for lock (held by x)" comes up and heads does not start. i remain stuck in an endless loop.

```
gpg: waiting for lock (held by 169) ... 
gpg: waiting for lock (held by 169) ...
gpg: waiting for lock (held by 169) ...
gpg: waiting for lock (held by 169) ...
gpg: waiting for lock (held by 169) ...
```

tlaurion, 2021-08-30:
@Niklas-Nitrokey This is what you get directly at boot? Prior of having `HOTP: Success` and a TOTP code?
When does that happen exactly?

Niklas-Nitrokey, 2021-08-30:
I get this output immediately after starting the laptop.
[image]

tlaurion, 2021-08-30:
Call tree:

- init calls [cbfs-init](https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/init#L48) (which outputs [Loading from CBFS](https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/bin/cbfs-init#L16)...)
- init then [calls key-init](https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/init#L57) which [loads keys from /etc/distro](https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/bin/key-init#L5-L17)

Anything funky in that directory in the rom you are using @Niklas-Nitrokey ?

tlaurion, 2021-08-30:
[Wiping the error redirections](https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/bin/key-init#L5-L17) (2> /dev/null) on your platform/fork should give useful input to troubleshoot this issue I unfortunately cannot replicate on my side on x230 with latest commit ID.

Not sure I understand what is happening here, but it seems that gpg waits a child to exit, which never happens.

@Niklas-Nitrokey : You say it happens randomly 1/5 times?
It would be useful to see the output of when it succeeds (with 2>/dev/null removed above) and when it doesn't.

tlaurion, 2021-09-02:
> The problem has already occurred for x230s and t430s.
> commit: [Nitrokey@5e5c36f](https://github.com/Nitrokey/heads/commit/5e5c36f35574aeddc435fa83f249c5ef90969819)
> 
> How can I change the key-init so that the changes are saved for the reboot?

@Niklas-Nitrokey The simplest solution is to edit the sources directly and build a new rom to be flashed internally with only changes under key-init as shown in previous post to give output of what is going on prior of looping for something locked for what we don't know why (I can't replicate).

A more complicated approach would be to inject those files with cbfs from within heads, in a rom previously backed up with flash.sh tool. If that is your intended approach, you can look into codebase for cbfs calls, which can read, delete and add files into rom with extraction path.

Here is an example of how the file [/etc/config.user is injected inside of the rom](https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/bin/config-gui.sh#L84), which is then [extracted back in place by cbfs-init at boot.](https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/bin/cbfs-init#L13-L29)

tlaurion, 2021-12-08:
@Niklas-Nitrokey still relevant on master? Please close otherwise.

tlaurion, 2022-03-28:
Please reopen otherwise

closed 2022-03-28 by tlaurion

Niklas-Nitrokey, 2022-03-29:
Sorry for the late reply we installed the new version last week on the corresponding laptops with this "lock problem" and it seems to have fixed the problem completely.

tlaurion, 2022-06-17:
@Niklap97 I'm having this issue right now on a x230 test laptop.
That laptop has a clock skew issue, and on boot I have "rtc_cmos: 00:03: hctosys: unale to read the hardware clock".

If I reboot and hit r to hit recovery console early, and reboot, the same error above appears but I do not have the "gpg: waiting for lock" issue anymore.

This happened directly after a maximized -> maximized internal upgrade keeping settings, so I my gpg pubring and trustdb have been migrated. On firmware upgrade, there is MRC cache that is rewritten on SPI flash.... Unfortunately, I cannot replicate further.

**But the hint here is that RTC clock is skew, and if the laptop has issues keeping time, RTC battery (button battery under keyboard) should be replaced.**

Also, before booting, it is recommended (if time skew) to call `network-recovery-init` from command line with Ethernet cable connected, or to set time manually through documented date and hwclock -w.

network-init-recovery is recommended because automatic. 
- It will randomize Ethernet MAC locally (does not survive botting into OS)
- It will get IP from DHCP server
- Attempt to sync time through NTP against obtained DNS server from DHCP response (normally internal router)
  - Otherwise will attempt NTP against ntp.pool.org (with IP of server obtained through name resolution from DNS server obtained through DHCP). 
 
 This is less error prone then setting date and proper input format, and will set time in UTC as Heads requires, leaving no room for user error.

Hope that helps.

githubuseravailable, 2022-09-14:
> Also, before booting, it is recommended (if time skew) to call network-recovery-init from command line with Ethernet cable connected

@tlaurion i also have the same issue, 
please inform, how to access command line, to call network-recovery-init, before booting ? 
also is it compulsory to use Ethernet cable ? is it okay with Wifi ?

the complete issue is in ticket [Starting new kernel takes long time](https://github.com/osresearch/heads/issues/1213)

tlaurion, 2022-09-14:
@githubuseravailable 

Hit 'r' keyboard's key repeatedly and frenetically until you reach recovery shell.
Then you can call `network-recovery-init` which will try to obtain DHCP address from router, then try to sync clock through NTP with router and otherwise will try to use router's IP address as DNS server to resolve ntp.pool.org and sync time with IP address returned for that DNS name.

Otherwise you would have to set time manually. Just searched through osresearch.net with the search tool, and was not able to find any reference to data nor skew, so that got lost somehow ?!?
Would be beneficial to open an issue under heads-wiki github repository so that that is documented properly.

> also is it compulsory to use Ethernet cable ? is it okay with Wifi ?
Ethernet cable only. There is nothing to deal with WPA key nor anything that fancy under Heads.

Diboby, 2024-03-16:
# List the files in the public-keys.d directory
ls -al ~/.gnupg/public-keys.d/

# Output might look like this:
# total 84
# drwxr-x--- 2 root root  4096 Mar 16 02:34 .
# -rw-r--r-- 2 root root    27 Mar 16 02:34 .#lk0x00007ff8c7261aa0.buildkitsandbox.7
# drwx------ 4 root root  4096 Mar 16 02:34 ..
# -rw-r--r-- 1 root root 65536 Mar 16 02:34 pubring.db
# -rw-r--r-- 2 root root    27 Mar 16 02:34 pubring.db.lock

# Check the contents of the lock file
cat ~/.gnupg/public-keys.d/pubring.db.lock 
# Output: 7 buildkitsandbox

# Try to list keys (this might fail due to the lock file)
gpg --list-keys
# Output: gpg: Note: database_open 134217901 waiting for lock (held by 7) ...

# Remove the lock file
rm -f ~/.gnupg/public-keys.d/pubring.db.lock 

# Try to list keys again (this should succeed now)
gpg --list-keys
# Output: [keyboxd] pub rsa2048 2024-03-15 [SC] B7244*******5341455F5D5*******1F96D uid [ unknown] GitLab ***** (032024) <c*******d@gitlab.com> sub rsa**** 2024-03-15 [E]

Diboby, 2024-03-16:
Hi there,

This indicates the presence of a `lock` file within `~/.gnupg` or its subfolders, likely located within the `\~/.gnupg/public-keys.d` directory, which shouldn't be there. It needs to be removed. For example: `\~/.gnupg/public-keys.d/pubring.db.lock`. If you run `cat` on the `lock` file, you'll find a `PID` inside the error message `removing stale lockfile (created by 17297)` or in my case `gpg: Note: database\_open 134217901 waiting for lock (held by 7) ...`.

Here are some other approaches to prevent this error:

- Docker:
  When creating your Docker image, include the `public-keys.d` folder like this:  
  ```RUN mkdir -p \~/.gnupg/public-keys.d``` before importing keys.

- On a local machine:
  If the folder `~/.gnupg/public-keys.d` doesn't exist, create it before importing keys.

```
568718040f3d:/# ls -al ~/.gnupg/public-keys.d/
total 84
drwxr-x--- 2 root root  4096 Mar 16 02:34 .
-rw-r--r-- 2 root root    27 Mar 16 02:34 .#lk0x00007ff8c7261aa0.buildkitsandbox.7
drwx------ 4 root root  4096 Mar 16 02:34 ..
-rw-r--r-- 1 root root 65536 Mar 16 02:34 pubring.db
-rw-r--r-- 2 root root    27 Mar 16 02:34 pubring.db.lock
568718040f3d:/# cat ~/.gnupg/public-keys.d/pubring.db.lock 
         7
buildkitsandbox
568718040f3d:/# gpg --list-keys
gpg: Note: database_open 134217901 waiting for lock (held by 7) ...
gpg: Note: database_open 134217901 waiting for lock (held by 7) ...
gpg: Note: database_open 134217901 waiting for lock (held by 7) ...
gpg: Note: database_open 134217901 waiting for lock (held by 7) ...
gpg: Note: database_open 134217901 waiting for lock (held by 7) ...
gpg: keydb_search_first failed: Operation timed out
568718040f3d:/# rm -f ~/.gnupg/public-keys.d/pubring.db.lock 
568718040f3d:/# gpg --list-keys
[keyboxd]
---------
pub   rsa**** 2024-03-15 [SC]
      B7244*******5341455F5D5*******1F96D
uid           [ unknown] GitLab ***** (032024) <c*******d@gitlab.com>
sub   rsa**** 2024-03-15 [E]

568718040f3d:/# 
```

Hope this helps!

tlaurion, 2024-11-22:
This was fixed a while ago no? Was linked to cmos system kept time being either way in the past compared to kept under repo public keys, and was thought to be fixed with early boot detection of date being < 2024 recently, and forcing the end user to set time with and helper, before those keys are loaded from firmware.

Is this still an issue for someone here? If so please open a new issue with proper screenshots. This was resolved with PR pointing here see above.