linuxboot/heads #1022: Librem 13 v3 support

ethark, 2021-08-31:
Hi, I have a Librem 13 v3. What porting effort would it take to get heads running on the Librem 13 v3? Thanks.

kylerankin, 2021-08-31:
On Mon, Aug 30, 2021 at 05:16:05PM -0700, ethark wrote:
> Hi, I have a Librem 13 v3. What porting effort would it take to get heads running on the Librem 13 v3? Thanks.
> 

Heads is already supported on the Librem 13v3. We even provide a custom
version of firmware we call "PureBoot" that includes Heads along with other
customizations we add for our hardware. You can use our Coreboot
utility/update script to switch over to PureBoot by following the
directions on https://puri.sm/projects/coreboot

MrChromebox, 2021-09-02:
@ethark one thing to note is that the Librem 13v2 and v3 use the same firmware image, so don't be alarmed if the Purism script identifies your device as a V2 / downloads/flashes the 13 v2 firmware

ethark, 2021-09-02:
Thanks, @MrChromebox. I was just confused because it wasn't listed on https://osresearch.net/Prerequisites#supported-devices. perhaps the under device it should say "Purism Librem 13 v2/v3" in the "librem_13v2" row.

Also the link on to "https://osresearch.net/Prerequisites/" on https://osresearch.net/Install-and-Configure is broken for me on firefox. I think it's due to the trailing "/" because the link on the sidebar works.

ethark, 2021-09-02:
Are there more boards with multiple products associated with them? i'll make a patch request

MrChromebox, 2021-09-02:
> Thanks, @MrChromebox. I was just confused because it wasn't listed on https://osresearch.net/Prerequisites#supported-devices. perhaps the under device it should say "Purism Librem 13 v2/v3" in the "librem_13v2" row.

that list is very out of date, as it's missing at least 2 other recent Librem Models (Mini v2, 14)

> Are there more boards with multiple products associated with them? i'll make a patch request

no, but the L13v3 isn't a separate product - it's a L13v2 with a TPM chip.

closed 2021-09-22 by ethark