linuxboot/heads #1036: Build error on Xubuntu 21.04

copyvar, 2021-10-06:
I need some help to build heads. I installed all packages mentioned in line 13 https://github.com/osresearch/heads/blob/master/.circleci/config.yml

`make[3]: Entering directory '/home/test/heads/build/make-4.2.1/doc'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/test/heads/build/make-4.2.1/doc'
make[3]: Entering directory '/home/test/heads/build/make-4.2.1'
gcc  -g -O2 -Wl,--export-dynamic  -o make ar.o arscan.o commands.o default.o dir.o expand.o file.o function.o getopt.o getopt1.o guile.o implicit.o job.o load.o loadapi.o main.o misc.o posixos.o output.o read.o remake.o rule.o signame.o strcache.o variable.o version.o vpath.o hash.o remote-stub.o glob/libglob.a   -ldl 
/usr/bin/ld: glob/libglob.a(glob.o): in function `glob_in_dir':
/home/test/heads/build/make-4.2.1/glob/glob.c:1269: undefined reference to `__stat'
/usr/bin/ld: glob/libglob.a(glob.o): in function `glob':
/home/test/heads/build/make-4.2.1/glob/glob.c:1036: undefined reference to `__stat'
/usr/bin/ld: /home/test/heads/build/make-4.2.1/glob/glob.c:814: undefined reference to `__stat'
/usr/bin/ld: /home/test/heads/build/make-4.2.1/glob/glob.c:960: undefined reference to `__stat'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:656: make] Error 1
make[3]: Leaving directory '/home/test/heads/build/make-4.2.1'
make[2]: *** [Makefile:798: all-recursive] Error 1
make[2]: Leaving directory '/home/test/heads/build/make-4.2.1'
make[1]: *** [Makefile:534: all] Error 2
make[1]: Leaving directory '/home/test/heads/build/make-4.2.1'
LANG=C MAKE=/home/test/heads/build/make-4.2.1/make /home/test/heads/build/make-4.2.1/make  all
/bin/sh: 1: /home/test/heads/build/make-4.2.1/make: not found
make: *** [Makefile:668: all] Error 127`

copyvar, 2021-10-10:
I am stuck after hours of research in the web. Anyone got an idea?

paulmenzel, 2021-10-11:
Please try the change from merge/pull request https://github.com/osresearch/heads/pull/1035.

copyvar, 2021-10-11:
> Please try the change from merge/pull request #1035.

Thank you, that helped. Now I got stuck at another point (should I open a new issue for that?):

make[2]: Entering directory '/home/test/heads/build/coreboot-4.8.1'
Welcome to the coreboot cross toolchain builder v1.52 (June 11th, 2018)

Building toolchain using 4 thread(s).

Downloading and verifing tarballs ...
 * acpica-unix2-20180531.tar.gz (downloading from http://acpica.org/sites/acpica/files/acpica-unix2-20180531.tar.gz)... 100%... hash verified ("17717140438d506533b4a56e34350749d7b84d6c")
Downloaded tarballs ... ok
Unpacking and patching ...
 * acpica-unix2-20180531.tar.gz
   o acpica-unix2-20180531_iasl.patch
Unpacked and patched ... ok
Building packages ...
Building IASL v20180531 for host ... failed. Check 'build-IASL/build.log'.
make[3]: *** [Makefile:36: build_iasl] Error 1
make[2]: *** [util/crossgcc/Makefile.inc:52: iasl] Error 2
make[2]: Leaving directory '/home/test/heads/build/coreboot-4.8.1'
make[1]: *** [modules/coreboot:83: "/home/test/heads/build/coreboot-4.8.1/util/crossgcc/xgcc/bin/iasl"] Error 2
make[1]: Leaving directory '/home/test/heads'
make: *** [Makefile:668: all] Error 2
make[2]: Entering directory '/home/test/heads/build/coreboot-4.8.1'
Welcome to the coreboot cross toolchain builder v1.52 (June 11th, 2018)

Building toolchain using 4 thread(s).
```
Downloading and verifing tarballs ...
 * acpica-unix2-20180531.tar.gz (downloading from http://acpica.org/sites/acpica/files/acpica-unix2-20180531.tar.gz)... 100%... hash verified ("17717140438d506533b4a56e34350749d7b84d6c")
Downloaded tarballs ... ok
Unpacking and patching ...
 * acpica-unix2-20180531.tar.gz
   o acpica-unix2-20180531_iasl.patch
Unpacked and patched ... ok
Building packages ...
Building IASL v20180531 for host ... failed. Check 'build-IASL/build.log'.
make[3]: *** [Makefile:36: build_iasl] Error 1
make[2]: *** [util/crossgcc/Makefile.inc:52: iasl] Error 2
make[2]: Leaving directory '/home/test/heads/build/coreboot-4.8.1'
make[1]: *** [modules/coreboot:83: "/home/test/heads/build/coreboot-4.8.1/util/crossgcc/xgcc/bin/iasl"] Error 2
make[1]: Leaving directory '/home/test/heads'
make: *** [Makefile:668: all] Error 2
```

copyvar, 2021-10-15:
@tlaurion @paulmenzel I am sorry if that is a dumb question, but I tried to manually change all the files which are different in https://github.com/tlaurion/heads/commit/60e908ec2794ae44cd0afa0fcdd15f46af3e2440 to the main heads repo. Still I am not able to build (I still get stuck at IASL part). Please tell me how to properly apply the patches via command line (without the need to manually edit files). What is the proper step after `git clone https://github.com/osresearch/heads` ? I hope when applying them correctly I can finally build heads.

tlaurion, 2021-10-16:
@copyvar I will rebase #1038 on master, which will build coreboot 4.13 and 4.11 based boards including xx30 and xx20 boards.

It is still impossible with current parches to build coreboot 4.8.1 based boards, unfortunately, which this PR is attempting to resolve the current raised issue.

copyvar, 2021-10-29:
Thanks a lot for your effort!

closed 2021-10-29 by copyvar