linuxboot/heads #1054: Documentation update: Highlight differences in x230 boards

copyvar, 2021-10-29:
Hello,

I have difficulties to understand the difference of the available x230 boards. In https://github.com/osresearch/heads-wiki/edit/master/Installing-and-Configuring/Prerequisites.md three boards namely `x230`, `x230-hotp-verification` and `x230-flash` are listed. However, looking at https://github.com/osresearch/heads/tree/master/boards five boards are listed: `x230-flash`, `x230-hotp-maximized`, `x230-hotp-verification`, `x230-maximized` and `x230`

What are the differences between them? Looking at the boards config file does not help alot (at least not for someone like me, who is relatively new to this). It would help a lot, if there was better documentation about the differences. I will update the config files of each board and the documentation if someone answers.

tlaurion, 2021-10-31:
Most of those questions are answered under #1015 ans under osresearch.net website, but let me know what is missing.

The quick answer would be: are toy using an external reprogrammer and would you intend to neuter ME.

tlaurion, 2021-10-31:
So to answer your question

x230-flash is to be flashed through skulls or 1vyrain, not touching IFD nor ME.

x230 is without HOTP, coming from unknown external flashing rom.

x230-hotp-verification is with hotp, coming from unknown external flashing from as above.

x230-hotp-maximized is internal flashing on top of unlocked IFD where ME can be overwritten. Can he flashed both internally or externally, while first internal flash coming from recovery shell needs to be a manual flashrom call, not specifying IFD bios region as o'er birnal internal upgrades from x230 and x230-flash

x230-maximized is like x230-hotp-maximized but without HOTP support.

tlaurion, 2021-10-31:
Situation is different for xx20 board which have 8MiB of SPI flash coming from a single SPI chip.

Consequently, xx20 boards came after, and do not loose features since board owners already cleaned their ME wlsince they had no other choice to be able to flash Heads initially.

So they have 7.5MiB of SPI available since the arrival of the boards, as opposed to xx30 boards which had 7MiB initially.

So xx20 boards owners are not loosing FBWhiptail as opposed to xx30 owners of not going the maximized path.

@copyvar Poke me directly if you have questions, I would gladly welcome help into documenting this properly.

tlaurion, 2021-11-01:
@copyvar please review/cherry-pick my branch https://github.com/osresearch/heads-wiki/pull/82/files?short_path=529f311#diff-529f311e8716c83389f3e386ff6b8e9a90113d07f137321554820739e2a7b81c in link with #1015

tlaurion, 2021-11-07:
@copyvar ?