linuxboot/heads #1056: No matter what build I get the same result - just the main heads rom

lukyjr, 2021-11-14:
Hi all!

I am new to heads. Trying to flash a T430, and I can't seem to compile the 2 roms I need.
I have tried other boards too, and I always get the same result: build/qemu-coreboot/heads--v0.2.0-1092-gfdbd9b2.rom
It never creates a new folder for a specific board build as the documentation says it should.

I am running ubuntu 20.04.3 on a live USB with persistence (right now installing linux on a machine for flashing isn't an option, but I will find a way to make it happen if I must)
I see there are warnings at the top and a couple "command not found" lines
What am I doing wrong?

Here is my log:

ubuntu@ubuntu:~/heads$ make BUILD=t430-hotp-maximized
2021-11-14 07:38:07-05:00 Wrong gawk detected: 
/home/ubuntu/heads/boards/qemu-coreboot/qemu-coreboot.config:58: warning: overriding recipe for target 'run'
Makefile:69: warning: ignoring old recipe for target 'run'
Makefile:212: warning: overriding recipe for target 'all'
Makefile:69: warning: ignoring old recipe for target 'all'
make -C "/home/ubuntu/heads/build/gawk-4.2.1/"  \
	2>&1 \
	| tee "/home/ubuntu/heads/build/log/gawk.log" \
	> /dev/null
make -C "/home/ubuntu/heads/build/coreboot-4.8.1" CPUS=12 iasl
make[1]: Entering directory '/home/ubuntu/heads/build/coreboot-4.8.1'
Welcome to the coreboot cross toolchain builder v1.52 (June 11th, 2018)

Building toolchain using 12 thread(s).

Downloading and verifing tarballs ...
 * acpica-unix2-20180531.tar.gz (cached)... hash verified ("17717140438d506533b4a56e34350749d7b84d6c")
Downloaded tarballs ... ok
Unpacking and patching ...
 * acpica-unix2-20180531.tar.gz
   o acpica-unix2-20180531_iasl.patch
Unpacked and patched ... ok
Building packages ...
Skipping IASL v20180531 for host as it is already built
Packages built ... ok
Copied EDK2 tools template ... ok
Cleaning up temporary files... ok

You can now run IASL ACPI compiler from /home/ubuntu/heads/build/coreboot-4.8.1/util/crossgcc/xgcc.
make[1]: Leaving directory '/home/ubuntu/heads/build/coreboot-4.8.1'
make -C "/home/ubuntu/heads/build/coreboot-4.8.1" CPUS=12 crossgcc-i386
make[1]: Entering directory '/home/ubuntu/heads/build/coreboot-4.8.1'
Welcome to the coreboot cross toolchain builder v1.52 (June 11th, 2018)

Building toolchain using 12 thread(s).

Target architecture is i386-elf

Found compatible Ada compiler, enabling Ada support by default.

./buildgcc: line 1198: (: command not found
./buildgcc: line 1199: (: command not found
Downloading and verifing tarballs ...
 * gmp-6.1.2.tar.xz (cached)... hash verified ("9dc6981197a7d92f339192eea974f5eca48fcffe")
 * mpfr-3.1.5.tar.xz (cached)... hash verified ("c0fab77c6da4cb710c81cc04092fb9bea11a9403")
 * mpc-1.0.3.tar.gz (cached)... hash verified ("b8be66396c726fdc36ebb0f692ed8a8cca3bcc66")
 * binutils-2.30.tar.xz (cached)... hash verified ("574d3b5650413d6ee65195a4f5ecbddc3a38f718")
 * gcc-8.1.0.tar.xz (cached)... hash verified ("b34031ba9ff3e248b2c62de0825e49a1e0e01998")
Downloaded tarballs ... ok
Unpacking and patching ...
 * gmp-6.1.2.tar.xz
   o gmp-6.1.2_freebsd-configure.patch
 * mpfr-3.1.5.tar.xz
 * mpc-1.0.3.tar.gz
 * binutils-2.30.tar.xz
   o binutils-2.30_mips-gold.patch
   o binutils-2.30_nds32.patch
[... 103 lines cut ...]
2021-11-14 07:42:31-05:00 UNCHANGED build/qemu-coreboot/tools.cpio
48c2d27ec3887851c66d0813c4b34e7dab43a2a5830bdcb716a350503b57f5a1  /home/ubuntu/heads/build/qemu-coreboot/tools.cpio
2021-11-14 07:42:31-05:00 HASHES    build/qemu-coreboot/tools.cpio
2021-11-14 07:42:31-05:00 CPIO      build/qemu-coreboot/heads.cpio
2021-11-14 07:42:31-05:00 UNCHANGED build/qemu-coreboot/heads.cpio
751925f7ed2b76518f62b53f8cd2373296dba3e4879e8497f067276924289835  /home/ubuntu/heads/build/qemu-coreboot/heads.cpio
2021-11-14 07:42:31-05:00 HASHES    build/qemu-coreboot/heads.cpio
2021-11-14 07:42:31-05:00 MAKE coreboot
2021-11-14 07:42:39-05:00 DONE coreboot
# Use coreboot.rom, because custom output files might not be processed by cbfstool
"/home/ubuntu/heads/build/coreboot-4.8.1/qemu-coreboot/cbfstool" "/home/ubuntu/heads/build/coreboot-4.8.1/qemu-coreboot/coreboot.rom" print
Name                           Offset     Type           Size   Comp
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           15020 none
cpu_microcode_blob.bin         0x3bc0     microcode           0 none
fallback/ramstage              0x3c40     stage           49767 none
cmos_layout.bin                0xff00     cmos_layout       548 none
fallback/dsdt.aml              0x10180    raw              6952 none
fallback/payload               0x11d00    simple elf    7111621 none
(empty)                        0x6da100   null           153880 none
bootblock                      0x6ffa40   bootblock         880 none
2021-11-14 07:42:39-05:00 INSTALL   build/coreboot-4.8.1/qemu-coreboot/coreboot.rom => build/qemu-coreboot/heads--v0.2.0-1092-gfdbd9b2.rom
2021-11-14 07:42:39-05:00 UNCHANGED build/coreboot-4.8.1/qemu-coreboot/coreboot.rom
1f4a059521a590b7e30d97970792e88a376bb02a0a4a140753862324f680aafe  build/qemu-coreboot/heads--v0.2.0-1092-gfdbd9b2.rom
8a7b2920d1b81f98941aa0211eccbbb09f90685831f5733f53a9b01b8fecaff0  /home/ubuntu/heads/build/gawk-4.2.1/gawk

tlaurion, 2021-11-14:
@lukyjr 

It would be 

make BOARD=t430-flash CPUS=12
make BOARD=t430 CPUS=12

or if you have a Nitrokey Pro/storage or a Librem Key

make BOARD=t430-flash CPUS=12
make BOARD=t430-hotp-verification CPUS=12

where coreboot 4.13 boards and newer configs are under PR #1015 for review

And where coreboot 4.15 boards and configs are under work and a new PR should land this week.

lukyjr, 2021-11-14:
You know what... I swear I saw "BUILD=xxx" somewhere, some readme or in the wiki, but now everywhere I look now it's BOARD, BOARD, BOARD!
My mistake!

Sorry for the waste of time and thanks so much for the quick response.

closed 2021-11-14 by lukyjr