linuxboot/heads #1057: Can't get heads to boot on t430

lukyjr, 2021-11-15:
did a fresh clone of heads
pulled and cleaned the ME blobs for xx30
built t430-hotp-maximized roms
flashed top & bottom
upon booting, my t430 made a continuous beep with microphone and power lights blinking

flashed backup dump and the t430 boots. hardware is good.

then I tried just the t430-flash rom, and I get the same boot error: blank screen, blinking lights, continuous beep. heat sink fan starts up then immediately slows to a stop.

any ideas?

since the backup bios flash worked-- is there some way i could get a prebuilt t430-flash heads rom just for the purpose of testing, to confirm/deny that the rom build is the problem?

walliams, 2021-11-15:
mobo has an nvidia GPU. I read this is an extra feature, not on all t430 boards. could this have something to do with my issue?

tlaurion, 2021-11-15:
@walliams : building #1015 for coreboot 4.13 or master for coreboot 4.8.1?

tlaurion, 2021-11-15:
> mobo has an nvidia GPU. I read this is an extra feature, not on all t430 boards. could this have something to do with my issue?

most probably yes, others reporting working boards didn't specify if they were using Intel i915 only.
The current 4.8.1 coreboot configuration uses Coreboot native graphical initialization, but doesn't initialize libgfx.
Coreboot 4.13 #1015 PR is deactivating coreboot graphic initialization altogether. We might need to push something specific to support nvidia support, so your input here on what you built would be a base for @nitrosimon to bring that further!

Thanks for the report, that confirms my basic insight, where without hardware, wat I can do is limited.

Adding coreboot graphic initialization, forcing coreboot to use privary display and having libgfx into coreboot might resolve this issue.

daringer, 2021-11-15:
Hey @walliams 

you could also try https://github.com/Nitrokey/heads/tree/nitropad as reference, this is what we use inside our Nitrokey T430 Nitropads and "should" work, although they come without a discrete GPU. We'll share some more insights during this week, we especially @nitrosimon is working on it

walliams, 2021-11-22:
well, we're making progress!!
the t430-flash board from the nitropad branch plus a cleaned stock 8mb flash got me into the minimal heads and then i was able to mount a usb and flash the t430 rom of the same branch, but then... back to the same boot error >:[

i'm going to retry this process with a few more roms on the internal flash step, but at least the external flash hurdle has been cleared.

thank you all for the guidance and continued support on this project.

by the way, as to the discrete nvidia gpu on my board, i am not so much interested in enabling the gpu as i am in just being able to get heads up and running with hotp verification for my librem key. goal is to run qubes, for which i don't think the gpu would offer any advantages, at least for my intended purposes. however, feel free to use me as a guinea pig to test builds on this board while i'm still figuring out what will work on the internal flash.

walliams, 2021-11-22:
internal flash attempts:

nitrokey fork, nitropad branch "heads-t430-v1.3.1.rom" - failed, blinking
nitrokey fork, nitropad branch "nitropad-v1.3.1-5e5c36f.rom" - failed, blinking
nitrokey fork, master - "heads-t430-v0.2.0-957-gb017309.rom" - failed, blinking
heads master - "heads-t430-hotp-maximized-v0.2.0-1092-gfdbd9b2.rom" - failed, solid power light, blank screen

let me know if there's another build I should try that might work. i will try the coreboot 4.13 build next time i have a chance

daringer, 2021-11-22:
Hey, yes based on the current PR #1015 the only change to do is: `CONFIG_NO_GFX_INIT=n` inside the coreboot config for your board, then it should work, just verified. There will be an update to this PR to coreboot 4.15 soon, which will include this fix. But keep in mind that our t430 is w/o the gpu, which might lead to some further adaptations.

The maximized builds have to be flashed externally, your ME and IFD are set to read-only, you should see this if you fire up `flashrom -p internal`.

walliams, 2021-11-22:
> The maximized builds have to be flashed externally, your ME and IFD are set to read-only, you should see this if you fire up `flashrom -p internal`.

ahhhh i see. i knew that it could be flashed externally, but i didn't realize that it must. good to know. one other question: is there a reason that one would want to go with the hotp-maximized build over the hotp-verification build? seems they both serve the same purpose as long as you clean the ME in the external flash, but i'm sure i'm missing something. ideally is one method preferred over the other and for what reasons?

icequbes1, 2021-11-23:
> Hey, yes based on the current PR #1015 the only change to do is: `CONFIG_NO_GFX_INIT=n` inside the coreboot config for your board, then it should work, just verified. There will be an update to this [...]

Is there a workaround to having to reflash externally if we've built without the above change?

icequbes1, 2021-11-23:
> Is there a workaround to having to reflash externally if we've built without the above change?

Spoke too soon - connecting an external monitor to the VGA port shows the Whiptail interface (ERROR: GPG Keyring empty!). "x" followed by "[enter]" will get to the recovery shell.

icequbes1, 2021-11-23:
I can confirm that the fix from @daringer to specify `CONFIG_NO_GFX_INIT=n` in the coreboot config resolved the "blank screen" issue on my T430 (`make BOARD=t430`).

walliams, 2021-11-25:
@icequbes1 do you have the t430 board with the nvidia gpu? I am still not having any luck, even on the #1015 4.13 build with CONFIG_NO_GFX_INIT=n set in the coreboot config file (~/heads/config/coreboot-t430-hotp-maximized.config) I am still getting a blank screen. Backlight turns on, and ctrl-alt-del reboots, so it does seem to be booting now, but not displaying

tlaurion, 2021-11-26:
@walliams : dirty means you have stuff changed locally when compared to remote branch, which would be normal since you modified ~/heads/config/coreboot-t430-hotp-maximized.config

icequbes1, 2021-11-26:
@walliams, No my T430 does not have the Nvidia GPU.

It is (Intel) Integrated Graphics HD 4000 (CPU is i5-3320M).

tlaurion, 2021-11-26:
@walliams can you post the content of your whole t430-hotp-maximized coreboot config here? I could then expend it so its not in a defconfig format and check if the primary display is forced, which should, per other user reports, be Intel one. Otherwise without a board and back and forths, I'm blind here.

tlaurion, 2021-11-26:
@eganonoa by any chance, you own a t430 with nvidia dedicated GPU?

tlaurion, 2021-11-26:
FYI, heads store kernel and coreboot configs I'm defconfig format.

That is, if you
cp configs/coreboot-t430-hotp-maximized.config build/coreboot-4.13/.config

Then
cd build/coreboot-4.13
make menuconfig

Then exit and save, the result of the modified defaults kept in defconfig stored under heads tree + actual coreboot config will be stored under build/coreboot-4.13/.config

That is what is interesting here.

We need to figure out a common configuration for all xx20 and xx30 boards, dedicated GPU or not, so that internal display is always initialized and used by Heads.

That .config is the base we need to tweak so that a make savedefconfig is then saved under Heads tree, where a specific commit would lead into building a functional board config under specified kernel and coreboot versions for all variations of a board config.

This is why #1015 is not merged still. And I haven't anticipated so much complications and conflicting reports for a same board which I do not own.

This is why board owners need to test configurations and all reports need to be positive before a merge. Otherwise bricks will occur. And not all owners are technical enough to externally reprogram.

eganonoa, 2021-11-26:
> @eganonoa by any chance, you own a t430 with nvidia dedicated GPU? Your PR would suffer from the same community reported problems for t520 and t530 if coreboot doesn't deal properly with the possibility of having dedicated GPU.

@tlaurion.  No. We're running four plain vanilla t430's with integrated graphics, all running fine. But this whole thing I find a bit confusing.  The t530 I used _**does**_ have a dGPU and it runs totally fine with a board and coreboot config simply copied and repurposed from the standard t430 and x230 boards and configs. If you are fine with simply ignoring and not initializing the dGPU my experience with coreboot is that you just run the normal board and everything works fine.  This has been my experience with Libreboot on a w500 with a dGPU, vanilla Coreboot (on a t440p with a dGPU), and Heads (t530 with a dGPU). 

Thing are certainly more challenging if you are trying to actually use the dGPU. But that is a totally different thing. Based on my experience with the t530 with a dGPU (and indeed the other experience with Libreboot and Coreboot), I would suggest that unless @walliams is trying to do something to initialize the dGPU in their t430, there is something else going on that's causing this problem.

eganonoa, 2021-11-26:
> We need to figure out a common configuration for all xx20 and xx30 boards, dedicated GPU or not, so that internal display is always initialized and used by Heads.
> 

@tlaurion Based on my experience if you want the dGPU to work, you will need a different coreboot config than any with integrated-only boards. But if you don't want it, you should be fine with the vanilla configs you already have. I'm not sure you want to try to have one config that works for all, but rather just have a config file that helps people who want to initialize their dGPU to do so (inc. with the relevant VGA rom extracted from Bios and placed in the blobs folder, as well as whatever is needed for graphic init). I'm really not 100% convinced that the issue here is derived from the dGPU in the t430.

tlaurion, 2021-11-26:
@eganonoa the problem is not just with @walliams board, but other reports for the t430. From my understanding, he is not trying to use the dGPU. Coreboot is probably selecting it by default, if not forced to use Intel GPU. That is my only hypothesis at this point, and since I'm not a board owner, unfortunately I can only point to that from intellectual guess. 

But it doesn't work for some owners, so the default config is not enough.

tlaurion, 2021-11-26:
https://github.com/osresearch/heads/issues/1057#issuecomment-976965583

Is confirming some behavior fix. But not others, like here.

eganonoa, 2021-11-26:
> @eganonoa the problem is not just with @walliams board, but other reports for the t430. From my understanding, he is not trying to use the dGPU. Coreboot is probably selecting it by default, if not [...]

This just doesn't reflect my experience and certainly not the experience I've just had with the t530 and Heads. Very strange.

eganonoa, 2021-11-26:
> [#1057 (comment)](https://github.com/osresearch/heads/issues/1057#issuecomment-976965583)
> 
> Is confirming some behavior fix. But not others, like here.

@tlaurion but the @icequbes1 t430 **does not** have a dGPU. So that adds again another strange thing. As I say, we have four t430's running, all opened and flashed externally, all integrated boards and I've never had to do anything but build off master and go.  This is true for coreboot 4.8.1, coreboot 4.13 and coreboot 4.14.

tlaurion, 2021-11-26:
```
user@heads-tests:~/heads$ cp config/coreboot-t430-hotp-maximized.config build/coreboot-4.13/.config
user@heads-tests:~/heads$ cat build/coreboot-4.13/.config
# CONFIG_USE_BLOBS is not set
CONFIG_USE_OPTION_TABLE=y
CONFIG_STATIC_OPTION_TABLE=y
CONFIG_VENDOR_LENOVO=y
CONFIG_NO_POST=y
CONFIG_CBFS_SIZE=0xB80000
CONFIG_IFD_BIN_PATH="../../blobs/xx30/ifd.bin"
CONFIG_ME_BIN_PATH="../../blobs/xx30/me.bin"
CONFIG_GBE_BIN_PATH="../../blobs/xx30/gbe.bin"
CONFIG_HAVE_IFD_BIN=y
CONFIG_BOARD_LENOVO_THINKPAD_T430=y
CONFIG_LINUX_COMMAND_LINE="intel_iommu=igfx_off quiet"
CONFIG_UART_PCI_ADDR=0
CONFIG_HAVE_ME_BIN=y
CONFIG_HAVE_GBE_BIN=y
CONFIG_NO_GFX_INIT=y
CONFIG_DRIVERS_PS2_KEYBOARD=y
CONFIG_TPM_MEASURED_BOOT=y
CONFIG_CONSOLE_CBMEM_BUFFER_SIZE=0x80000
CONFIG_PAYLOAD_LINUX=y
CONFIG_PAYLOAD_FILE="../../build/t430-hotp-maximized/bzImage"
CONFIG_LINUX_INITRD="../../build/t430-hotp-maximized/initrd.cpio.xz"
user@heads-tests:~/heads$ cd build/coreboot-4.13/
user@heads-tests:~/heads/build/coreboot-4.13$ make menuconfig
configuration written to /home/user/heads/build/coreboot-4.13/.config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

user@heads-tests:~/heads/build/coreboot-4.13$ cat .config
#
# Automatically generated file; DO NOT EDIT.
# coreboot configuration
#

#
# General setup
#
CONFIG_COREBOOT_BUILD=y
CONFIG_LOCALVERSION=""
CONFIG_CBFS_PREFIX="fallback"
CONFIG_COMPILER_GCC=y
# CONFIG_COMPILER_LLVM_CLANG is not set
# CONFIG_ANY_TOOLCHAIN is not set
# CONFIG_CCACHE is not set
# CONFIG_FMD_GENPARSER is not set
# CONFIG_UTIL_GENPARSER is not set
CONFIG_USE_OPTION_TABLE=y
CONFIG_STATIC_OPTION_TABLE=y
CONFIG_COMPRESS_RAMSTAGE=y
CONFIG_INCLUDE_CONFIG_FILE=y
CONFIG_COLLECT_TIMESTAMPS=y
# CONFIG_TIMESTAMPS_ON_CONSOLE is not set
# CONFIG_USE_BLOBS is not set
# CONFIG_COVERAGE is not set
# CONFIG_UBSAN is not set
# CONFIG_HAVE_ASAN_IN_ROMSTAGE is not set
# CONFIG_ASAN_IN_ROMSTAGE is not set
CONFIG_HAVE_ASAN_IN_RAMSTAGE=y
# CONFIG_ASAN_IN_RAMSTAGE is not set
# CONFIG_ASAN is not set
# CONFIG_NO_STAGE_CACHE is not set
CONFIG_TSEG_STAGE_CACHE=y
# CONFIG_UPDATE_IMAGE is not set
# CONFIG_BOOTSPLASH_IMAGE is not set
# CONFIG_FW_CONFIG is not set

[... 690 lines cut ...]
# CONFIG_DEBUG_CONSOLE_INIT is not set
# CONFIG_DEBUG_SPI_FLASH is not set
# CONFIG_TRACE is not set
# CONFIG_DEBUG_FUNC is not set
# CONFIG_DEBUG_BOOT_STATE is not set
# CONFIG_DEBUG_ADA_CODE is not set
CONFIG_HAVE_EM100_SUPPORT=y
# CONFIG_EM100 is not set
CONFIG_NO_EDID_FILL_FB=y
CONFIG_WARNINGS_ARE_ERRORS=y
# CONFIG_POWER_BUTTON_DEFAULT_ENABLE is not set
# CONFIG_POWER_BUTTON_DEFAULT_DISABLE is not set
# CONFIG_POWER_BUTTON_FORCE_ENABLE is not set
# CONFIG_POWER_BUTTON_FORCE_DISABLE is not set
# CONFIG_POWER_BUTTON_IS_OPTIONAL is not set
# CONFIG_REG_SCRIPT is not set
CONFIG_MAX_REBOOT_CNT=3
# CONFIG_NO_XIP_EARLY_STAGES is not set
# CONFIG_EARLY_CBMEM_LIST is not set
CONFIG_RELOCATABLE_MODULES=y
CONFIG_HAVE_BOOTBLOCK=y
CONFIG_HAVE_ROMSTAGE=y
CONFIG_HAVE_RAMSTAGE=y

```

eganonoa, 2021-11-26:
@tlaurion how sure are we that we're not still just dealing with an issue caused by a locked IFD? The behaviour reported above is consistent with that and, while he says he's now unlocked, perhaps it wasn't really?  I would probably suggest @walliams goes and grabs [Skulls for the t430](https://github.com/merge/skulls/blob/master/t430/README.md) and see if that works.  If it does try the upgrade and if that fails then we know. I put my money on it working.

Beyond that perhaps try to use nvramcui to switch off any mention of the dGPU to the extent that it is in there (see e.g. [here](https://github.com/merge/skulls/blob/master/t430/README.md)).

tlaurion, 2021-11-26:
From there, `make menuconfig` on top of copied .config file, results in:
Devices->Graphics Initialization (Use libgfxinit)->Use libgfxinit
Devices->Display->Framebuffer->Linear "high-resolution" framebuffer
Note that under this, resolution of 2560x1600 is the default now, which might/might not be problematic.

Then
Exit until save prompt.

Then now .config looks like this:
```
user@heads-tests:~/heads/build/coreboot-4.13$ cat .config
#
# Automatically generated file; DO NOT EDIT.
# coreboot configuration
#

#
# General setup
#
CONFIG_COREBOOT_BUILD=y
CONFIG_LOCALVERSION=""
CONFIG_CBFS_PREFIX="fallback"
CONFIG_COMPILER_GCC=y
# CONFIG_COMPILER_LLVM_CLANG is not set
# CONFIG_ANY_TOOLCHAIN is not set
# CONFIG_CCACHE is not set
# CONFIG_FMD_GENPARSER is not set
# CONFIG_UTIL_GENPARSER is not set
CONFIG_USE_OPTION_TABLE=y
CONFIG_STATIC_OPTION_TABLE=y
CONFIG_COMPRESS_RAMSTAGE=y
CONFIG_INCLUDE_CONFIG_FILE=y
CONFIG_COLLECT_TIMESTAMPS=y
# CONFIG_TIMESTAMPS_ON_CONSOLE is not set
# CONFIG_USE_BLOBS is not set
# CONFIG_COVERAGE is not set
# CONFIG_UBSAN is not set
# CONFIG_HAVE_ASAN_IN_ROMSTAGE is not set
# CONFIG_ASAN_IN_ROMSTAGE is not set
CONFIG_HAVE_ASAN_IN_RAMSTAGE=y
# CONFIG_ASAN_IN_RAMSTAGE is not set
# CONFIG_ASAN is not set
# CONFIG_NO_STAGE_CACHE is not set
CONFIG_TSEG_STAGE_CACHE=y
# CONFIG_UPDATE_IMAGE is not set
# CONFIG_BOOTSPLASH_IMAGE is not set
# CONFIG_FW_CONFIG is not set

#
# Mainboard
#

#
# Important: Run 'make distclean' before switching boards
#
# CONFIG_VENDOR_51NB is not set
# CONFIG_VENDOR_ACER is not set
# CONFIG_VENDOR_ADLINK is not set
# CONFIG_VENDOR_AMD is not set
# CONFIG_VENDOR_AOPEN is not set
# CONFIG_VENDOR_APPLE is not set
# CONFIG_VENDOR_ASROCK is not set
# CONFIG_VENDOR_ASUS is not set
# CONFIG_VENDOR_BAP is not set
# CONFIG_VENDOR_BIOSTAR is not set
# CONFIG_VENDOR_BOSTENTECH is not set
# CONFIG_VENDOR_CAVIUM is not set
# CONFIG_VENDOR_CLEVO is not set
# CONFIG_VENDOR_COMPULAB is not set
# CONFIG_VENDOR_DELL is not set
[... 790 lines cut ...]
```
user@heads-tests:~/heads$ cat config/coreboot-t430-hotp-maximized.config 
CONFIG_USE_OPTION_TABLE=y
CONFIG_STATIC_OPTION_TABLE=y
# CONFIG_USE_BLOBS is not set
CONFIG_VENDOR_LENOVO=y
CONFIG_NO_POST=y
CONFIG_CBFS_SIZE=0xB80000
CONFIG_IFD_BIN_PATH="../../blobs/xx30/ifd.bin"
CONFIG_ME_BIN_PATH="../../blobs/xx30/me.bin"
CONFIG_GBE_BIN_PATH="../../blobs/xx30/gbe.bin"
CONFIG_HAVE_IFD_BIN=y
CONFIG_BOARD_LENOVO_THINKPAD_T430=y
CONFIG_LINUX_COMMAND_LINE="intel_iommu=igfx_off quiet"
CONFIG_UART_PCI_ADDR=0
CONFIG_HAVE_ME_BIN=y
CONFIG_HAVE_GBE_BIN=y
CONFIG_GENERIC_LINEAR_FRAMEBUFFER=y
CONFIG_DRIVERS_PS2_KEYBOARD=y
CONFIG_TPM_MEASURED_BOOT=y
CONFIG_CONSOLE_CBMEM_BUFFER_SIZE=0x80000
CONFIG_PAYLOAD_LINUX=y
CONFIG_PAYLOAD_FILE="../../build/t430-hotp-maximized/bzImage"
CONFIG_LINUX_INITRD="../../build/t430-hotp-maximized/initrd.cpio.xz"
```

tlaurion, 2021-11-26:
So, from above "full" (not defconfig), we can confirm that:
```
#
# Devices
#
CONFIG_HAVE_VGA_TEXT_FRAMEBUFFER=y
CONFIG_HAVE_LINEAR_FRAMEBUFFER=y
# CONFIG_MAINBOARD_HAS_NATIVE_VGA_INIT is not set
# CONFIG_MAINBOARD_FORCE_NATIVE_VGA_INIT is not set
CONFIG_MAINBOARD_HAS_LIBGFXINIT=y
CONFIG_MAINBOARD_USE_LIBGFXINIT=y
# CONFIG_VGA_ROM_RUN is not set
# CONFIG_NO_GFX_INIT is not set
# CONFIG_MULTIPLE_VGA_ADAPTERS is not set
```

And:
```
#
# Display
#
# CONFIG_VGA_TEXT_FRAMEBUFFER is not set
CONFIG_GENERIC_LINEAR_FRAMEBUFFER=y
CONFIG_LINEAR_FRAMEBUFFER=y
# CONFIG_BOOTSPLASH is not set
CONFIG_PCI=y
# CONFIG_NO_MMCONF_SUPPORT is not set
CONFIG_MMCONF_SUPPORT=y
# CONFIG_HYPERTRANSPORT_PLUGIN_SUPPORT is not set
CONFIG_HT_CHAIN_UNITID_BASE=0
CONFIG_HT_CHAIN_END_UNITID_BASE=0
CONFIG_PCIX_PLUGIN_SUPPORT=y
CONFIG_CARDBUS_PLUGIN_SUPPORT=y
# CONFIG_AZALIA_PLUGIN_SUPPORT is not set
CONFIG_PCIEXP_PLUGIN_SUPPORT=y
CONFIG_PCI_ALLOW_BUS_MASTER=y
CONFIG_PCI_SET_BUS_MASTER_PCI_BRIDGES=y
CONFIG_PCI_ALLOW_BUS_MASTER_ANY_DEVICE=y
# CONFIG_PCIEXP_HOTPLUG is not set
# CONFIG_EARLY_PCI_BRIDGE is not set
CONFIG_SUBSYSTEM_VENDOR_ID=0x0000
CONFIG_SUBSYSTEM_DEVICE_ID=0x0000
CONFIG_INTEL_GMA_HAVE_VBT=y
CONFIG_INTEL_GMA_ADD_VBT=y
# CONFIG_SOFTWARE_I2C is not set
# CONFIG_RESOURCE_ALLOCATOR_V3 is not set
CONFIG_RESOURCE_ALLOCATOR_V4=y
# CONFIG_XHCI_UTILS is not set
```

Which I think, from my best knowledge, should be enough. But @eganonoa @icequbes1 @walliams board owners should confirm, while regression testing for @alex-nitrokey @nitrosimon @jans23 should be done.

tlaurion, 2021-11-26:
> @tlaurion how sure are we that we're not still just dealing with an issue caused by a locked IFD? The behaviour reported above is consistent with that and, while he says he's now unlocked, perhaps it wasn't really by mistake? I would probably suggest @walliams goes and grabs Skulls for the t430 and see if that works. If it does try the upgrade and if that fails then we know. I put my money on it working.
> 
> Beyond that perhaps try to use nvramcui to switch off any mention of the dGPU to the extent that it is in there (see e.g. here).

@eganonoa from my experience, and logically, board owner would jot be able to type ctrl-alt-del.

Which is Head's Linux dealing with sysq magic there. We are dealing with graphic issues here, only.

tlaurion, 2021-11-26:
> Beyond that perhaps try to use nvramcui to switch off any mention of the dGPU to the extent that it is in there (see e.g. here).

Per actual setting, it would not be possible since we are hardcoding settings from setting:

CONFIG_USE_OPTION_TABLE=y
CONFIG_STATIC_OPTION_TABLE=y

eganonoa, 2021-11-26:
> > @tlaurion how sure are we that we're not still just dealing with an issue caused by a locked IFD? The behaviour reported above is consistent with that and, while he says he's now unlocked, perhaps it wasn't really by mistake? I would probably suggest @walliams goes and grabs Skulls for the t430 and see if that works. If it does try the upgrade and if that fails then we know. I put my money on it working.
> > Beyond that perhaps try to use nvramcui to switch off any mention of the dGPU to the extent that it is in there (see e.g. here).
> 
> @eganonoa from my experience, and logically, board owner would not be able to type ctrl-alt-del and reboot system.
> 
> Which is Head's Linux dealing with sysq magic there. We are dealing with graphic issues here, only.

@tlaurion Sorry, I meant nvramtool. The tool you use post-build to change certain CMOS values, not the one you build into the bios itself. You can use that to switch off the dGPU to the extent that it is in there.

tlaurion, 2021-11-26:
> @tlaurion Sorry, I meant nvramtool. The tool you use post-build to change certain CMOS values, not the one you build into the bios itself. You can use that to switch off the dGPU to the extent that it is in there.

@eganonoa two different things. If we want to be able to save nvram with nvramcui, we would need to remove  CONFIG_STATIC_OPTION_TABLE but also have working coreboot config at the first place :)

tlaurion, 2021-11-26:
> user@heads-tests:~/heads$ cat config/coreboot-t430-hotp-maximized.config
> CONFIG_USE_OPTION_TABLE=y
> CONFIG_STATIC_OPTION_TABLE=y
> # CONFIG_USE_BLOBS is not set
> CONFIG_VENDOR_LENOVO=y
> CONFIG_NO_POST=y [...]

So I would love dGPU owners to test this and report back.

icequbes1, 2021-11-26:
@tlaurion

On my T430 with iGPU-only, specifying `CONFIG_GENERIC_LINEAR_FRAMEBUFFER=y` in `heads/config/coreboot-t430.config` worked, in the sense that the internal display came up fine after initial flashing.

Qubes R4.1 booted fine with no obvious graphics issues, but that is the extent of my testing.

Branch tlaurion/maximized_boards-coreboot-4_13, commit acd60611, `make BOARD=t430`.

Only diff from your commit and my local change is replacing `NO_GFX_INIT` with `GENERIC_LINEAR_FRAMEBUFFER`:

```patch
diff --git a/config/coreboot-t430.config b/config/coreboot-t430.conf
index 5127195..c0f8b2a 100644
--- a/config/coreboot-t430.config
--- b/config/coreboot-t430.config
@@ -5,7 +5,7 @@ CONFIG_CBFS_SIZE=0x700000
 CONFIG_BOARD_LENOVO_THINKPAD_T430=y
 CONFIG_UART_PCI_ADDR=0
 CONFIG_CPU_MICROCODE_CBFS_NONE=y
-CONFIG_NO_GFX_INIT=y
+CONFIG_GENERIC_LINEAR_FRAMEBUFFER=y
 CONFIG_DRIVERS_PS2_KEYBOARD=y
 CONFIG_TPM_MEASURED_BOOT=y
 CONFIG_CONSOLE_CBMEM_BUFFER_SIZE=0x80000
```

tlaurion, 2021-11-26:
We need dGPU owners to test this to merge #1015.

walliams, 2021-11-27:
> We need dGPU owners to test this to merge #1015.

excellent work tlaurion. I am up and running with that new coreboot config you generated for the t430-hotp-maximized board. thanks so much for the help. now to figure out GPG key generation.

tlaurion, 2021-11-27:
@walliams thank you so much for your report. 

The issue you are reporting is in progress under https://github.com/osresearch/heads/pull/1063 which I haven't tested yet.

Sorry for the current mess. You are the guinea pig for the moment. Thanks for your feedback there and I will fix t4xx board configs following your report under #1015

walliams, 2021-12-03:
@tlaurion feel free to close this issue unless you believe there is work to be done yet. i was about to post again that my qubes 4.0 installer was freezing, but now that i have installed sufficient RAM in the machine (was 4gb, now 16gb) it seems to be running perfectly. i will reopen if I run into any issues i believe to be related to the firmware,

tlaurion, 2021-12-04:
#1015 merged

closed 2021-12-04 by tlaurion

walliams, 2021-12-27:
thanks again for the help @tlaurion. been smooth sailing for a month now with no major issues. i just wanted to ask out of curiosity: any idea how you could get heads to initialize the nvidia dGPU on boards like mine?

confirmed with pciids that the only recognized VGA controller is the iGPU. this is by no means an issue-- just wondering if it's a possibility.

eganonoa, 2021-12-27:
@walliams I have scripts, configs and instructions in a PR pending merger (#1080) for the t530 and w530 with dGPU that can easily be converted for the t430 to get the dGPU to run.  I just don't have a t430 with dgpu to actually do test it.  If you would be willing to test it, I would very happily put everything together for you.

walliams, 2021-12-27:
@eganonoa i would be very interested in testing for you! not looking forward to having to pull the machine apart for an external reflash if it doesn't work... so i'm especially interested if you are confident you can get me up and running with an internal flash!

let me know if there is any specific info you need from me to make this happen.
thanks!

eganonoa, 2021-12-27:
@walliams. So long as you are now running a maximized board (i.e. either t430-hotp-maximized or t430-maximized) I am very confident.  I have run and tested these builds multiple times on my dgpu T530 and w530 without error. And I would be simply taking the currently-working t430 boards in heads:master and simply adding the necessary for the dgpu following the t530 and w530 model. Nonetheless, of course, you would be testing something not tested before so there is risk involved.

Here's a link to a branch I've just created with the dgpu support for the t430: https://github.com/eganonoa/heads/tree/t430-dgpu. It takes current heads:master and adds specific boards for t430-dgpu-hotp-maximized and t430-hotp-maximized with the relevant dgpu extraction script and coreboot configs. I've successfully built these on my local machine running Debian 10. So you can build yourself. But it is also running on [CircleCi right now](https://app.circleci.com/pipelines/github/eganonoa/heads/40/workflows/6236bcce-7e6f-4ffd-9560-3f97b78bb9c8) and should have the roms ready in the artefacts folder there in about an hour or so.

Either way, once you have the 12 MB rom, you then need to take that rom and run nvramtool on it to change the default setting from integrated-only to hybrid (or discrete-only) before flashing. Instructions are in the README_vbios file in the xx30 blobs directory, but essentially all it takes is to build nvramtool and then run the following on the 12MB rom before flashing it: `nvramtool -C /path/to/12MB/rom -w hybrid_graphics_mode="Dual Graphics"` 

Once that's done the dgpu should be capable of working, with of course all the other OS-level things you need to do to make sure that the dGPU is working there.

eganonoa, 2021-12-27:
@walliams, the boards have built on CircleCi:

- [T430 dgpu hotp maximized](https://app.circleci.com/pipelines/github/eganonoa/heads/40/workflows/6236bcce-7e6f-4ffd-9560-3f97b78bb9c8/jobs/984/artifacts)
- [T430 dgpu maximized]( https://app.circleci.com/pipelines/github/eganonoa/heads/40/workflows/6236bcce-7e6f-4ffd-9560-3f97b78bb9c8/jobs/985/artifacts)

If you don't already have it, nvramtool can be gotten in two different ways:

- On debian-based systems via apt (`sudo apt install nvramtool`)
- Otherwise build from coreboot;

`git clone https://github.com/coreboot/coreboot.git
cd coreboot/util/nvramtool
make
sudo make install` (last part is optional)

See https://www.coreboot.org/Nvramtool.

walliams, 2021-12-27:
currently building t430-dgpu-hotp-maximized on debian 11
if successful, will run it thru nvramtool and reflash
any tips on how to retain my hotp verification settings on reflash?
thanks @eganonoa !

walliams, 2021-12-28:
done! the build worked with one small caveat-- my wifi card stopped initializing. ethernet is still good luckily.
lmk if you have any ideas to fix this @eganonoa. maybe i should move this to a new issue in your t430-dgpu branch?

eganonoa, 2021-12-28:
@walliams Strange. I made no changes over what's now in the current osresearch master related to the wifi adapter and indeed if I look at the difference between the generated coreboot .config of the regular t430-hotp-maximized and the t430-dgpu-hotp-maximized I don't see any difference that would relate to the wifi card. 

I'd suggest trying a couple things to double-check whether its this board that's causing this:

1. Check the cmos values in the rom you flashed to make sure that wlan is enabled:

`nvramtool -C /path/to/12MB/rom -a`

You should see "wlan = Enable" (I do). If not, change it to Enable and reflash.

2. Build the non-dgpu version of the board you used and flash that, just to double-check that there isn't something in the current osresearch:master that's causing this.

If the current rom you flashed does have wlan=Enable and the non-dgpu version of the current board doesn't disable the wifi card then I'd imagine it's something caused by this board.

tlaurion, 2021-12-28:
@walliams @eganonoa you both use Intel WiFi?

May be an error with x230 config having been used as base for all other xx30 boards, where in my tests, I always swap that Intel card with an Atheros,so never tested this.

tlaurion, 2021-12-28:
@walliams @eganonoa nevermind my last comment, Intel Wifi card is initialized in coreboot per default config, which isn't changed in defconfig provided in tree

`CONFIG_DRIVERS_INTEL_WIFI=y` is included. So no hypothesis here.

eganonoa, 2021-12-28:
> @walliams @eganonoa nevermind my last comment, Intel Wifi card is initialized in coreboot per default config, which isn't changed in defconfig provided in tree
> 
> `CONFIG_DRIVERS_INTEL_WIFI=y` is included. So no hypothesis here.

@tlaurion Yep it's a default value it seems (e.g. librem configs have to specify that its not set). As a general matter I eventually switch everything to Atheros because we use an FSF-certified distro. But typically I wait until I've confirmed everything is working with heads (and whatever old device I've managed to source) before investing in upgrades. That means that all my boards get tested on both Intel and Atheros cards, which has been true of both the t530 dgpu and w530 dgpu boards. And I can confirm that both those boards work out-of-the-box on both types of wifi card. The t430 dgpu-version uses the very same dgpu as the t530. So a wifi issue is surprising here. Fingers crossed for an errant change in the cmos values before flashing or something similar.

walliams, 2021-12-28:
checked with nvramtool and wlan = Enabled
also, noticed bluetooth was enabled and I don't see a bluetooth controller when I run lspci so I disabled that on a hunch and reflashed, but it didn't change anything with the wifi. I'm going to make the board from the master branch now and see how that goes.

tlaurion, 2021-12-28:
@walliams basic troublshooting first.

Ils the wifi slider off?

walliams, 2021-12-28:
i figured it out! somehow on reflashing, the Centrino card was dropped off of my sys-net "Devices" list. added it back in and it's working.!
so, not a problem with heads, but i guess something to be aware of that could cause people trouble with existing qubes installations.

thanks again to both of you for the advice. i am now very happy with this heads rom. 
now if only i could get my fedora template to boot with the nvidia drivers :(
@eganonoa since you have gone to some lengths to enable dGPU's in heads, any rec's on a qubes template that plays nice with nvidia drivers? I read something about the fedora kernel being an issue. I was running fedora-34 in a cloned template, and now it still starts but no apps will open, not even terminal.

eganonoa, 2021-12-28:
@walliams Good stuff. Glad this worked out. I'm not a regular Qubes user. But I do have one (non-Heads, but coreboot nonetheless) device running Fedora 35 (silverblue). Getting the nvidia card set up was no problem at all (on 33, 34 and 35). I just have the nvidia-470xx driver installed (esp. xorg-x11-nvidia-470xx) and use system76-power (via this COPR https://copr.fedorainfracloud.org/coprs/szydell/system76/) for graphics switching.

@tlaurion I have set up a new clean branch for this rebased on master, one commit, etc (see https://github.com/eganonoa/heads/tree/dgpu-t430). Happy to make a PR for it, though obviously I can't test on it going forward and will rely on @walliams and others with a t430 dgpu to test. Let me know.

walliams, 2021-12-28:
@eganonoa great. qubes doesn't provide a fedora-35 template yet, but it looks like i can upgrade from a 34 template. I'll try that, then i might try an ubuntu template if that won't work. goal is to make this an all purpose machine as well as a security beast.

really appreciate the help getting setup. i'm happy to test any boards you are trying to merge with master for the good of the project. feel free to mention me in a thread if you have another t430-dgpu build that needs testing. obviously i would appreciate if you verify that the updates work on your T530 first to minimize risk ;)

would you like me to verify that the new t430-dgpu branch you just made is working, or is it essentially the same as what I just built?

eganonoa, 2021-12-29:
@walliams You should be able to get it to work with Fedora 33 and 34 also. I certainly have, always using the latest nvidia driver and system76-power. Ubuntu probably the same, though I have no experience with it beyond servers. 

Thanks re testing on the t430-dgpu going forward. The latest branch is just a copy of the one you used, just with the additions to master (with dgpu boards for t530 and w530) that happened today included. So there's nothing new related to what you've tested already.

tlaurion, 2021-12-29:
@walliams @eganonoa 

About https://github.com/eganonoa/heads/tree/dgpu-t430

There are validation concerns (nothing gets validated because of lack of { do this | pipe here } || { echo failing; exit 1 } thorough code as reported through Matrix @eganonoa 

Also, those scripts lack manual installation of bundle in apt install, and then upgrading to proper bundle version. 

I'm not sure yet why Qubes fails to call rom-tool without sudo in its call to UEFIExtract in your scripts.

But that should be looked on independently prior of merging in another PR fixing other vbios extract scripts and then deduplicating here.

No problem on debian-10 docker image. 

@walliams is https://github.com/eganonoa/heads/tree/dgpu-t430 working for you?

walliams, 2021-12-29:
> @walliams is https://github.com/eganonoa/heads/tree/dgpu-t430 working for you?

@tlaurion no problems for me so far. the vbios script took two tries-- one without sudo and then one with. i can't say i understand why this was necessary.

i had some issues installing nvramtool... the command wouldn't call the script, so i had to manually identify it after cloning the coreboot git.
But once I got the rom built & flashed and figured out why my wifi went out i have had no problems

walliams, 2021-12-30:
to be clear though, I am still having trouble getting the dGPU to run properly on the machine. every time i add it into a qube my system freezes up... but that's more likely a comination of qubes & user error than a heads build issue.

tlaurion, 2021-12-31:
> to be clear though, I am still having trouble getting the dGPU to run properly on the machine. every time i add it into a qube my system freezes up... but that's more likely a combination of qubes & user error than a heads build issue.

Those needs to be HVMs, and pci/gpu passthrough guides need to be followed.
If you have issues there, I would recommend posting under Qubes forums and come back to create a new seperate issue if something inside of coreboot/heads is missing.

Some links:
- https://forum.qubes-os.org/t/qubes-gpu-passthrough/661
- https://www.qubes-os.org/faq/#can-i-run-applications-like-games-which-require-hardware-acceleration