linuxboot/heads #1065: Heads on w530 (dGPU wanted?)

eganonoa, 2021-11-29:
I have successfully built and flashed a w530 with coreboot 4.14 using the maximized boards and flash (see my amended PR #1059). As with the T530 this was tested with the dGPU (I believe all W530's must have the dGPU), building off of master with minimal changes. The issue that appears to impact t430 boards with the dGPU (#1057) does not appear to impact the T530 or W530. However, I have built acting as if the dGPU is not present on these boards at all, and as such they are not initialized. I see some advantages to this, specifically:

1. Lower power consumption means that you do not need the **giant** 135w charger to power the device and charge the battery and get a bit better battery life generally;
2. You can make full use of the additional super-chunky heat sinks that the T530 and W530 models with the dGPU have keeping things incredibly cool

As the dGPU in the T530 is old and not really useful today, I had decided not to bother working to build coreboot in order to activate it. However, as I understand it the dGPU's in the w530 are a bit more useful. I would be happy to play around with figuring out how to get it working if other owners of w530's would find it interested and could help with some testing.

tlaurion, 2021-12-04:
The dGPU is an interesting feature for Qubes passthrough and gaming, while I see this being pertinent more in platforms like the kgpe-d16 where a performant GPU is chosen and initialized, let it be for mining and or a gaming workstation.

Since I am not a gamer myself and the dGPU of those machines is quite old, I would leave for comment from the community to decide or not if this beast would be interesting or not to be enabled properly. 

That would be the resort of the OS though. Coreboot has little to play here, if not already done, setting ACPI tables correctly so that the PCI device is made available to the OS with proper VBT blobs, which I think were pushed with corehoot 4.13

eganonoa, 2021-12-12:
@tlaurion: It turns out that the w530 cannot use an external display (either in the dock or directly) if you do not have the dGPU initialized. For some this won't be an issue, but for others (like me) it will be necessary. So I have done that and have everything working.  I have a working board for this. However, I am wondering whether or not this is something that should be added in as a new board (e.g. w530-dgpu) or whether it is something better just to have documented for people to do on their own. It is a multi-stage process, and one that might be a challenge to automate. The process is as follows:

1. Extract the integrated and discrete VGA roms. A tool such as VBiosFinder (https://github.com/coderobe/VBiosFinder) can automate this. But it is not included in Heads and isn't the simplest thing to build for any given system.
2. Save and rename the roms, placing them into the xx30 blobs directory. I use the relevant ID as the file name. The standard VGA bios is 8086,0166.rom; the dgpu is either 10de,0ffc.rom (for the K1000m) or 10de,0ffb.rom (for K2000m)
3. Amend the relevant coreboot config (or if like me creating a new board, create a new one) to include the following:

`CONFIG_VGA_BIOS_DGPU_ID="10de,0ffc"
CONFIG_VGA_BIOS_DGPU_FILE="../../blobs/xx30/10de,0ffc.rom"
CONFIG_VGA_BIOS=y
CONFIG_VGA_BIOS_FILE="../../blobs/xx30/8086,0166.rom"
# CONFIG_VGA_BIOS_SECOND is not set
CONFIG_VGA_ROM_RUN_DEFAULT=y
CONFIG_VGA_BIOS_DGPU=y`

Note here to change the CONFIG_VGA_BIOS_DGPU_ID and CONFIG_VGA_BIOS_DGPU_FILE depending on which dGPU is present.

Note also, I'm using the generic framebuffer coreboot config as per my earlier PR #1078.

4. After building, make the nvramtool in the relevant coreboot build's util directory and using nvramtool on the newly generated rom change the settings for hybrid_graphics_mode from Integrated Only to Dual Graphics. 

After this, you will have a working dGPU on your w530 (assuming of course you have the relevant driver installed in your OS).

All of this can easily be put into new boards and I'm happy to make a PR, but note (a) a lot of work would be needed to make it automated for CircleCI; and (b) you will end up with **_two more_** boards at least (i.e. w530-dgpu-hotp-maximized and w530-dgpu-maximized), if not four more boards if you are aiming to run via CircleCI and produce roms for people to use (i.e. w530-dgpu**-K1000-**hotp-maximized and w530-dgpu-**-K1000-**maximized w530-dgpu-**-K2000-**hotp-maximized and w530-dgpu-**-K2000-**maximized.

Other option is to just stick with the w530 not initializing the dGPU and simply document (a) the lack of external monitor without initialization; and (b) the process above. 

Let me know what you'd like me to do here.

tlaurion, 2021-12-12:
From what I read, user need to known if he owns a k1000m or k2000 first. Then have boards with instructions building for those. Or better, scripts downloading and extracting vbios blobs for them where boards specify proper coreboot specifics for them.

I'm not clear of who would he happy of having a w530 without expected VGA external display.

A board is normally fully functional or not. And here we talk about having multiple w530. Without external display, k1000m and k2000m

eganonoa, 2021-12-13:
@tlaurion my experience with coreboot/heads is that you are always sacrificing some functionality or other. And there are definitely some people that use desktop replacements like the w530 specifically to replace using an external display. So as long as it's properly documented, I'd suggest the non-dGPU one is useful. It's also the one that is probably the most reproducible right now in terms of CircleCI. Will need to do a lot more to automate the dGPU process. So I tend to agree that what's needed is to put documentation in to highlight what you get with each board, what you don't get and what you need to do to get more, much like what coreboot has (e.g. https://doc.coreboot.org/mainboard/lenovo/t440p.html). 

Nonetheless, I have working boards to share. Will just need to put together a script to install the VBiosFinder dependencies, pull the bios update, and extract the roms. And then we can have individual boards for (at a minimum0, w530 K1000, w530 K2000, t530, t530 dgpu.

eganonoa, 2021-12-13:
@tlaurion Have a working extractor script for the dGPU bios that tracks the concept for the me.bin, pulling the bios update from Lenovo's website, building and using VBiosFinder to extract. Tested on Debian 10 and Ubuntu 21.04 and can be fully automated. Was easier than I thought. Will raise a PR tonight or tomorrow. But for now it's available in my [dGPU branch](https://github.com/eganonoa/heads/tree/w530_t530_dgpu) along with working and tested configs for T530 dGPU and w530 dGPU (tested K1000M).  So closing this issue here now in advance of the PR.

closed 2021-12-13 by eganonoa