linuxboot/heads #1066: Trouble with ECC keys and GPG digest algo

Unb0rn, 2021-11-30:
While trying to make Purism's Pureboot to work with a Nitrokey containing ECC keys (brainpoolp384r1 for signing) I got stuck.
It looks like kexec-sign-config script script has a hardocded value of sha2-256 but I think those functions require something larger like sha2-384...
I've also tried creating an issue in Purism's [repo](https://source.puri.sm/firmware/pureboot/-/issues/22) but this fork repo is not so active...

tlaurion, 2021-11-30:
@MrChromebox

MrChromebox, 2021-11-30:
@tlaurion I don't see how this is a Pureboot (vs Heads) issue.

Also looks like I'm not getting notifications by default for newly-created issues on the Pureboot repo, so I'll have to fix that

tlaurion, 2021-11-30:
@Unb0rn the first step would be to run this from the command line to give output, otherwise we have to generate keys ourselves. My insight is that the kernel might not have the requirements to deal with this cipher requirement.

Can you run the following from recovery shell and give screen captures?
`kexec-sign-config -p /boot -u`

Unb0rn, 2021-12-02:
@tlaurion looks like my suspicions were correct:
```
gpg using ECDSA key
gpg: ECDSA key requires a 384 bit or larger hash (hash is SHA256)
gpg: Can't check signature: Genereal error
Invalid signature on kexec boot params
```
No other errors present.

Unb0rn, 2022-07-21:
@tlaurion Do we really need to specify the digest algo in `kexec-sign-config`?
After removing it and enabling CONFIG_CRYPTO_SHA512(with 384 obviously) everything seems to be ok...

I am asking because it seems like using ECC keys requires using the corresponding digest algorithms (looks like you can detach-sign something with 384 bit EC using 256-bit sha2, but at the same time its impossible to verify such a signature with `gpgv`. It also seems like its impossible to sign anything with nistp521 using hashes smaller than sha2-512 `gpg: signing failed: Invalid length` or to use sha2-512 with brainpoolp384 `gpg: signing failed: Conditions of use not satisfied`)
So to support ECC keys we either need to introduce some checks of key type and length or trust gpg to select the proper algo?

tlaurion, 2022-07-21:
@Unb0rn as of today, if key gen inside of Heads supports proper algos, and Heads scripts can remove hashing specifications, I think we should go that way.

I also think the factory reset / re-ownership wizard should evolve in giving alternatives to its current hardcoded rsa-3072 option.

We could iterate on that.
If you have a PR that works for your use case, I'll make sure it doesn't break things.

PR #1194 (2022-07-29): EC keys support
EC signatures require the corresponding digest length. Enabling gpg to auto select the algo and supporting those algos in kernel should fix #1066
closed 2022-07-29 by tlaurion