linuxboot/heads #1067: oem-factory-reset

tlaurion, 2021-12-01:
The PINS should be given back to the end-user:

User PIN by default is: 123456
Admin PIN by default is: 12345678

Otherwise end-user needs to google stuff around. Heads should tell what passphrases are.

@alex-nitrokey @nitrosimon @jans23

tlaurion, 2021-12-01:
[image]
[image]

tlaurion, 2021-12-01:
The PINS should be given back to the end-user:

User PIN by default is: 123456
Admin PIN by default is: 12345678
TPM Ownership PIN: 12345678

Otherwise end-user needs to google stuff around to/from Nitrokey/Purism/Other brands of GPG enforcing USB Security dongles. Heads should ease that process and tell what passphrases setuped are.

@alex-nitrokey @nitrosimon @jans23 @MrChromebox 

Note: when it comes to manually selecting passphrases, I personally prefer when those are not echoed back in plaintext, but when there is a prompt requiring me to type it to confirm my input. That would be valid for all secrets to be typed, but that is just a personal opinion.

https://github.com/osresearch/heads/pull/1063#issuecomment-984088318

jans23, 2021-12-02:
I understand during OEM factory reset also Nitrokeys are reset. In this case HEADS could use the default PINs automatically without user intervention.

tlaurion, 2021-12-03:
> I understand during OEM factory reset also Nitrokeys are reset. In this case HEADS could use the default PINs automatically without user intervention.

@jans23 this is the case here. The thing is that the end user is not told what thoe default PINs are. We imply they do. It's never written on the screen, since the output is burried into redirected output.

[This is what the user sees](https://user-images.githubusercontent.com/827570/144450329-6f9dc2f5-b3bc-4887-82c2-121090c12d03.jpeg)

(Without the error shown on screen because it was troubleshooting. Image comes from https://github.com/osresearch/heads/pull/1063#issuecomment-984728806)

jans23, 2021-12-06:
> We should not imply that an end user knows what is the value of those default PINs unless we imply that the users search for those online and dig GPG documentation.

As the name implicates, the OEM Factory Reset aims for OEMs (like Nitrokey), not for end users. We use it to prepare laptops in our workshop.

I agree that User PIN, Admin PIN and TPM PIN should differ and ideally changed from default values to custom values. 

> (or generating them from Diceware dictionnary for user/OEM in ALL cases)

Nitrokeys uses PINs and not passphrases. The difference is that PINs can be short (e.g. 6 digits) and in combination with the device's PIN counter provide sufficient strong protection against brute force attacks. Therefore no need for long passphrases.

Not sure if this answers all your questions. Does it?

daringer, 2021-12-06:
I think simply outputting the PINs is clearly something that imposes no disadvantage for the vendor (e.g., Nitrokey), but still serves as a certain added value for the end-user doing a factory reset for whichever reason. So here I would fully second this approach.

> I would advise into at least having custom PINs prompt asking for 2 distinct PINs

Also cool with that, but there should be a option for OEMs to stick to default pins. Both always come with trade-offs, means either a "default pin" or a "communicated pin" (even if the OEM sets an random pin, this has to be communicated to the user - this is very customer unfriendly and not really an option).

This being said, the imho "cleanest" solution would be to have something like a `first start` flag (which is set after a oem-factory-reset), which will lead to a prompt for the user (forcing him/her) to update the pin. The `first start` flag would then only be cleaned up if the user successfully sets a non-default pin, which would remind the user to update the pins (with the drawback that this would not be cleared if the user chooses to change the pin outside of heads).

tlaurion, 2021-12-07:
Pureboot way is to burn one PIN attempt changing it.

daringer, 2021-12-07:
Means Pureboot will try 1234 on every boot and warn the user accordingly if the default pin is set ?

MrChromebox, 2021-12-07:
@daringer correct. See:
https://source.puri.sm/firmware/pureboot/-/commit/c912f256fde7bcd0039e23170260f629521fb819#a6c94f7187de316661024d0ce4f8837609a83516_89_91

closed 2022-03-11 by tlaurion