linuxboot/heads #1077: Bundle a Release

eganonoa, 2021-12-09:
The last release from Heads:Master is from April 2017. Thanks to the incredible work of the main folks involved, Heads has moved on significantly since then. From x230 and the Librem 13v1, we now have 8 Purism boards, including the server, the kgpe server/workstation board, and nearly all thinkpad 2- and 3- series devices.  There's the amazing GUI.  The excellent Nitrokey/LibremKey tools. And now the huge task of migrating to a fairly recent coreboot build (4.13) complete. It seems to me that now is an appropriate time to bundle up the Heads:Master into a release to serve as a nice, working point in time, before a lot of the heavy lifting starts with the move to coreboot 4.15 and a 5.x linux kernel, which will likely require quite some tinkering and potentially some added instability in the build process. I don't think that actual rom's need to be added. But it would be good to have all these gains consolidated in a release on the main page.  Right now all the development seems a little hidden with such an old release sitting there on the releases page.

icequbes1, 2021-12-12:
I agree. While I personally would always build Heads myself, having a reference and a clearly defined state is a plus for debugging future issues.

tlaurion, 2021-12-12:
@icequbes1 @eganonoa for transparency needs, this is currently discussed. The thing is that the repository ownership is currently not permitting to create tags and releases by anyone else then the owner.

The solution proposed would be to move the repos (heads and heads-wiki) and own a new domain to replace osresearch.net where heads-wiki is currently rendered.

I am not completely comfortable moving the community and looking into other options so that the current repos and current documentation stays the same, while opening ownership of the project or have @osresearch do those tags.

While ongoing problems (moving back to gitlabci, trying github actions etc) needs some kind of ownership delegation or something which I'm not really clear hownit could work so that @osresearch is not needing to be the one doing those changes.

Any idea welcomed.
@jans23 @mrchromebox @kylerankins

eganonoa, 2021-12-13:
@tlaurion Wow, that's a lot more complicated than I thought!  Sorry about that.  Obviously, moving off of @osresearch ownership and moving the community around would be horrible. So you are definitely right about that!

eganonoa, 2021-12-14:
@tlaurion One option could be to simply update the README and highlight that osresearch:heads is a rolling release, but that various organizations have fixed releases built off of it.  You have Purism's Pureboot releases; Nitrokey has releases specific to their computers; You have releases with nice release notes.  Last week I bundled a [release](https://github.com/eganonoa/heads/releases) on my organization's fork simply to create a fixed point in time for debugging purposes because of the rolling-release model over here. The main README needs a refresh anyway as it really does not highlight how actively developed Heads is, nor that it is use commercially.

q2dg, 2026-08-17:
Well...have you arrived to any solution?
Thanks!

tlaurion, 2026-08-17:
Points addressed, maybe it should be more clear and contributions welcomed

Done:
- I have the rights to do a release. 
- documentation contains vendors to show who do releases for commercial solutions

Todo:
- have Circleci comment on PR so that hashes for each boards produced final rom files have posterity
- do tags for a release
- do release on last commit produced roms so hashes match commit (each commit changes hashes so chicken egg problem since commit hash part of /etc/config being part of tools.cpio being part of initrd.cpio.xz being part of final rom. 

May edit after re-reading whole content of issue to point to tracking issues missing before doing point release. 

Meanwhile, please point if I missed something.