linuxboot/heads #1083: Fan Speed on T430

walliams, 2021-12-27:
iI'm not sure if this is heads related or what, so feel free to close if this is an OS/driver problem.

my CPU fan is jumping from half speed to full speed and then back to half speed rapidly when doing heavy processing. doesn't seem to be any in between. it would be nice if there were manual control or better automatic control, possibly in the heads gui menu.

Thrilleratplay, 2021-12-27:
@walliams What you are probably looking for is [Thinkfan](https://github.com/vmatare/thinkfan).  It is can be frustrating to set up but once you have it the way you want it, it is something that you'll be able to forget about.  There are a few GUIs, like [Thinkfan-UI](https://github.com/zocker-160/thinkfan-ui) which may help you tune your settings but ultimately you should not need them.

tlaurion, 2021-12-27:
Improvements welcome/moving out this section to not be hidden deep down for a single board, welcome

https://osresearch.net/InstallingOS/#installing-extra-software

SvenSemmler, 2021-12-27:
I run thinkfan on a T430 with an i7-3840QM. Here's what's working well for me:

1) install thinkfan
2) add these lines to /etc/defaul/thinkfan

START=yes
DAEMON_ARGS="-q -b 1 -s 3"

3) add these lines to /etc/thinkfan.conf

tp_thermal /proc/acpi/ibm/thermal
tp_fan     /proc/acpi/ibm/fan

# t_junction 105
# t_max       90
# t_idle      65

(0,     0,      55)
(1,     45,     60)
(2,     50,     65)
(3,     55,     70)
(4,     60,     75)
(5,     65,     80)
(6,     70,     85)
(7,     75,     90)
(127,   80,     32767)

-- 
  public key: https://www.svensemmler.org/2A632C537D744BC7.asc
fingerprint: DA59 75C9 ABC4 0C83 3B2F 620B 2A63 2C53 7D74 4BC7

walliams, 2021-12-30:
thanks for the config @SvenSemmler 
say i'm running qubes... i'm very hesitant to install anything to dom0. should i then install thinkfan in all my heavy usage vm's, or perhaps just one template that is most likely to need it? not sure where i should start with this.

SvenSemmler, 2021-12-30:
On 12/30/21 4:33 AM, walliams wrote:

> say i'm running qubes... 

Please post to ***@***.*** or use https://forum.qubes-os.org/ for further discussion about this, so we don't overstay our welcome here.

> i'm very hesitant to install anything to dom0

You should. However, that's the only place where installing thinkfan will have any effect. DomU's can't access your fans. A simple ...

	sudo qubes-dom0-update thinkfan

... will do the trick.

/Sven

-- 
  public key: https://www.svensemmler.org/2A632C537D744BC7.asc
fingerprint: DA59 75C9 ABC4 0C83 3B2F 620B 2A63 2C53 7D74 4BC7

closed 2021-12-30 by walliams