linuxboot/heads #1086: make: *** [Makefile:668: all] Error 2

christmasisforturkeys, 2022-01-02:
@tlaurion I have been following this guide [Zanthed t430 heads guide](https://github.com/Zanthed/t430-heads-guide) and am trying to build heads tools on arch and am receiving this error: `make: *** [Makefile:668: all] Error 2` and `make[1]: *** [Makefile:455: /home/beeswax/heads/build/json-c-0.14/.configured] Error 127`. 

Is this perhaps because I'm trying to build heads on arch, should I try a Live usb with debian or ubuntu 21.04? I have attached my full build log below but it seems pretty uneventful apart from the 2 errors I've already mentioned. 

I'm pretty new to all of this but I'm very willing to learn and research further if someone can point me in the right direction, as I'm starting to chase my tail here, thanks in advance

[coreboot build log.txt](https://github.com/osresearch/heads/files/7798385/coreboot.build.log.txt)

closed 2022-01-03 by christmasisforturkeys

AdmerStroh, 2022-01-05:
Hey, I get the same error 
`make: *** [Makefile:668: all] Error 2`
when building the x230-hotp-maximized board on a clean debian 11 with all dependencies installed..
@christmasisforturkeys how did you resolve the issue?

Thanks in advance!

christmasisforturkeys, 2022-01-05:
I used Ubuntu 20.04 and it compiled fine after one further error, still not had time to flash it though, super busy with work recently. I'm certainly no expert in this for you to be asking my advice though I have to say.

Although that being said I did run into a further error that I managed to solve and it may be the same problem for you.. It is so far undocumented in the main instructions but it is in the readme in the blobs folder of your cloned heads directory. 

As per tlaurion's response to copyvar's issue; https://github.com/osresearch/heads/issues/1036#issuecomment-954340542

Once I ran that script I was able to build a full rom, which as I mentioned earlier I've yet to have time to flash.

Good luck

tlaurion, 2022-01-05:
@AdmerStroh on Ubuntu/Debian?

AdmerStroh, 2022-01-05:
@tlaurion I am on Qubes 4.1 and use the debian 11 template. Thanks for your replies. I resolved my issue by compiling on debian 10 instead of debian 11. I am not sure but maybe some of the build dependencies differ in debian 11?