linuxboot/heads #1087: Swap Fn and Ctrl key

copyvar, 2022-01-02:
I successfully built heads x230-hotp-maximized for my X230. I installed classic keyboard (see https://www.thinkwiki.org/wiki/Install_Classic_Keyboard_on_xx30_Series_ThinkPads). I want to swap the Fn and Ctrl key. How can I do this?

Does heads support all types of authentic batteries or do I have to allow the use of "non official" batteries via thinkpad-ec beforehand? If so, is there an easy way to apply later on (after Heads building and external flashing)?

I am confused about this, because thinkpad-ec is not responsible for allowing the free use of any wifi card. At least, that is my understanding.

tlaurion, 2022-01-03:
> I successfully built heads x230-hotp-maximized for my X230. I installed classic keyboard (see https://www.thinkwiki.org/wiki/Install_Classic_Keyboard_on_xx30_Series_ThinkPads). I want to swap the Fn and Ctrl key. How can I do this?

Through nvramtool from your OS (debian deploys it) or by building it yourself inside of the coreboot tree.
This is standard cmos settings, and can be configured at build time by changing cmos.default or through nvramtool.

[You can do this from EC, or through coreboot.](https://github.com/merge/skulls/issues/57)

For your two other questions, those should be separate issues, linked to documentation, not code.

Better suited for heads-wiki. Still:

- Battery whitelist is from [EC](https://github.com/hamishcoleman/thinkpad-ec#intro), not firmware.
- Wifi withelisting is from firmware, not EC. When using coreboot, you can use whatever wifi chipset you want. That is, flash first, then replace wifi.

copyvar, 2022-01-08:
@tlaurion Thanks for your reply. Could you please write a step by step instruction on what to do (which files to modify where and when) for building process? As I understand I can rebuild heads with specific options and flash it again via external programmer?

tlaurion, 2022-01-08:
@copyvar 
Instructions are in board configs for normal instructions builds.

Using unapproved batteries is unsupported by Heads. thinkpad-ec project covers it.

For your use case of swapping FN and Ctrl keys, you can build normally, and then use nvramtool from OS, or modify built ROM with CMOS overrides prior of building the ROM.

For your additional question here on how to build ROMs, please open additional issues.

This one is for swap of FN and Ctrl key. Let's keep issues separated otherwise nobody but you will find answer to their questions.

copyvar, 2022-01-08:
@tlaurion Thanks again. And thank you a lot for the hint using nvramtool. Could you please add a step by step instruction on how to do that? That would be great, if I would not have to flash it again, but I do not have the skills to understand how to apply nvramtool properly. I tried to understand, but I lack a lot of background knowledge I guess.

copyvar, 2022-01-08:
Btw: how can I find out, which EC options I flashed (to the BIOS?)? You see I am still confused a bit about how everything works together.

tlaurion, 2022-01-08:
@copyvar 

I do not use it.
You could hijack t430 issue and ask users there, or referred ticket to skulls to do the same.

Doc https://www.coreboot.org/Nvramtool

Option to modify:
fn_ctrl_swap=Enable

tlaurion, 2022-01-08:
> btw: how can I find out, which EC options I flashed (to the BIOS/to the mainboard?)? You see I am still confused a bit about how everything works together.

I do not understand the question.
If you used thinkpad-ec project, options chosen there when modifying original firmware should be known?

Modifying EC happens only through lenovo BIOS upgrade over original firmware. If you are unelder Heads, you would have to flash back original backups, use thinkpad-ec tool to modify EC as needed, flash, and then flash back Heads.

That would be needed for removing battery whitelist. Not needed for FN CTRL keyboard key swap.

tlaurion, 2022-01-12:
@copyvar  (replying here for your https://github.com/osresearch/heads/issues/711#issuecomment-1010607143 comment)

> nano coreboot/src/mainboard/lenovo/x230/cmos.default 

will modify the cmos.default to be included in a local coreboot build... Which in what you are doing, have nothing to do with Heads locally used coreboot version, which is found under heads/build/coreboot-VERSION/src/mainboard/lenovo/x230/cmos.default after having build the board at least once with make BOARD=x230-VARIANT 

> git diff cmos.default > ~/heads/patches/coreboot-4.14/0061-cmos-settings.patch

Not sure what is attempted there, since there is no git tracking of change there. As a result, i'm pretty sure 
>git diff cmos.default > ~/heads/patches/coreboot-4.14/0061-cmos-settings.patch

Results in an invalid patch under 0061-cmos-settings.patch (path for patched file is necessary invalid)

From what I understand, you want to local build a rom containing cmos.default with the swap of FN and CTRL keys, by modifying x230 board. You should then do:
- make BOARD=x230-maximized (or x230-hotp-maximized)
- cp heads/build/coreboot-4.13/src/mainboard/lenovo/x230/cmos.default heads/build/coreboot-4.13/src/mainboard/lenovo/x230/cmos.default.orig
- modify heads/build/coreboot-4.13/src/mainboard/lenovo/x230/cmos.default
- make BOARD=x230-maximized (or x230-hotp-maximized)

By doing so, you are changing the defaults values that coreboot reverts to in case CMOS memory is wiped (CMOS battery being dead for example), and its default. The better way is through nvramtool as said https://github.com/osresearch/heads/issues/1087#issuecomment-1008085841 and https://github.com/osresearch/heads/issues/711#issuecomment-1008313982

Unfortunately, there is no way to modify heads/build/coreboot-4.13/src/mainboard/lenovo/x230/cmos.default only for one heads' board. (we could not create a x230-maximized-fn_ctrl_swap board, since that file is shared across all x230 board builds). 
This is why, unless desired by all x230 owners, this change needs to be applied by local builders for advanced users.

If you want to use Heads upstream without modification, you should :
- modify EC firmware by reflashing original lenovo firmware and following the thinkpad-ec project steps to swap keys.
- modify CMOS values at runtime, taken into consideration by coreboot and next flashes, through nvramtool as said previously with nvramtool and https://github.com/osresearch/heads/issues/711#issuecomment-1008313982. That is, modifying not the defaults, but what is kept alive under CMOS memory until the CMOS battery dies, hwere your swap would revert to cmos.default values.

Short:

1. Manually modifying cmos.default value will be valid only for local builds containing the change under heads' coreboot build directory, where board configuration file specifies the coreboot version to be used. You could create a local patch under patches/coreboot-version/NUMBER-keyboard-swap.patch but that patch could not be upstreamed nor applied to all x230 boards. 
2. You could go the thinkpad-ec way, so that your change is applied forever for that machine.
3. You could go the nvramtool way.

Why are you taking 1 as being the best solution for you?

> Is that correct? I am confused about "dirty".
Heads applies the dirty suffix for filenames and git tag of Heads included /etc/config to let the user know that there is something under its Heads tree that is being tracked but non committed. In your case, that dirty tag is related to the patch file that you created that is not commited to a branch.

Actually, Heads is right: your build environment is dirty. You mixed two ideas/goals here. [Wanting to build nvramtool ](https://github.com/osresearch/heads/issues/711#issuecomment-1008313982) and patching Heads' coreboot' cmos.default file to contain proper defaults containing your key swap, valid only for this local build. Patches are applied right after decompression of an downloaded archive per Heads build system into its relative module decompressed directory. 

So here on future builds, your modified cmos.default will be taken into consideration in building a rom. 
Your created patch file won't be applied unless the build/module relative directory (coreboot-VERSION here) is deleted.

I would still recommend going 2 or 3 above, not 1.

tlaurion, 2022-03-28:
@copyvar ?

closed 2022-03-28 by tlaurion