linuxboot/heads #1088: t430 wont boot after flashing

christmasisforturkeys, 2022-01-08:
I think I've got the wrong end of the stick here but following my interpretation of this comment https://github.com/osresearch/heads/issues/1056#issuecomment-968321625 I built coreboot 4.13 on ubuntu 20.04 and although I have a backup of my stock roms.. I thought I may be able to just run download_clean_me.sh from blobs and then build by using make BOARD=t430-maximized and flashing top spi & bottom spi externally. Needless to say it doesn't work

So am I right in thinking that I should; 

- run me_cleaner.py on my 8mb rom and flash that externally 

- then run make BOARD=t430-flash followed by make BOARD=t430-maximized 

- then externally flash t430-flash to my 4mb chip

- then flash t430-maximized rom internally from USB

also should I reflash to stock with my backups before this process?

Sorry I have tried to answer my own questions by going through the issues and reading and rereading the documentation only to grow more confused.

tlaurion, 2022-01-08:
> ntu 20.04 and although I have a backup of my stock roms.. I thought I may be able to just run download_clean_me.sh from blobs and then build by using make BOARD=t430-maximized and flashing top spi & bottom spi externally. 

This is exactly what should be done.

>  Needless to say it doesn't work

Please be more specific, since "it" is not helping troubleshooting. What is the expected behavior? What is the result? What t430? iGPU only? Nvidia?

christmasisforturkeys, 2022-01-08:
Oh so I got it right, well at least I don't feel foolish now.

Sorry I would have been more specific if I thought it was an issue with heads, I thought it was my own error in the process.

t430, flashed from completely from stock externally, me not cleaned, no dGPU

When I try to power on the machine the fan ramps up pretty loud for a second and then the machine powers off immediately, nothing on the display

tlaurion, 2022-01-08:
@christmasisforturkeys 

Please flash externally https://app.circleci.com/pipelines/github/osresearch/heads/366/workflows/88371e89-6f34-4042-b91c-cea8668e6699/jobs/1496/artifacts and report back

Please flash externally

christmasisforturkeys, 2022-01-08:
OK so circleci is like an auto builder thing?

No problem, just to clarify as there are a few files there, I should only need to flash ab top and ab bottom to the corresponding chips externally and then the whole rom internally if I can get it to boot? thanks

tlaurion, 2022-01-08:
> OK so circleci is like an auto builder thing?

Yes
> 
> No problem, just to clarify as there are a few files there, I should only need to flash ab top and ab bottom to the corresponding chips externally

Yes. Initial flash for maximized builds is external.

> and then the whole rom internally if I can get it to boot? thanks

No. Upgrades are then internal from the 12mb image, which is splitted in top and bottom files for external initial flashing.

christmasisforturkeys, 2022-01-08:
I have downloaded the bottom rom file but I can't seem to download the top rom file 'gfba79ab-top.rom' when I click it, it just displays the file on screen?

tlaurion, 2022-01-08:
Right click, save as

christmasisforturkeys, 2022-01-08:
same result with the link you provided, reflashing to stock now

christmasisforturkeys, 2022-01-08:
Nope same result again, I guess I messed up somewhere. The SOIC clip is a cheap Chinese one that I've had to hold on to the chip whilst reading and writing as its pretty crap and would just slip off otherwise, but I have been pretty meticulous in holding it perfectly still. although I have made multiple dumps and then reassembled and used the laptop in the meantime so the reading didn't do the damage just the most recent writing operation

I am only partially reassembling the laptop to test though, display cables, ram, bios battery, keyboard, cpu and fan could it be something else I am missing, perhaps a grounding issue? have any other t430 owners experienced anything like this? I'm probably clutching at straws but I'll try a full assembly tomorrow and report back if I get any joy.

Thanks for all of your help tlaurion, very much appreciated

tlaurion, 2022-01-11:
@christmasisforturkeys reflashing should give you a message from flashrom that content of SPI is exactly the same?

Having a good pair of clips is mandatory. Hopefully you haven't fried your board.

What did you use to flash?

christmasisforturkeys, 2022-01-11:
I used a Raspberry PI 4 and an SOIC clip that I got years ago on eBay cheaply, it is pretty poor but I have always managed to have success with it by holding it rigidly in place with my hand during flashing. So I'm really not sure what happened there, hate to think that I have fried a board with so much potential. 

I tried to diagnose the board with a multimeter but I couldn't find the schematics for the exact board type: NZM4I-8 only NZM4I-7 not sure how much difference that would have made, there probably isn't that much difference. I have all the tools needed for rework but I am just starting out with complex hardware really and have never worked on a board as complex as laptop motherboard, I usually just work on very well documented issues on simpler boards. I will keep the most likely fried board and diagnose it if my skills improve. 

Does anyone know what are the most likely components or areas to fail after a bad external BIOS flash? The board definitely gets power, the power light comes on and the fan ramps up for a second then it turns off, no light on the screen or keyboard apart from the power light briefly.

Yes, I reflashed multiple times and each time the content of the chips was the same, so the write was good but the board must have got fried somewhere in the process.

Luckily I was able to acquire another bricked motherboard with a corrupted Lenovo BIOS from a friend, (which turned out to be exactly the same model: NZM4I-8 NM-A082) and flashed that last night with the circle ci build you gave me and immediately I could see something was different, the light at the top of the screen flashed, the power light began blinking continuously accompanied by the microphone mute button flashing red and the fan ramped up. I power cycled that a few times with the same result each time then disconnected the CMOS battery and powered on and it booted right into heads, I then turned off reattached the CMOS battery and it continued to boot into heads without issue. 

After checking system information, only one stick of RAM was detected, so I flashed the full ROM that I built (not the circle ci build) internally via USB, merely as I had it already had it close to hand and then both RAM sticks were detected and I now have a fully working build and couldn't be happier.

one question about the order I should set things up

Am I right in thinking I should do the setup in this order;

1. reset the TPM, 
2. generate new TOTP 
3. add GPG key to the running bios and reflash
4. add a hard drive and begin to install qubes
5. enter heads and update checksums and sign all files in /boot
6. optionally use TPM to seal the secret used to unlock my LUKS volumes
7. optionally arrange LUKS headers be measured and the key re-sealed with those parameters

Thank you for all of your help once again, if there is anything I can do for community please let me know.

tlaurion, 2022-01-11:
@christmasisforturkeys please open issue under heads-wiki if https://osresearch.net is not clear enough.

You should insert hard drive, and Heads should propose to boot from USB and deal with generation of keys or import of keys.

Please close this issue if resolved.

closed 2022-01-11 by christmasisforturkeys