linuxboot/heads #1095: Can not boot after flashing x230-flash

vilaemail, 2022-01-23:
I followed instructions from [this website](https://osresearch.net/general-building/), here is what I did:
1. Built (`make BOARD=x230-flash` and `make BOARD=x230`) on Ubuntu 20.04.3 LTS server from commit b4b0bc4a7afa. Got following files "heads-x230-flash-v0.2.0-1135-gb4b0bc4-top.rom" with SHA-256=12DD7592B99B62920546A5B4D2D9975D969B064DA0A0B40BA7C80BFCC7B1EC82
 and "heads-x230-v0.2.0-1135-gb4b0bc4.rom" with SHA-256=7C50B1CDAB7F1461523072534A5D9D4553113BAA04C1EC20E0657D098EEF64C7
1. Updated bios to 2.76 using [Lenovo official iso](https://download.lenovo.com/pccbbs/mobiles/g2uj32us.iso) by following [these](https://osresearch.net/Clean-the-ME-firmware/) instructions. Did it by converting it to .img file, used `dd` to write it to USB drive, plugged it into X230 and followed on-screen instructions - it was successful.
1. Followed [this](https://osresearch.net/x230-flashing/) to flash the top (4MB) chip.
1. Followed [this](https://osresearch.net/Clean-the-ME-firmware/) to read bottom chip, unlock ROM, neutralize ME, and flash the bottom (8MB) chip.
1. Created ext4 partition on USB drive and copied "heads-x230-v0.2.0-1135-gb4b0bc4.rom" to it.
1. Put battery into X230, plug it into the wall and press button to power up.

Expected: Keyboard LED flash, and within a second the Heads recovery splash screen appears.
Actual: Keyboard LED flash. Power button keeps on glowing. Screen is black. Nothing happens :(

Any idea how to fix this? I am thinking about flashing original (backup) bottom chip and see if that works.
Did I do the right thing neutralizing ME *before* doing "full" flash with USB I prepared in step 5 or should I do that after the "full" flash?

vilaemail, 2022-01-23:
Flashing original (backup) to bottom chip did not help. I ended up building and flashing x230-maximized which worked as expected.

tlaurion, 2022-01-24:
Thank you for you report @vilaemail 

Will check, but hypothesis is that x230-flash doesn't enforce cmos.default. x230-maximized builts does, from memory.

Will review other boards.
This means that disconnecting CMOS battery would probably have resolved the issue.

tlaurion, 2022-02-05:
#1107 confirms lack of virtualization on legacy x230 builds when no black screen.

I said in the past that I was not really interested into legacy boards anymore. But will try to fix this

tlaurion, 2022-02-06:
Some weekend insights on this.
xx30-flash boards depend on coreboot-xx30 coreboot configs.

Those configs generate a 12mb ROM which is then generating 4mb ROM.
The boards build successfully because coreboot configs specify 7mb of CBFS available space: https://github.com/osresearch/heads/blob/master/config/coreboot-x230-flash.config#L4

I think its time to tinker with those configurations to change it to real CBFS space.

My theory here is that the space is so limited that maybe upgrading Tails public key makes the ROM span across 4mb.

If that's the case, then simply splitting the ROM as specified in xx30-flash board config's last lines is not sufficient anymore and results in faulty, incomplete xx30-flash 4mb ROMs.

Time to limit CBFS space for xx30-flash ROM in coreboot config, and that space be limited son that building xx30-flash fails instead of generating ROMs that won't boot.

As said previously in other issues, I personally do not see why keeping maintaining heads for legacy boards as relevant.

The other avenue would be to pass -O2 in global Makefile MAKE_OPTS to see if we can squeeze some additional free ROM space.

tlaurion, 2022-02-06:
Will test @fiedka :)

tlaurion, 2022-02-07:
Well, @fiedka was helpful into showing quickly the problem and showing used blocks in rom.

A new built rom atop of latest commit:
[image]

Locally used, old version (coreboot 4.8.1) x230-flash for current deployment:
[image]

Changing CONFIG_CBFS_SIZE to 4mb to test produces:
[image]

commit 065cbfda (2022-02-18): boards/xx30-flash: change board configs to be solely include flashrom module.
	Those boards now produce 4MB coreboot ROM and according CBFS small size, and remove the logic to extract 4Mb ROM out of the 12Mb rom which for some reason, was now misaligned.
config/coreboot-xx30-flash : remove all unnneded stuff to xx30-flash boards.
config/linux-x230-flash: used commonly for all xx30-flash boards, this is now finally saved with savedeconfig, and removes another bunch of unneeded stuff.

Tested working. Fixes #1095