linuxboot/heads #1103: cannot seal disk unlock key in tpm

lsdaok, 2022-01-31:
Expected:  seal luks key to tpm
actual: appears to accept luks key (which is a 100 character ASCII password) then asks for boot password which it seems to accept. Then it loops back the menu and reprompts for boot option again and asks if I want to make default again, asks for luks key again etc.

details:
Flashed heads on T430 all went as expected, although heads would not accept my public GPG key from USB, not a problem, I figured would do factory reset let heads generate new keys on my yubikey and export the public key to USB and use that one instead. This also failed, although after reading [1076](https://github.com/osresearch/heads/issues/1076) I ran forcesig and was then able to successfully OEM factory reset, thus generating a gpg key on my yubikey and exporting the public key to usb, I then did a test with a test qubes setup on SSD with luks password set to 123456. 

Although as I am not happy with default passwords/pin for yubikey, TPM (and obvioudly qubes luks password being 123456) I reinstalled qubes and setup luks with a 100 character ASCII password, reflashed ROM and erased settings. 

Since then I have tried manual setup multiple times with the same result, this is my process in order; 

I have reflashed and erased settings

reset tpm with its own 100 character ASCII password which it seems to accept. 

Then reflashed running BIOS with my public GPG key which it now accepts and flashes (previously it would read the flash then ask to write but not write anything, no error produced)

regenerated TOTP

signed all files in /boot

then to setup default boot: I go to boot options, select option 1 qubes with xen, select yes to seal tpm with disk unlock key and get the issue stated above. 

As my test with all default passwords and a short qubes luks password worked fine I guess it is either the password length or characters I am using somewhere, what is the maximum length and characters allowed for the tpm password and the luks key it can seal? thanks

tlaurion, 2022-01-31:
@rudolfocode 
1- Yubikey issue is #1076 as you found.
1- LUKS key ([Disk Recovery Key](https://osresearch.net/Keys/#disk-recovery-key)) != LUKS key (Disk Recovery key) passphrase. There is no reason for typing a passphrase at OS installation that would be so difficult to type there, since the key itself is derived before being stored in LUKS header. I always suggest Diceware passphrases of 6 words or more depending of your threat model. This passphrase is consequently easier to type and less prone to user error.
1- [TPM disk encryption key](https://osresearch.net/Keys/#tpm-disk-encryption-key) passphrase is setuped through setting a boot default, and is rate limited by the TPM. As opposed to the Disk Recovery Key, this key passphrase can be shorter.
1- The GPG User/Admin/TPM owner PIN are setuped through the OEM factory reset and are all the same. You can customize those at provisioning, while they are provisioned with conviviality in mind (being all the same) over security. You can always change those PINs as desired, after provisioning through gpg, within the OS or under Heads recovery shell.

The Disk Recovery Key passphrase, if eavesdropped, permits an evil maid to clone disk and decrypt it at leisure.
The TPM disk encryption key is valid only on your computer, and is considering firmware measurements, your public key, LUKS header and normal boot path (no recovery shell involved) to seal/unseal the TPM  NVRAM space when provided with the TPM disk encryption key passphrase. 

It seems that you entered an invalid character in the Disk Recovery Key passphrase/TPM disk encryption key passphrase that is invalid and Heads doesn't properly keep it in variables passed along in the shell scripts, resulting in error.

rudolfocode, 2022-01-31:
Thank you for your prompt reply Tlaurion and detailed descriptions, I will certainly reduce my password size for luks and use diceware as per your advice. 

> "It seems that you entered an invalid character in the Disk Recovery Key passphrase/TPM disk encryption key passphrase that is invalid and Heads doesn't properly keep it in variables passed along in the shell scripts, resulting in error."

I'm not sure what you mean here, if I made an error typing my password or if I am using a character, maybe something from ASCII that heads rejects as invalid, so I will answer both options

I'm pretty sure I haven't made a mistake with any of my passwords, I have taken great care in typing them and with the sheer number of times I have now done it, plus they always have been accepted when typing and confirming. I do honestly think it must be something regarding an ASCII character that is not accepted or perhaps the sheer length of the passwords I am using, what are the limits for characters and length?

I don't understand exactly what you mean by "is rate limited by the TPM" sorry, do you mean that there is a sort of timeout on retries?

how long do you suggest the password for the TPM owner PIN be? as I mentioned mine is currently 100 character alphanumeric with ASCII and seems to be accepted when resetting and taking owership of the TPM.

Thank you for your help

tlaurion, 2022-01-31:
> Thank you for your prompt reply Tlaurion and detailed descriptions, I will certainly reduce my password size for luks and use diceware as per your advice.
>
>
> I'm not sure what you mean here, if I [...]

For LUKS Disk recovery key passphrase, there is no limit in size AFAIK. When failing to use TPM disk encryption key passphrase, Heads will not try to unlock the LUKS container, and defer the unlocking to the operating system. This is where Qubes(Oses) will, as part of its boot process, ask for the passphrase and will deal with the disk unlocking.

When setting a TPM disk encryption key, Heads will use the Disk Recovery Key to add another LUKS key in slot 1, unlocking LUKS with slot 0 (Disk Recovery Key passphrase). What seems to be the problem here is the TPM disk encryption key, where for some reason, doesn't provide the same validation at seal/unseal for invalid characters? I'm unsure of what is the problem here. Will need to check code. Just to make sure, you are not using to the recovery shell to call scripts manually here? Doing so would take different TPM measurements at sealing/unsealing time here. And might as well be the reason why you have a problem. Not sure.

> 
> I don't understand exactly what you mean by:
> 
> > "is rate limited by the TPM"
> 
> sorry, do you mean that there is a sort of timeout on retries?

When attempting to unlock LUKS with TPM disk encryption key, user will have 3 attempts prior of being prompted to defer LUKS unlock to OS with Disk Recovery Key passphrase. the TPM will also refuse to release the TPM disk encryption key after a certain number of attempts (I don't recall how many as of now and should be documented once and for all). This requires waiting, or powering down the machine and emptying hardware capacitors to be able to try again. Those measures are really slowing down bruteforces from an external attacker. This is what rate limiting of the TPM disk encrypted key offers as opposed to the Disk Recovery Key. Once could clone disk at bruteforce at leisure a Disk Recovery Key, while the TPM disk unlock key can only be attempted to be unlocked from Heads.

Consequently, this is why the TPM disk encryption key is a strong security measure provided under Heads, refusing to prompt for the TPM disk encryption key passphrase only if firmware measurements are valid (including injected public key), when /boot integrity is valid (signed) and when LUKS header integrity is verified, when provided with valid TPM disk encryption key passphrase.
> 
> How long do you suggest the password for the TPM owner PIN be? as I mentioned mine is currently 100 character alphanumeric with ASCII and seems to be accepted when resetting and taking owership of the TPM.

I suggest TPM owner PIN to be the same as GPG Admin PIN, since it is never used outside of TPM ownership.
I also suggest the GPG Admin PIN to be of 2-3 diceware words, since GPG PINs are not ony rate limited, but will lock user out after 3 bad attempts. 
> 
> Thank you for your help

rudolfocode, 2022-01-31:
That's great info once again, thanks for such a thorough reply and I will implement all of your suggestions regarding password/pin length.

one note regarding keyslots: I can't remember the exact wording, as I am away from the laptop at the moment, but I will check asap and will report what heads displays exactly.. but I'm pretty sure just before/after when selecting the default boot and being asked for the disk  key (for the luks disc encryption, so that it can be sealed in the TPM) heads says something then says 'switching to keyslot 2' 

also is my manual process correct and in order?

1. reset tpm and setup password/pin
2. add gpg key to running bios and reflash
3. regenerate totp
4. sign files in /boot
5. set default boot and seal disk recovery key

tlaurion, 2022-01-31:
2 1 4 5

rudolfocode, 2022-02-01:
A foolish mistake, I can only think it was due to emulating the OEM factory reset order.

Thank you

tlaurion, 2022-02-01:
@rudolfocode please tell me if closed too early.

closed 2022-02-01 by tlaurion

rudolfocode, 2022-02-01:
No I got it to work, as a note for anyone else maybe experiencing the same issues though I can confirm the TPM does not accept sealing a Disk Recovery Key with ASCII characters.

Thanks again Tlaurion