linuxboot/heads #1107: x230 qubeos

ohrrkan, 2022-02-04:
Hello,

x230 with the Legacy last version of Heads

Installing qubeos 4.04 and get warning message missing HAP/SLAT/EPT/RVD  > how to activate them?

tlaurion, 2022-02-04:
@ohrrkan First thing first.

Under Heads main menu, hit the System Info menu.
You should see under there i7 or i5. If you have an i3, there is no way you can boot install Qubes with proper HAP/SLAT/EPT/RVD requirements.

That is, i3 laptops do not provide the proper CPU extensions.

tlaurion, 2022-02-04:
(Otherwise a bit confused you were able to have a functional x230 Legacy (x230-flash gives confirmed blackscreen at #1095)

tlaurion, 2022-02-04:
@ohrrkan I would be interested in knowing FW_VER you see under System Info (mostly the last segment after the last "-g" which corresponds to the commit ID  the ROM was build on top of.

ohrrkan, 2022-02-04:
It is the i5.
I was on skulls and the same Qubeos version before with virtual checked and activate so know that all good on the hardware part and Qubeos. 

sadly I do not read your message before now I get the black screen and cannot go in the sysinfo anymore.

This happen because to resolve my problem I have try to pass on the maximized version. see if this can resolve. I do it from internal with the flash menu proposal and flashing say everything work good but after the reboot black screen.

tomorrow I will try to re-flash it from external with the two (top and bottom) maximized version see I I resolve the black screen and then HAP/SLAT/EPT/RVD disable

ohrrkan, 2022-02-04:
Also for information I have compile my rom and do it from Debian 11. I precise this because see lot of people  who failed the compile with Debian 11 and say that you should go back on Debian 10 for resolving. 

This make me think that reading at you message of the config, do i have to precise something to take the coreboot /x230 config (other than BOARD=x230) when compile. Because this can be the explanation if he have take the basic coreboot config who have this parameter disable by default) is there a way to check (with the rom) after compile what was the defconf used?

tlaurion, 2022-02-04:
@ohrrkan I'm confused.

You should build passing
make BOARD=x230 
And you saying you build using something else then that is the thing to be known now.

How did you build this?

ohrrkan, 2022-02-04:
Yes I build it with make BOARD=x230 (for the legacy) and of course x230-maximized for maximized. My supposition inline with you note was that maybe I failed in another need parameter that make my rom have use the default coreboot config and not the good with the enable virtualization extension. 

Seems not the explanation.

ohrrkan, 2022-02-05:
Ok everything work after another external flash with the maximized.  No black screen all HAP/SLAT/EPT/RVD enable and qubeos 4.04 seems ok.

for information @tlaurion my heads build version is g4b26007.

if I have time in next month I will try to make a video or small tuto on how to bypass the difficulties that I get from the begin of the process.

tlaurion, 2022-02-17:
Confused here, since coreboot config is the same between x230 legacy and x230-maximized versions.
Also, all xx20 and xx30 boards share the linux-x230.config, where iommu kernel passed arguments (inside coreboot config) are deactivating igfx iommu for Heads' coreboot linux payload and where the kexec call to Qubes Xen and kernel is configured properly and hasn't changed (passed per grub).

Will need to review....

tlaurion, 2022-02-17:
Ok. So not sure why maximized versions are not having the same problem here. Testing

tlaurion, 2022-02-17:
So the conclusion here, testing on i5 laptop here, is that microcode is playing with results differently. 
maximized boards were different then legacy boards in the sense that maximized boards were including microcode updates where legacy were not.

A PR will point and close this issue where fixed. Should come soon as testing under my branch, will clean commits and link here.

tlaurion, 2022-02-18:
> Also for information I have compile my rom and do it from Debian 11. I precise this because see lot of people who failed the compile with Debian 11 and say that you should go back on Debian 10 for resolving.

@ohrrkan debian-11 support has been merged recently.

commit c1409a8b (2022-02-18): config/coreboot-xx30(legacy platforms): add back microcode updates that were preventing vmx to work on i5 without them.

Fixes #1107