linuxboot/heads #1110: Error compiling w530-dgpu-K1000m-maximized

zifxify, 2022-02-12:
I'm currently stuck at compiling an w530 image on Debian 11.2
Compiling an x230 image works...

7f89b099264ae15e72d423823f236f45f1cdf960237402ef3c6f8c4eafa33504  build/w530-dgpu-K1000m-maximized/initrd.cpio.xz
2022-02-12 13:27:43+01:00 MAKE coreboot
tail /home/zifxify/Documenten/HEADS/heads/build/log/coreboot.log
-----
    CC            firmware/lib20/api_kernel.o
    CC            firmware/lib20/kernel.o
    CC            firmware/lib/vboot_api_kernel.o
    CC            firmware/lib20/misc.o
    CC            firmware/lib/vboot_audio.o
    CC            firmware/lib/vboot_kernel.o
    CC            firmware/lib20/packed_key.o
    CC            firmware/lib/vboot_ui_legacy.o
    CC            firmware/lib/vboot_ui_legacy_clamshell.o
    RM            vboot_fw.a
    AR            vboot_fw.a
    CC            firmware/lib/vboot_ui_legacy_menu.o
    CC            firmware/lib/vboot_ui_legacy_wilco.o
    CC            firmware/lib20/api_kernel.o
    CC            firmware/lib20/kernel.o
    CC            firmware/lib20/misc.o
    CC            firmware/lib20/packed_key.o
    RM            vboot_fw.a
    AR            vboot_fw.a
make[1]: Leaving directory '/home/zifxify/Documenten/HEADS/heads/build/coreboot-4.13'
make: *** [Makefile:505: /home/zifxify/Documenten/HEADS/heads/build/coreboot-4.13/w530-dgpu-K1000m-maximized/.build] Fout 1
`
`

zifxify, 2022-02-12:
make BOARD=w530-flash is working. I have flashed this and there's also coreboot.rom when building. Should I flash this from within heads ? (I have unlocked the descriptor and ME regions with ifdtool and cleaned ME with me_cleaner)
What is the actual difference between w530-flash and w530-dgpu-K1000m-maximized and w530-dgpu-K1000m-hotp-maximized ? It's not exactly clear for me what I have to do next.

tlaurion, 2022-02-12:
You should run blobs/xx30 scripts to generate vbios and download and extract me, and then make the maximized  board corresponding to your GPU (dgpu or igpu).

@eganoa?

tlaurion, 2022-02-12:
CircleCI is building on Debian 11 now.

Here the host dependencies
https://github.com/osresearch/heads/blob/master/.circleci/config.yml#L16

Here the download and extract script for ME so that the blob is available when building xx30 boards:
https://github.com/osresearch/heads/blob/master/.circleci/config.yml#L97

Here the w530 script that downloads and extracts the vbios files so they are available when building board: https://github.com/osresearch/heads/blob/master/.circleci/config.yml#L107

Your original post doesn't show the real error, so my guess is that the blobs scripts were not ran prior of building w530-dgpu1000m

@zifxify does that help?

It would be nice if the documentation for those board additions were added under heads-wiki
@eganonoa ?

tlaurion, 2022-02-12:
On difference between boards, xx30-flash boards are meant for legacy configuration and is explained under https://osresearch.net (which renders heads-wiki)

Maximized boards take the freed ME space and make it available for maximized boards. Maximized boards also include generic GBE blob, required for Ethernet bring up and fixates the MAC address. The outcome is a fully external and flashable ROM image, not only reflashing the BIOS region of the SPI flash. This permits to take full advantage of the 11.5mb space available inside of combined SPI chips, as opposed to the really limited 7mb BIOS region of legacy boards counterpart. Legacy boards requires a two steps flashing process, flashing xx30-flash on top SPI (4mb image) and then to flash the whole limited BIOS region (7mb) defined under xx30 board.

When building a maximized board, 3 images are created. A full ROM (12mb) then splitted into top and bottom.
This doesn't just overwrite the BIOS region internally, but flash the whole image.

Maximized boards includes an unlocked IFD, which defines ME and BIOS region to maximized BIOS region.

Legacy boards will be limited in features in the future. If one opens hardware to flash externally, it is better to flash maximized boards then legacy.

Please help me document this properly under heads-wiki.

@zifxify @eganonoa @Thrilleratplay @Tonux599

We could also modify coreboot module to call the download and extract scripts to ease the process, while the reference is CircleCI creating those images and making them available to download from each successfully built boards, from gree n checkmarks of latest commits. Those can he downloaded and flashed directly.

zifxify, 2022-02-12:
@tlaurion thanks for clarifying the difference between boards

> There is no such board configuration. 

I was also using @eganonoa fork because main branch didn't compile the maximized build for w530.
I see now that there's is indeed no such board configuration in the main branch.

Running `./blobs/xx30/download_clean_me.sh -m $(readlink -f ./blobs/xx30/me_cleaner.py)`
works fine.

With following command `./blobs/xx30/vbios_w530.sh` I get this error

`/tmp/tmp.C1fmXlssw1/rom-parser-94a615302f89b94e70446270197e0f5138d678f3/VBiosFinder-c2d764975115de466fdb4963d7773b5bc8468a06/vendor/bundle/ruby/2.7.0/gems/terrapin-0.6.0/lib/terrapin/command_line.rb:82:in `run': Terrapin::CommandNotFoundError (Terrapin::CommandNotFoundError)
Verifying expected hash of extracted roms
./blobs/xx30/vbios_w530.sh: regel 65: cd: output: Bestand of map bestaat niet
sha256sum: vbios_10de_0ffb_1.rom: Bestand of map bestaat niet
vbios_10de_0ffb_1.rom: openen of lezen is MISLUKT
sha256sum: WAARSCHUWING: 1 getoond bestand kan niet gelezen worden
K2000M rom failed sha256sum verification...`

tlaurion, 2022-02-12:
As reported in separate issue for code of @eganonoa, there is a total lack of validation and script goes on without exiting on errors.

May I ask you if you are on latest commit? I changed his script to be compliant with Debian 11.

do git log under heads
And make sure you do a 
git pull master 
To be on latest commit.

I fixed this error like a week ago.

zifxify, 2022-02-13:
I'm on last commit and still getting these errors with vbios_w530.sh. Strange that it is working on CircleCI.

zifxify, 2022-02-13:
Seems to have the same issue: https://webcache.googleusercontent.com/search?q=cache:roU1lJZ6JQMJ:https://githubmemory.com/repo/thoughtbot/terrapin/issues/14+&cd=7&hl=nl&ct=clnk&gl=be&client=firefox-b-e

Still don't know what to do.

zifxify, 2022-02-13:
Script isn't working but I have the blobs downloaded from https://github.com/eric-pierce/Coreboot-W530.
Now I have successfully build an image for the w530.

tlaurion, 2022-02-15:
@zifxify #1111 fixes https://github.com/osresearch/heads/issues/1110#issuecomment-1037955019

@eganonoa : you have a local branch for documentation ready to push upstream?

tlaurion, 2022-02-20:
@zifxify can this issue be closed?

closed 2022-02-20 by zifxify