linuxboot/heads #1113: Error TPM deactivated - Lenovo W530 MTM 2447-6S8

zifxify, 2022-02-17:
There seems something wrong with the TPM-1.2 on my motherboard, not sure if this is a global issue.
Output of dmesg | grep TPM
`[    0.480076] tpm_tis 00:06: 1.2 TPM (device-id 0x0, rev-id 78)`
`[    0.544078] tpm tpm0: A TPM error (6) occurred attempting to read a pcr value`
`[    0.544108] tpm tpm0: TPM is disabled/deactivated (0x6)`

Reset TPM from GUI:

[image]

Under windows I get the following when opening tpm.msc :

`TPM_E_INVALID_POSTINIT 0x80280026 `
`"The command was received in the wrong sequence relative to TPM_Init and a subsequent TPM_Startup"`

tlaurion, 2022-02-17:
@zifxify power off. Retry ownership by resetting TPM.

Otherwise I have never seen that error on my side on x230.

You can ping board owners directly from pinned issue where board owners are specified there.

One question would be if you upgraded Lenovo firmware to latest version prior, and if the TPM was activated and locked down?

Can you revert back to stock ROM and check state from there?

Another question is how you are getting to windows?

zifxify, 2022-02-17:
> You can ping board owners directly from pinned issue where board owners are specified there.

Not sure how to do that

> One question would be if you upgraded Lenovo firmware to latest version prior, and if the TPM was activated and locked down?

I flashed back to latest stock firmware from back-up to make sure that  "Security Chip" was set active in BIOS.  
Next I cleared from coldboot the "Security Chip".
Under Windows TPM 1.2 is visible but I'm getting this error: TPM_E_INVALID_POSTINIT 

> Can you revert back to stock ROM and check state from there?

Have done that but it's also not working "correctly" in stock ROM I guess.

> Another question is how you are getting to windows?

Well I flashed back to stock first ;) and reinstalled Windows on a old SSD.

zifxify, 2022-02-17:
Output of tpm_version from tpm-tools debian 11.2

`TPM 1.2 Version Info:` 
`Chip Version:        1.2.13.12`
`Spec Level:          2`
`Errata Revision:     3`
`TPM Vendor ID:       STM` 
`Vendor Specific data: 50`
`TPM Version:         01010000`
`Manufacturer Info:   53544d20`

tlaurion, 2022-02-17:
Cannot help. Tag using @ users from this pinned issue : https://github.com/osresearch/heads/issues/692

tlaurion, 2022-02-18:
@eganonoa?

zifxify, 2022-02-20:
I disassembled my W530 to manually reset the TPM chip ( ST19NP18 - U39 on motherboard). 
This does not work for me as I just get a blank screen and I need to reboot to get the laptop working again.

[image]

[image]

[image]

zifxify, 2022-02-21:
Running tpm_takeownership from tpm-tools on Debian 11.2

`root@debian:/home/user# tpm_takeownership` 
`Enter owner password:` 
`Confirm password:` 
`Enter SRK password:` 
`Confirm password:` 
`Tspi_TPM_TakeOwnership failed: 0x00001087 - layer=tddl, code=0087 (135), I/O error`

`root@debian:/home/user# /usr/sbin/tcsd -f`
`TCSD resetting mode of /var/lib/tpm from 40755 to: 700`
`TCSD TDDL ioctl: (25) Inappropriate ioctl for device`
`TCSD TDDL Falling back to Read/Write device support.`
`TCSD trousers 0.3.13: TCSD up and running.`
`TCSD TCS Unloading a public key of size 0!`
`TCSD TDDL ERROR: write to device /dev/tpm0 failed: Timer expired`

`root@debian:/home/user# dmesg | grep tpm`
`[   13.022604] tpm_tis 00:05: 1.2 TPM (device-id 0x0, rev-id 78)`
`[  582.532287] tpm tpm0: invalid TPM_STS.x 0xff, dumping stack for forensics`
`[  582.532327]  tpm_tis_status.cold+0x19/0x20 [tpm_tis_core]`
`[  582.532336]  tpm_transmit+0x15f/0x3d0 [tpm]`
`[  582.532344]  tpm_dev_transmit.constprop.0+0x67/0xc0 [tpm]`
`[  582.532351]  ? tpm_try_get_ops+0x44/0x90 [tpm]`
`[  582.532356]  tpm_common_write+0x112/0x1c0 [tpm]`
`[  762.484413] tpm tpm0: Operation Timed out`

tlaurion, 2022-02-21:
Found some unfixed issue with same error, but would not understand why you have that issue.

https://www.spinics.net/lists/linux-integrity/msg16909.html

tlaurion, 2022-02-21:
My recommendation would be to GI back to stock ROM and try to reset the TPM from OEM's BIOS?

tlaurion, 2022-02-21:
Seems like its in an unknown state on which I don't have any input at all :(

zifxify, 2022-02-22:
> My recommendation would be to GI back to stock ROM and try to reset the TPM from OEM's BIOS?

Have done that, tried to clear TPM from coldboot and make sure it's set to active along with intel TXT.
Maybe @eganonoa knows something I haven't seen yet.

What's also not working is the fingerprint reader on stock bios. Although enabled it isn't detected. Not sure if that's related to the TPM to store fingerprint data ?

zifxify, 2022-02-23:
With IBM's TPM 2.0 TSS, I get the following:

`ln -s /dev/tpm0 /dev/tpmrm0`

root@debian:/usr/local/bin# tss1startup -v
TSS_Execute12: Command 00000099 marshal
TSS_Execute12_valist: Step 1: initialization
TSS_Execute12_valist: Step 6: calculate HMACs
TSS_Execute12_valist: Step 7: set command authorizations
TSS_Execute12_valist: Step 8: process the command
TSS_AuthExecute: Executing TPM_ORD_Startup
TSS_Dev_Open: Opening /dev/tpmrm0
TSS_Dev_SendCommand: TPM_ORD_Startup
 TSS_Dev_SendCommand length 12
 00 c1 00 00 00 0c 00 00 00 99 00 01 
TSS_Dev_ReceiveResponse:
 TSS_Dev_ReceiveResponse length 10
 00 c4 00 00 00 0a 00 00 00 26 
TSS_Dev_ReceiveResponse: rc 00000026
TSS_Dev_Close: Closing /dev/tpmrm0
startup: failed, rc 00000026
TPM 1.2 TPM_INVALID_POSTINIT - The command was received in the wrong sequence relative to TPM_Init and a subsequent TPM_Startup

zifxify, 2022-02-23:
Made some progress thanks to this site: https://www.electronicayciencia.com/2020/09/practicas-tpm-virtual.html.
Booted into a live environment Debian 11.2 and installed tpm-tools.
Now before running tcsd -f, we manually write following commands to /dev/tpm0 from user space:

**TPM_Init**
`python3 -c 'f=open("/dev/tpm0", "r+b", buffering=0); f.write(b"\x00\xC1\x00\x00\x00\x0A\x00\x00\x00\x97")'`

**TPM_Startup**
`python3 -c 'f=open("/dev/tpm0", "r+b", buffering=0); f.write(b"\x00\xc1\x00\x00\x00\x0c\x00\x00\x00\x99\x00\x01")'`

**TMP_PhysicalPresence -> CMD_ENABLE** 
`python3 -c 'f=open("/dev/tpm0", "r+b", buffering=0); f.write(b"\x00\xC1\x00\x00\x00\x0C\x40\x00\x00\x0A\x00\x20")'`

**TMP_PhysicalPresence -> PRESENCE_PRESENT** 
`python3 -c 'f=open("/dev/tpm0", "r+b", buffering=0); f.write(b"\x00\xC1\x00\x00\x00\x0C\x40\x00\x00\x0A\x00\x08")'`

**Start tcsd**
`/usr/sbin/tcsd -f`

Run **tpm_selftest** otherwise tpm_takeownership will complain about no selftest has successfully being run.
`tpm_selftest`

Run **tpm_takeownership** (no errors should anymore popup)
`tpm_takeownership`

Not needed but did a `tpm_clear`and rebooted into Windows 10.
TPM is also under Windows recognized and running get-tpm under powershell confirms this.
Not sure why it was in that "unknown state" in first place but it's finally solved now.
Only thing that isn't working at the moment is my fingerprint reader.
Is it related to my tpm for storing fingerprint data ? I don't know but it could maybe also be an hardware defect because it's not detected by "stock" bios.
Now I finally can start using "heads" with my new nitrokey ;-)

tlaurion, 2022-02-23:
This is awesome news and awesome debugging notes! Thanks you @zifxify !

Since it is noe resolved, would you mind modifying OP issue name to "Resolved: how to troubleshoot xyz TPM error", point original post to resolution on top of the posy and leave this open (so I can cleanup and point to this issue from the wiki one day?)

tlaurion, 2022-02-23:
For the fingerprint reader, I had numerous models having one and never even tried to have it work, so again sorry I cannot help with that.

But if windows on stock bios doesn't detect it, its probably not functional :/

zifxify, 2022-02-24:
This method isn't working anymore and I can not replicate correct behavior.

`python3 -c 'f=open("/dev/tpm0", "r+b", buffering=0); f.write(b"\x00\xC1\x00\x00\x00\x0C\x40\x00\x00\x0A\x00\x20"
); print(f.readall())'`

**TMP_PhysicalPresence -> CMD_ENABLE** returns `\x00\xc4\x00\x00\x00\n\x00\x00\x00\x03` and we expect `\x00\xc4\x00\x00\x00\n\x00\x00\x00&`

zifxify, 2022-07-22:
Probably I have set **TPM_PHYSICAL_PRESENCE_LIFETIME_LOCK** to TRUE which explains why I'm getting TPM_BAD_PARAMETER when trying to change permanent flags.
Either by sending:

`tpm physicalpresence -x 0x0080h (heads)`
`tpm_setpresence --lock (trousers)`

or by sending 

`python3 -c 'f=open("/dev/tpm0", "r+b", buffering=0); f.write(b"\x00\xC1\x00\x00\x00\x0C\x40\x00\x00\x0A\x00\x80")'`

I have done a lot trial of error so I can't remember exactly if  I had run one of the above commands when testing things out.
Can someone run the following command within heads to check if  "Physical Presence Lifetime Lock" permanent flag is also TRUE for your TPM.

`tpm getcapability -cap 4 -scap 108  (heads)`

tlaurion, 2022-07-25:
Sorry you have issues like that where I have no clue how to help

Here is output requested

[image]

zifxify, 2022-07-27:
@tlaurion thanks for your screenshot. Everything is the same with me except "Read Pubek".
We also have volatile flags we can check running `tpm getcapability -cap 4 -scap 109`
But these "values" are good I think.
"Physical Presence Lock" = FALSE and "Physical Presence" = TRUE
You can alter the state of "Physical Presence" by sending `tpm physicalpresence -x 8` (enable) or `tpm physicalpresence -x 10` (disable) as long as "Physical Presence Lock" = FALSE so don't send  `tpm physicalpresence -x 4`

But at this point I don't know what I can try next.
Maybe I buy another W530 which hopefully hasn't these weird tpm problems

Page 25 ( http://ww1.microchip.com/downloads/en/Appnotes/Atmel-8882-TPM-System-Design-Mftg-Recommendations-ApplicationNote.pdf )

TPM_PHYSICAL_PRESENCE_HW_DISABLE
0x0200h sets the physicalPresenceHWEnable to FALSE
TPM_PHYSICAL_PRESENCE_CMD_DISABLE
0x0100h Sets the physicalPresenceCMDEnable to FALSE
TPM_PHYSICAL_PRESENCE_LIFETIME_LOCK
0x0080h sets the physicalPresenceLifetimeLock to TRUE
TPM_PHYSICAL_PRESENCE_HW_ENABLE
0x0040h Sets the physicalPresenceHWEnable to TRUE
TPM_PHYSICAL_PRESENCE_CMD_ENABLE
0x0020h Sets the physicalPresenceCMDEnable to TRUE
TPM_PHYSICAL_PRESENCE_NOTPRESENT
0x0010h Sets PhysicalPresence = FALSE
TPM_PHYSICAL_PRESENCE_PRESENT
0x0008h Sets PhysicalPresence = TRUE
TPM_PHYSICAL_PRESENCE_LOCK
0x0004h Sets PhysicalPresenceLock = TRUE

tlaurion, 2022-07-27:
TPM reset is doing the following.
I still cannot wrap my head around your issue and what is not enough in tpm-reset

https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/bin/tpm-reset#L23-L34

zifxify, 2023-05-11:
Closing this as it only effects my W530 and  seems not reproducible on other W530 Thinkpads.

closed 2023-05-11 by zifxify