linuxboot/heads #1126: Error 1, Coreboot.org Certificate not trusted

grawawa23, 2022-02-25:
Hello everyone,
I am trying to build x230-maximized Board on Debian 10.4.0.
I have installed all the extra required packages.
Right at the start of building I get error 1, The certificate of coreboot.org cannot be trusted, the certificate is invalid.
I have already failed building on Debian 9 (had the wrong gawk and make versions), but this is a new error for me.
I have even tried and downloaded the coreboot-blob file with --no-certificate-check and put it in heads/packages
but the installer still gives out the same error.
I have also updated ca-certificates package on debian, but still no luck.
Any ideas of what I'm doing wrong?

Thanks in advance for your help!

```
root:xxxxxx....~/heads# make BOARD=x230-maximized
2022-02-25 15:00:13+01:00 Local built HEADS_GAWK only if different then provided by OS:
2022-02-25 15:00:13+01:00 Local built HEADS_MAKE only if different then provided by OS:
2022-02-25 15:00:13+01:00 Heads build system will call make from now on as: MAKE: make.
2022-02-25 15:00:13+01:00 WGET https://www.coreboot.org/releases/coreboot-blobs-4.13.tar.xz
--2022-02-25 15:00:13--  https://www.coreboot.org/releases/coreboot-blobs-4.13.tar.xz
Auflösen des Hostnamens www.coreboot.org (www.coreboot.org)… 2a01:4f8:121:1254::2, 78.46.105.101
Verbindungsaufbau zu www.coreboot.org (www.coreboot.org)|2a01:4f8:121:1254::2|:443 … verbunden.
FEHLER: Dem Zertifikat von »www.coreboot.org« wird nicht vertraut.
FEHLER: Das Zertifikat von »www.coreboot.org« ist abgelaufen.
make: *** [Makefile:505: /root/heads/packages/coreboot-blobs-4.13.tar.xz] Fehler 1
root:xxxxxx....~/heads#
```

grawawa23, 2022-02-25:
Hey tlaurion,
thanks for the quick reply!
I did apt update and apt upgrade and rebooted.
Now suddenly the build starts, takes about 30-40 minutes, and then at the end (or at least for me what seems like the end) I still get Error 1.

I will post the code below (I executed make BOARD=xxx a second time after the first build, still same error but now everything is much faster, the code below is from the second time), the os is in german, I hope it still makes sense.

I have tried building on Debian 11 bullseye, the latest stable release, a few days ago, didn't work.
Then, as mentioned above on Debian 9, still not working, and now on Debian 10.4.0 at least I seem to have the right make and gawk versions, but it stills delivers an error.

I really appreciate any help and comments 😃 

Thanks in advance!

```
root@:~# cd heads
root@:~/heads# make BOARD=x230-maximized
2022-02-25 16:57:19+01:00 Local built HEADS_GAWK only if different then provided by OS: 
2022-02-25 16:57:19+01:00 Local built HEADS_MAKE only if different then provided by OS: 
2022-02-25 16:57:19+01:00 Heads build system will call make from now on as: MAKE: make.
make -C "/root/heads/build/coreboot-4.13" CPUS=4 crossgcc-i386
make[1]: Verzeichnis „/root/heads/build/coreboot-4.13“ wird betreten
Welcome to the coreboot cross toolchain builder v ()

Building toolchain using 4 thread(s).

Target architecture is i386-elf

Found compatible Ada compiler, enabling Ada support by default.

Downloading and verifying tarballs ...
 * gmp-6.2.0.tar.xz (cached)... hash verified (052a5411dc74054240eec58132d2cf41211d0ff6)
 * mpfr-4.1.0.tar.xz (cached)... hash verified (159c3a58705662bfde4dc93f2617f3660855ead6)
 * mpc-1.2.0.tar.gz (cached)... hash verified (0abdc94acab0c9bfdaa391347cdfd7bbdb1cf017)
 * binutils-2.35.tar.xz (cached)... hash verified (6bdd090ce268b6d6c3442516021c4e4b5019e303)
 * gcc-8.3.0.tar.xz (cached)... hash verified (c27f4499dd263fe4fb01bcc5565917f3698583b2)
Downloaded tarballs ... ok
Unpacking and patching ...
 * gmp-6.2.0.tar.xz
   o gmp-6.2.0_generic-build.patch
 * mpfr-4.1.0.tar.xz
 * mpc-1.2.0.tar.gz
 * binutils-2.35.tar.xz
   o binutils-2.35_as-ipxe.patch
   o binutils-2.35_mips-gold.patch
   o binutils-2.35_no-bfd-doc.patch
   o binutils-2.35_no-makeinfo.patch
 * gcc-8.3.0.tar.xz
   o gcc-8.3.0_ada-musl_workaround.patch
   o gcc-8.3.0_asan_shadow_offset_callback.patch
   o gcc-8.3.0_gnat-bad_constant.patch
   o gcc-8.3.0_gnat.patch
   o gcc-8.3.0_gnat_eh.patch
   o gcc-8.3.0_libcpp.patch
   o gcc-8.3.0_libgcc.patch
   o gcc-8.3.0_nds32_ite.patch
Unpacked and patched ... ok
Building packages ...
Skipping GMP v6.2.0 for host as it is already built
Skipping MPFR v4.1.0 for host as it is already built
Skipping MPC v1.2.0 for host as it is already built
Skipping BINUTILS v2.35 for target as it is already built
Skipping GCC v8.3.0 for target as it is already built
Packages built ... ok
Copied EDK2 tools template ... ok
Cleaning up temporary files... ok

You can now run i386-elf cross GCC from /root/heads/build/coreboot-4.13/util/crossgcc/xgcc.
Welcome to the coreboot cross toolchain builder v ()

[... 101 lines cut ...]
3979157f7d201d43660328ac6826bf988dd5f8cba9929eeee67c2cba639ae98d  /root/heads/build/x230-maximized/tools.cpio
2022-02-25 16:57:55+01:00 HASHES    build/x230-maximized/tools.cpio
2022-02-25 16:57:56+01:00 CPIO      build/x230-maximized/heads.cpio
2022-02-25 16:57:56+01:00 UNCHANGED build/x230-maximized/heads.cpio
a42675ab5f238fe99deb056c33aa608c3a498d99df792623970e499e555e25f5  /root/heads/build/x230-maximized/heads.cpio
2022-02-25 16:57:56+01:00 HASHES    build/x230-maximized/heads.cpio
2022-02-25 16:57:56+01:00 MAKE coreboot
tail /root/heads/build/log/coreboot.log
-----
make -C "/root/heads/build/coreboot-4.13/x230-maximized" -C /root/heads/build/coreboot-4.13 obj=/root/heads/build/coreboot-4.13/x230-maximized CROSS=/root/heads/crossgcc/bin/i386-linux-musl- IASL=/root/heads/build/coreboot-4.13/util/crossgcc/xgcc/bin/iasl DOTCONFIG=/root/heads/build/coreboot-4.13/x230-maximized/.config BUILD_TIMELESS=1 CFLAGS_x86_32=-fdebug-prefix-map=/root/heads=heads -gno-record-gcc-switches -Wno-error=packed-not-aligned CFLAGS_x86_64=-fdebug-prefix-map=/root/heads=heads -gno-record-gcc-switches -Wno-error=packed-not-aligned -j4 --max-load 16
make[1]: Verzeichnis „/root/heads/build/coreboot-4.13“ wird betreten
    OPTION     option_table.h
    GEN        generated/assembly.inc
    CREATE     x230-maximized/mainboard/lenovo/x230/cbfs-file.zFDVaK.out (from /root/heads/build/coreboot-4.13/x230-maximized/.config)
#
# configuration written to /root/heads/build/coreboot-4.13/x230-maximized/.config
#
make[1]: *** Keine Regel vorhanden, um das Ziel „../../blobs/xx30/me.bin“, 
  benötigt von „add_intel_firmware“, zu erstellen.  Schluss.
make[1]: *** Es wird auf noch nicht beendete Prozesse gewartet....
make[1]: Verzeichnis „/root/heads/build/coreboot-4.13“ wird verlassen
make: *** [Makefile:505: /root/heads/build/coreboot-4.13/x230-maximized/.build] Fehler 1
root@:~/heads# 

```

tlaurion, 2022-02-26:
@grawawa23 
cat /root/heads/build/log/gawk.log

Shows more details?

grawawa23, 2022-02-26:
/root/heads/build/log/gawk.log shows:
```
/bin/bash: line 1: /root/heads/build/make-4.2.1/make: No such file or directory
```

Creating the directory hasn't helped, then it just says .... make: Is a directory

I ran the commands above, still no luck 😞

tlaurion, 2022-02-26:
Please rm ./packages/* and try again

grawawa23, 2022-02-26:
> rm -rf ./build/* ./packages/* and try again?

Nope same error I'm afraid

tlaurion, 2022-02-26:
rm -rf ./build/* ./packages/*
make BOARD=x230-maximized bootststrap V=1

tlaurion, 2022-02-26:
Seems like bootstrap is currently needed.

Will fix that.

For now `make BOARD=x230-maximized bootstrap`  prior of calling `make BOARD=x230-maximized` will fix this.

Trace:
```
user@heads-x230-clean:~$ rm -rf heads
user@heads-x230-clean:~$ git clone https://github.com/osresearch/heads
Cloning into 'heads'...
remote: Enumerating objects: 8447, done.
remote: Counting objects: 100% (195/195), done.
remote: Compressing objects: 100% (92/92), done.
remote: Total 8447 (delta 96), reused 159 (delta 85), pack-reused 8252
Receiving objects: 100% (8447/8447), 3.24 MiB | 239.00 KiB/s, done.
Resolving deltas: 100% (5565/5565), done.
user@heads-x230-clean:~$ cd heads
user@heads-x230-clean:~/heads$ grep apt .circleci/config.yml 
            apt update
            apt install -y build-essential zlib1g-dev uuid-dev libdigest-sha-perl libelf-dev bc bzip2 bison flex git gnupg gawk iasl m4 nasm patch python python2 python3 wget gnat cpio ccache pkg-config cmake libusb-1.0-0-dev autoconf texinfo ncurses-dev doxygen graphviz udev libudev1 libudev-dev automake libtool rsync innoextract sudo
            apt update
            apt install -y build-essential zlib1g-dev uuid-dev libdigest-sha-perl libelf-dev bc bzip2 bison flex git gnupg gawk iasl m4 nasm patch python python2 python3 wget gnat cpio ccache pkg-config cmake libusb-1.0-0-dev autoconf texinfo ncurses-dev doxygen graphviz udev libudev1 libudev-dev automake libtool rsync innoextract sudo
user@heads-x230-clean:~/heads$ sudo apt update
Hit:1 https://deb.debian.org/debian bullseye InRelease                
Get:2 https://deb.debian.org/debian-security bullseye-security InRelease [44.1 kB]
Hit:3 https://deb.qubes-os.org/r4.0/vm bullseye InRelease 
Get:4 https://deb.debian.org/debian-security bullseye-security/main amd64 Packages [121 kB]
Get:5 https://deb.debian.org/debian-security bullseye-security/main Translation-en [77.8 kB]
Fetched 243 kB in 2s (109 kB/s)      
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
9 packages can be upgraded. Run 'apt list --upgradable' to see them.
user@heads-x230-clean:~/heads$ sudo apt install -y build-essential zlib1g-dev uuid-dev libdigest-sha-perl libelf-dev bc bzip2 bison flex git gnupg gawk iasl m4 nasm patch python python2 python3 wget gnat cpio ccache pkg-config cmake libusb-1.0-0-dev autoconf texinfo ncurses-dev doxygen graphviz udev libudev1 libudev-dev automake libtool rsync innoextract sudo
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'acpica-tools' instead of 'iasl'
Note, selecting 'python-is-python2' instead of 'python'
Note, selecting 'libncurses-dev' instead of 'ncurses-dev'
acpica-tools is already the newest version (20200925-1.2).
autoconf is already the newest version (2.69-14).
automake is already the newest version (1:1.16.3-2).
bc is already the newest version (1.07.1-2+b2).
bison is already the newest version (2:3.7.5+dfsg-1).
build-essential is already the newest version (12.9).
bzip2 is already the newest version (1.0.8-4).
ccache is already the newest version (4.2-1).
cmake is already the newest version (3.18.4-2+deb11u1).
cpio is already the newest version (2.13+dfsg-4).
doxygen is already the newest version (1.9.1-1).
libelf-dev is already the newest version (0.183-1).
flex is already the newest version (2.6.4-8).
gawk is already the newest version (1:5.1.0-1).
gawk set to manually installed.
git is already the newest version (1:2.30.2-1).
gnat is already the newest version (10.1).
gnupg is already the newest version (2.2.27-2).
graphviz is already the newest version (2.42.2-5).
innoextract is already the newest version (1.8-1.2+b1).
libdigest-sha-perl is already the newest version (6.02-1+b3).
libtool is already the newest version (2.4.6-15).
libusb-1.0-0-dev is already the newest version (2:1.0.24-3).
m4 is already the newest version (1.4.18-5).
nasm is already the newest version (2.15.05-1).
libncurses-dev is already the newest version (6.2+20201114-2).
patch is already the newest version (2.7.6-7).
pkg-config is already the newest version (0.29.2-1).
python2 is already the newest version (2.7.18-3).
[... 581 lines cut ...]
	| tee "/home/user/heads/build/log/make.log" \
	> /dev/null
/home/user/heads/build/make-4.2.1/make -C "/home/user/heads/build/gawk-4.2.1/" -j2 --max-load 16 \
	2>&1 \
	| tee "/home/user/heads/build/log/gawk.log" \
	> /dev/null
user@heads-x230-clean:~/heads$ make BOARD=x230-maximized V=1
2022-02-26 17:06:03-05:00 Wrong OS deployed make detected: 4.3. Building and using 4.2.1...
2022-02-26 17:06:03-05:00 Wrong OS deployed gawk detected: 5.1.0. Building and using 4.2.1...
2022-02-26 17:06:03-05:00 Local built HEADS_GAWK only if different then provided by OS: /home/user/heads/build/gawk-4.2.1/gawk
2022-02-26 17:06:03-05:00 Local built HEADS_MAKE only if different then provided by OS: /home/user/heads/build/make-4.2.1/make
2022-02-26 17:06:03-05:00 Heads build system will call make from now on as: MAKE: /home/user/heads/build/make-4.2.1/make.
Makefile:262: warning: overriding recipe for target 'all'
Makefile:139: warning: ignoring old recipe for target 'all'
2022-02-26 17:06:03-05:00 WGET https://www.coreboot.org/releases/coreboot-blobs-4.13.tar.xz
if ! wget -O "/home/user/heads/packages/coreboot-blobs-4.13.tar.xz.tmp" https://www.coreboot.org/releases/coreboot-blobs-4.13.tar.xz ; then exit 1 ; fi ; mv "/home/user/heads/packages/coreboot-blobs-4.13.tar.xz.tmp" "/home/user/ [... cut ...]
Resolving www.coreboot.org (www.coreboot.org)... 78.46.105.101, 2a01:4f8:121:1254::2
Connecting to www.coreboot.org (www.coreboot.org)|78.46.105.101|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 38318160 (37M) [application/octet-stream]
Saving to: '/home/user/heads/packages/coreboot-blobs-4.13.tar.xz.tmp'

ads/packages/coreboot-blobs-4.13.tar.   8%[=====>                                                                  ]   3.23M   270KB/s    eta 89s    ^Cmake: *** [Makefile:505: /home/user/heads/packages/coreboot-blobs-4.13.tar.xz] Interrupt
```

grawawa23, 2022-02-26:
It worked! Thank you so much, without your help i'd never figured it out!
Thanks!!

closed 2022-02-26 by grawawa23