linuxboot/heads #1129: Mounting USB Invalid Argument, sh: 0: unknown operand

grawawa23, 2022-03-02:
Hello all,

trying to install Qubes 4.1.0 on Heads on a x230 i5 Board.
Inital external flashing through Skulls (top and bottom chip) then updated to Heads x230-maximized 12mb ROM (built on seperate laptop, and have also tried this through Flash/Upgrade BIOS in Heads with the latest CircleCI x230-maximized pre-built ROM)

I've added GPG keys, have a new TOTP and reset the TPM, but when I mount my usb to install Qubes (happens both when I have ISO file + asc copied BUT also when I dd just the iso on the usb stick) it always says:

```
mount: mount /dev/sdb1 on /media failed: Invalid argument
sh:0: unknown operand
!!! Something failed during USB boot
New value of PCR(4): xxxxxx
Starting recovery shell
```

Do you know what the problem could be?

System Info gives me:
```
Thinkpad X230-maximized
FW_VER: CBET 4000 Heads-v5.0.1-102-g7601400
Kernel: Linux 4.14.62-heads
CPU: Intel Core i5-3320M@2,60 GHz
RAM 8 GB (although I have 16 gb installed)

battery charge & health

Disk /dev/sda: 466GB
```

Thanks in advance for any help!!

tlaurion, 2022-03-02:
@grawawa23 hello there g7601400 refers to the commit id you are building from. Will edit.

grawawa23, 2022-03-03:
Thanks for the reply!
I have reflashed the bios via a heads internal update with my own x230 maximized build, following the steps described in the wiki and your help in (https://github.com/osresearch/heads/issues/1126) and now system info says 16gb ram and heads is booting fine (erased all the previous info and signed new keys & totp & tpm).

I have copied the Qubes-R4.1.0-x86_64.iso and the detached PGP signature (Qubes-R4.1.0-x86_64.iso.asc) from the qubes Website onto a usb stick via extFS for Mac on Apple M1 Macbook running Monterey 12.2.1 and like you said, it was infact the file system, I can now use the usb stick in heads (before was exfat, didn't work).

But now I get this when I boot from usb at the start at Verifying ISO:
```
Bad Signature from Qubes OS Release 4 Signing Key" [ultimate]
ISO signature failed
```
and then exit into recovery shell.
I have tried copying the .iso and .asc again onto another usb stick, same error.
When I verify the files on my macbook everything is fine.
😕 ❓

grawawa23, 2022-03-05:
I am now on `heads-x230-maximized-v0.2.0-1158-g70b93be` (reflashed using recovery shell not the upgrade option in gui) and am noticing the following:

Using ISO's for Qubes OS 4.1.0 but also 4.0.4 along with .asc files on the usb stick I always get BAD Signature, Verification of ISOs fails. The same with tails-amd64-4.27 iso & iso.sig

When I dd onto usb I get a freeze after `kexec_core: Starting new kernel` with both Qubes versions and also the tails distribution.

Sorry, but what am I missing? Have I overlooked something?

tlaurion, 2022-03-05:
@grawawa23 please copy the files to USB thumb drive and follow verification guides from either qubes website/tails.

If the verification fails, it is most probably because the file on the USB thumb drive is corrupted. Dd'ing on a bad USB thumb drive would explain your situation. Verifying from the drive you are copying from is no guarantee that what you are copying to (thumb drive) is good. The only way to test this is to verify integrity from the boot the media.

Integrity validation doesn't lie. There is at least one bit different, somewhere, on the iso. And dd'ing as I said is just trusting blindly integrity, which just complexities this whole process because of blind trust.

I reinstalled Q4.1 on a test machine yesterday from iso. The validation succeeded. Which is how I created the screenshots in the q4.1 PR over heads-wiki.

tlaurion, 2022-03-05:
@grawawa23 please open issue for extfat not being supported under Heads. I think it would be a good improvement.

grawawa23, 2022-03-05:
Thanks for the quick reply!

So, I used 3 USB sticks and 2 SD cards.
All formated as extfs4 with Master Boot Record.

I copied the files Qubes-R4.1.0-x86_64.iso and Qubes-R4.1.0-x86_64.iso.asc.
On every device authenticating the files on my laptop is success.

I tried all 5 devices in my x230 and get the following every single time:
```
Verifying ISO
gpg: Signature made xxx
gpg: using RSA Key xxxxxx
gpg: BAD signature from "Qubes OS Release 4 Signing Key" [ultimate]
ISO signature failed
!!!! Something failed during USB boot
New value of PCR[4]: xxxxx
!!!! Starting recovery shell
```

The RSA key is the same as when I verify on my laptop on the usb/sd device (obviously).
But still no success.

Do I have to import the Qubes master signing key or release key into heads or something beforehand?
I don't know what the issue could be, and since the files are verified on my laptop, 
I don't know why dd boot in heads is giving me the kernel error/freeze.

grawawa23, 2022-03-05:
This is the output if I dd the iso onto USB and choose Install_Qubes_OS_R4.1.0_[/isolinux/xen.gz]:
```
Loading the new kernel:
kexec -l /media/isolinux/xen.gz --command-line "console=none no-real-mode reboot=no vga=current" --module "/media/isolinux/vmlinuz inst.stage2=hd:LABEL=QUBES-R4-1-0-X86-64 plymouth.ignore-serial-consoles i915.alpha_support=1 rhgb intel_iommu=on intel_iommu=igfx_off " --module "/media/isolinux/initrd.img "
Starting the new kernel
[      45.809694] kexec_core: Starting new kernel
```
and then freeze

tlaurion, 2022-03-06:
@grawawa23 iso boot should work from internally maximized build and output of valid iso and iso.asc is given on https://github.com/osresearch/heads-wiki/blob/ba55a0d9dca814cbb0c4ec7dbb46dda21677c215/Installing-and-Configuring/install-os.md

Please take a screenshot of what happen when booting from iso, best being by launching usb-scan from recovery shell so that you have good output.

Please, if failing, take measurement from media there by doing a sha256sum of the iso to validate the hashes of the iso.

tlaurion, 2022-03-07:
@grawawa23 
> I am now on `heads-x230-maximized-v0.2.0-1158-g70b93be` (reflashed using recovery shell not the upgrade option in gui) and am noticing the following:

Retesting this [build](https://app.circleci.com/pipelines/github/osresearch/heads/390/workflows/3f2f722d-fc48-453e-b200-9cd940e32b4f/jobs/2474/artifacts) from this whole internally flashable ROM (taking for granted intial external flash of bottom and top ROM has been flashed once. No need to reflash externally unless you bricked something): https://2474-64810881-gh.circle-artifacts.com/0/build/x230-maximized/heads-x230-maximized-v0.2.0-1158-g70b93be.rom 
(flashed internally as a clean flash, not keeping config. Then re-add public key from wizard on next reboot).

[image]

[image]

 
> 
> Using ISO's for Qubes OS 4.1.0 but also 4.0.4 along with .asc files on the usb stick I always get BAD Signature, Verification of ISOs fails. 

Verification of the ISO from USB3 (blue tagged port) takes around 2 minutes:

[image]

"Starting new kernel" takes around 30 seconds prior of showing "Q" from Qubes ISO boot and then show normal installer. You should now be under normal Qubes installer.

Checking local iso integrity with sha256sum of Q4.1 iso should give the following:

[image]

Which matches Q4.1 DIGESTS file:
[image]

[image]

[image]

> The same with tails-amd64-4.27 iso & iso.sig
Validated against local media and confirmed working as well if integrity is good and time is good, and booting from first option proposed there.

Heads shows system time in UTC (GMT) timezone:
[image]

For me being UTC-5, the time and date is right (I have 5 hours less then the hour shown there. Time being saved in RTC clock in UTC timezone, not local time.)
 

- Please go into recovery shell and type `usb-scan` to boot from iso. Since you are launching this command from shell, the output of the error should be more meaningful.
- Give output of `sha256sum /media/*.iso`

We will go from there.

tlaurion, 2022-03-07:
For technical implementations details
- Distribution public keys are integrated under Heads [here](https://github.com/osresearch/heads/tree/master/initrd/etc/distro/keys)
  -  public key management (user's keys keychain and distro keychains being imported into distinct directories, where user's keys are at the end also considered as distro keys so ISOs signed by users can also be booted) is happening under [key-init](https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/bin/key-init#L5-L17)
- USB booting logic is implemented under [usb-scan](https://github.com/osresearch/heads/blob/master/initrd/bin/usb-scan)
  - Including booting from dd'ed drives. 
- USB ISO boot is implemented in this [script](https://github.com/osresearch/heads/blob/master/initrd/bin/kexec-iso-init)
  - As you can see, distro processed public keys are used to validate authenticity of iso.asc/iso.sig files where detached signature is used to validate integrity of the ISO.
  - If you get `ISO signature failed` [here](https://github.com/osresearch/heads/blob/c3b0bd6ffbe816430dd41ef54e649af52ed1ff3b/initrd/bin/kexec-iso-init#L21) it needs to be because the ISO is corrupted. This is why doing a `sha256sum /media/Qubes-R4.1.0-x86_64.iso` is expected to give a different result then `b9be56830d9b472a9de425b3e8fc70f75a18e3b21da61d2cb2c3849e25821608`, since GPG is considered reporting truth unless proven otherwise.

Hope those helps you and others facing same issue.

tlaurion, 2022-03-07:
@grawawa23 I created seperate issues to track them properly and renamed this issue to initial problem (so that exfat users find this issue). 

This issue is not solely for booting into detached signed isos.

tlaurion, 2022-03-07:
Same applies to Tails (4.26 iso and detached signature) on replicated test on same rom version, where kexec screen stays "frozen" for around 10 seconds.
[image]
[image]

grawawa23, 2022-03-07:
@tlaurion:
Thanks for separating the issues, sorry for the mess!
Sooo... I did three steps:

Checked the shasum256 of the downloaded iso, corresponds to .digests file.
(`b9be56830d9b472a9de425b3e8fc70f75a18e3b21da61d2cb2c3849e25821608`)
Copied on my usb stick and checked the usb stick file on my laptop, same shasum256, corresponds to .digests file.
(`b9be56830d9b472a9de425b3e8fc70f75a18e3b21da61d2cb2c3849e25821608`)
I plug the same usb stick into my x230 (blue usb3 port), calculate again under heads, get a different shasum256
(see screenshot)

[image]

**What is also weird, each time I calculate the sha256sum on heads for the same Qubes iso, I get a different value 😕 
I have tried all three x230 usb ports, always invalid iso, always different hash value.**

Heads date is correct, I am also on g70b93be...

[image]
[image]