linuxboot/heads #1133: Integrate T440p Support from Osboot

shmalebx9, 2022-03-07:
I saw that a few others are interested in heads for the t440p #975 and so I thought I should let you know that I recently added this board to [osboot.](https://notabug.org/osboot/osbmk) I wrote the infrastructure for blob extraction on osboot largely inspired by what I saw on heads, so I thought I'd let you know about my work over there as a thank-you. The only available download for the me (9.1) is listed in our blobs [sources.](https://notabug.org/osboot/osbmk/src/master/resources/blobs/sources) That ME isn't compatible with the ifd you'd get from a donor board. I created my own ifd by modifying one from a donor board, and you can find it in our source [here.](https://notabug.org/osboot/osbmk/src/master/blobs/t440p) That ifd will work with a truncated+relocated ME.

I also saw that there was a [worry over the mrc blob for this board.](https://github.com/osresearch/heads/issues/616#issuecomment-546331826) On osboot, we forked a version of [coreboot mrc downloader](https://notabug.org/osboot/osbmk/src/master/resources/scripts/download/mrc) which uses an archive.org download link. That script should make sure that you'll get reproducible builds without having to distribute the mrc.

Adding the T440p to heads shouldn't be difficult at all since we have everything ready to go on osboot. I could add it myself but we're far from release ready over at osboot, so I'm focusing my time there.

tlaurion, 2022-03-07:
@shmalebx9 awesome!

githubisnonfree, 2022-03-15:
shmalebx9's work is the bees knees. i'm very grateful for his osboot contributions

tlaurion, 2022-08-14:
Note that this board, for users tolerating mrc, and opened to tolerate  ACM init and ACM BIOS blobs could have TXT and measured boot with IBB encompassing bootblock.

That would mean that the CPU would be useable as a hardware root of trust, and PCR0 would be populated by the ACM, measuring what is defined to be part of IBB.

There was a failed experiment on x230 and a PR created to download and extract ACM bios and ACM init blob.

That PR will probably go nowhere for Ivy bridge boards, since there is no IBB support.

But the story is different for Haswell, which supports IBB measurement from rge ACM blobs.

Any t440p owner willing to replicate and push forward an equivalent PR for the t440p?

I know. More blobs. But also a hardware root of trust for those interested.

tlaurion, 2022-09-06:
Supporting t440p would require
- cloning board config, coreboot and linux config from platform being the nearest. Librem 13v1 or similar should provide similar configuration for Linux drivers (Linux comfig file)
- borrow and adapt osboot blob download script (mrc, refcode, me)
- borrow osboot ifd (can be distributed under gihub)
- borrow and adapt osboot coreboot config
- do a PR for maximized board additions (with without hotp)
- Iterate from there on github and link to t440p related support requests

tlaurion, 2022-09-06:
> I could add it myself but we're far from release ready over at osboot, so I'm focusing my time there.

@shmalebx9 there is interest from the community.

What is the status of the board under osboot?

shmalebx9, 2022-10-07:
> > I could add it myself but we're far from release ready over at osboot, so I'm focusing my time there.
> 
> @shmalebx9 there is some interest from the community.
> 
> What is the status of the board under osboot?

Sorry to get back so late. I have a job working in the dirt (not tech related) and a baby so time is tight. The board is stable in osboot. Other than the grub bug, it should be good to go. I have a t440p board which I could use to test with heads as a payload. Let me know if you're interested. I'm not super familiar with your build system, but I figure I could build a linux payload for testing with osbmk and update in this issue.

tlaurion, 2022-10-07:
So sad this mrc blob.

So sad work on reversing that mrc to support 16gb chips didn't go forward
https://github.com/eurus13/T440P-32GB/issues/1

Alright, checked the code for mrc download and me extract. Should not be hard to add coreboot config, deduplicate/extraction scripts and wrap it under circleci.

Seems like cos download is already dealt with by coreboot, didn't know https://doc.coreboot.org/northbridge/intel/haswell/mrc.bin.html#obtaining-mrc-bin

Cannot guarantee when I will have time to do that, would love to see interested people do it but might do it later on.

tlaurion, 2022-10-16:
@shmalebx9 this issue on skulls and upstream coreboot related partial fix for dGPU might be of interest for osboot as well. Again I do not own the hardware but t440p support still seems partial and including the board under Heads would result in lots of support requests I could not handle until it is known to work: 
https://github.com/merge/skulls/issues/220

PR #1282 (2023-02-28): Add initial support for the ThinkPad T440p
## Overview

I made the following changes to support the ThinkPad T440p:

- Add script for obtaining Haswell `mrc.bin` blob.
- Add script for extracting blobs from an original T440p ROM.
- Add Coreboot and Linux configs for the T440p.
- Add Heads configs for the T440p (`maximized`/`hotp-maximized`).
- Add script for obtaining a T440p `me.bin` blob from Lenovo's website.
- Add `ifd.bin` and `gbe.bin` blobs extracted from my T440p.
- Build everything on CircleCI.

I did not perform any research into T440p support that doesn't involve external flashing.  [... cut]
closed 2023-02-28 by tlaurion