linuxboot/heads #1135: t430-hotp-verification cannot detect libremkey

ItsAlwaysBeenWankershim, 2022-03-09:
So I recently upgraded from non-hotp t430 to the hotp-verification version. Flashing went very well, as did totp and hotp secret generation, with libremkey. However, when I boot, no matter which usb port I attempt to use, heads cannot find the security dongle. It prompts me to connect it, and hit retry. It appears as if heads does not initialize the usb ports before searching for the security key.

tlaurion, 2022-03-13:
Can you give version that shows under System Information please?

A commit made USB init happen earlier at boot when security dongle (hotp) is specified under board configs.

I would like to know if the flashed version was from before that commit or after.

ItsAlwaysBeenWankershim, 2022-03-15:
We're looking at Heads-v0.2.0-1156-g13a12d1

Commit 13a12d1

tlaurion, 2022-03-15:
Yep. You flashed the commit that cause a regression.

Please flash the latest commit.
Build or download from CircleCI latest build for t430-hotp-verification board.

Go to recovery hitting r frenetically at boot.
Type
flash-gui.sh
And flash from there, conserving settings.

Sorry about the inconvenience.

tlaurion, 2022-03-15:
It was fixed on next commit @ItsAlwaysBeenWankershim 

Being https://github.com/osresearch/heads/commit/ab6425cc7eabebb112312e870f9a060e9160611b#diff-9A66E05586B37D4779EE1EB88802E759

tlaurion, 2022-03-15:
It would be
https://2597-64810881-gh.circle-artifacts.com/0/build/t430-hotp-verification/heads-t430-hotp-verification-v0.2.0-1162-g9016ebc.rom

https://app.circleci.com/pipelines/github/osresearch/heads/392/workflows/110484b2-d799-4feb-a829-bff45f7849c4/jobs/2597/artifacts

From:

ItsAlwaysBeenWankershim, 2022-03-15:
No inconvenience incurred, I find it hilarious!

Confirmed, 9016ebc is beautiful and works flawlessly

closed 2022-03-16 by tlaurion