linuxboot/heads #1137: Setting a TPM Disk Unlock Key fails without error and is not user friendly.

tlaurion, 2022-03-10:
To fix:
- Initial setup of disk unlock key setups kexec_key_devices.txt
  - We can use cryptsetup to check for all blkid and check if we have a LUKS header and suggest those devices automatically so the user only enters it once, to be saved to be reused. 
  - That file contains encrypted devices/encrypted LVM(still not tested on my side)
  - That is, /dev/sdaX UUID in case of encrypted devices
- Next runs should not ask again to 
  - setup Encrypted LVM group if not defined under that file 
  - setup Encrypted devices
- It should actually prompt user to reuse already defined key devices under kexec_key_devices.txt and simply reuse them if user accepts
- Note that if Disk Recovery Key (passphrase should be added here) entered is bad, it will silently fails and go back to boot selection when trying to set a Disk Unlock Key, instead of looping, or at least stop so user can read error.
- Note that initial setup of TPM Disk Unlock Key is bad:
 - `Do you wish to add a disk encryption to the TPM` is just a wrong sentence and should be TPM Disk Encryption Key and passphrase. Otherwise we are training users to mix keys and passphrases and its already complicated here. If not ok with glossary, modify glossary otherwise we have no clue what we are talking about. I will also move glossary terms with Capitals.

Additionally, for traces:
 - Show boot options silently fails if signature check fails, going back to menu

Sirwhofromwhere, 2022-06-12:
This is what is happening to me, posted at the bottom of #1170 

How do I fix this?

tlaurion, 2022-07-27:
> What do you mean by signature check fails?

Heads checks and returns an error on console in between two fbwhiotail graphic output, hence the user doesn't see failing signature validation failing.

It requires the user to sign /boot from the menu.

Solution to this is to switch those "warn" to "error" messages and have an error logging function that waits for user input.