linuxboot/heads #1139: 16MB IFD on X230

githubisnonfree, 2022-03-15:
Hi,

If I recall, Heads does distribute IFD files for the default 12MB (8+4) setup on a ThinkPad X230. Other thinkpads might use a similar configuration.

I made a discovery recently. I upgrade 12MB flash to 16MB on X230's but there are 2 methods:

1) As defined in schematics, remove a bunch of resistors/caps and both flash ICs, replace just SPI1 with 16MB chip. System will be in single flash mode, and just configure regions in ifdtool and you're good to go

2) Replace just SPI2 (4MB) with a 8MB chip e.g. W25Q64FVSIG. This is physically much easier, but didn't work. I've recently got it working, and here's how:

Byte at offset 0x20 in hexedit, on the IFD. Default value is 0x25, change it to 0x2D. That's literally it. Then set regions accordingly (flreg base/limit) using ifdtool as usual. My project osboot, has IFD in this 2x8 configuration for X230.

Explanation:

In that byte, the least 3 significant bits define "component 1 density" and the 3 bits next to it define component 2 density. 5 means 8MB, 4 means 4MB. 0x25 is 00100101 -> 4 and 5, see? change 100 to 101, making the byte 0x2D

Only make this modification if actually upgrading to 2x8MB (soldering required). If using the other method, converting to single IC of 16MB, the component 1/2 density settings is a Do Not Care by the hardware.

Enjoy!

githubisnonfree, 2022-03-15:
this 2x8MB config is also possible on other ivy/sandy thinkpads probably, like t430, t530, w530

i think the S models (t430s, t530s if that exists, x230s) all are in single by default (one large 16mb chip, from factory)

this also works on x230i/x230t. i tested it myself

githubisnonfree, 2022-03-15:
yeah now you can fit vim in boot flash

tlaurion, 2022-03-15:
Really interesting @githubisnonfree, thanks for contributing this here!
On another note, as you may know, heads-wiki is the place to store documentation prior of PRs which is rendered at https://osresearch.net, while this heads repo is more for bug tracking of code/features so that PR can address those issues directly later on.

I'm pretty sure some people will be willing to desolder/solder as a service and might open the door to other contributors wanting to fit bigger things in the firmware. I am thinking once again of python3, thin-lvm-tools and the most dreamed project of mine being to make wyng-backups work directly under Heads, so that "safe state as a service" could be deployed/restored remotely. And then maybe even wifi tools, instead of requiring the user to connect with Ethernet in case something is needed to be done from measured firmware state in Heads recovery shell.

> yeah now you can fit vim in boot flash
vi is there thanks to busybox :)

As you may know from this project (I think you take advantage of ME freed space by default is osboot project) 11.5mb is still enough for Heads on xx30,, where your suggestions would open the door more advantageously to xx20 models, which are already struggling with their 8mb single SPI flash chip, suffering from features reduction. 

Heads requires TPM for its added security features (including TPM Disk Unlock Key), so unfortunately that would not fit the bill for GS505s (?) that is long waiting for basic Heads support through HOTP flashrom based measurement (that is another topic). If we talk of models which would benefit from Heads detached signed ISO boot (Tails iso and iso.asc being on usb drive) I'm also thinking of this being beneficial from RYF certified x200 and other GM45 models (gpg toolstack is an heavy beast from a ROM fingerprint perspective, so is FBWhiptail and cryptsetup).

That's about it! This issue can now be tracking improvements that would target 8mb and smaller SPI machines, with/without virtualization support to support more software features, while some other models not supporting Qubes (xxx0) would  then provide enough space to go diskless and be used by whisleblowers not needing data persistence/compartmentalization, where amnesiac operating systems are sufficient.

githubisnonfree, 2022-03-21:
tlaurion, i've discovered something else. when you do 2x8MB setup, it works, and boots properly, but if running flashrom -p internal afterwards, it won't work. flashrom will report it still as 12MB, even though it's 16.

So: 2x8MB works, but means you'll only be able to externally re-flash later on.

if you want internal flashing capability, you should do 1x16MB; if doing *that*, there are some resistors to remove also. the schematics say which ones. those resistors are either placed, or not placed, to tell the pch whether it's 12MB or not; if not, and the resistors (plus 1 cap) aren't placed, then you install either a single 4MB IC, or a single 8MB IC (X220 has the single 8MB IC and those resistors aren't placed. X230 by default has 8MB+4MB and has those resistors placed)

i wrote about this today on twitter, and thought i'd warn people here too, on this issue page

the good news is that 2x8MB *does* work on X230, and it's absolutely OK so long as you are prepared to only flash externally afterwards. i assume you'll want to flash protect anyway, and if you're handy with a soldering iron to do the upgrade anyway, you won't mind external reflashing!

the IFD with the 0x25 byte changed to 0x2D should also still work on a 1x16MB setup, with those resistors (as defined in schematics) not placed

githubisnonfree, 2022-03-21:
[
[image]
](url)

githubisnonfree, 2022-03-21:
if you attempt internal flashing after doing 2x8MB, the hardware still reports to flashrom on host cpu, that you have 12MB of flash

there might be a way to overcome this, but i don't know of a way yet, if it even exists. i'm going to assume it's not possible

githubisnonfree, 2022-03-21:
the fact that the machine does in fact boot, but flashrom reports 12MB, tells me to maybe check flashrom. it could be that flashrom is at fault here, but that the configuration i established *is* technically correct.

I need to investigate it further.

fhvyhjriur, 2022-06-07:
Side note with my words:
I installed in a x230 two 16MiB SPI chips by replacing the 4MiB and the 8MiB SPI chips. To not have to deal at the moment with software configuration in any kind, i simply create two 16MiB ROMS. One top and one bottom. For the 4MiB i have now 4 times the same content of the 4MiB heads-maximized part to fill it up to remove the flashrom warning because image is too small. For the 8MiB SPI i have twice the content of the 8MiB heads-maximized image for the second SPI chip.
I got told, that the Intel chipset can address at max 16MiB. You have to configure in the IFD where those 16MiB are. That is done like now with the 12MiB that are split on two chips.
This dual 16MiB setup helps me to not have to deal with anything else (software or hardware(resistors)) and when in the future some release is out that could make use of 16MiB instead of 12MiB i dont have any compatibility issues in any sort. I can now use and update coreboot/heads with flashrom without any issue still using the 8+4MiB release. If a future release need two SPI chips with each 8MiB it would still be working with my dual 16MiB chips setup. If a release would be out there that make use of one single 16MiB SPI chip and dont use the other, then my setup with two 16MiB chips is also working.

Conclusion: If someone would like to do some testing with images that use more space then 12MiB and would like to play around with the IFD in different ways, just install in any coreboot thinkpad you own 8Pin-16MiB chips. Then you have the max compatibility with every possible future setup.

What is nice: You NEVER have to deal any more with ANY preinstalled closed source thing on the SPI chips. Coreboot/Libreboot/heads installation is done by preparing the loose SPI chip (no in circuit flashing needed any more). After the SPI chip have the software you would like to use, you put it in your Thinkpad with soldering iron. Like you remove the old spinning disk drive and dont care what was there installed (you never power up the device with it), its the same with libreboot 16MiB images for the gm45 devices https://github.com/osresearch/heads/issues/925 or the newer devices that you have heads-maximized images for. Get and thinkpad and never turn it on. Remove any memory (Hard disk + SPI), install the SPI with the content you like and then insert a empty SSD you like and install on it the content you like.

tlaurion, 2026-09-17:
@githubisnonfree is it something you sell preinstalled? If so, please reopen this issue, if not, sorry for DIYs: cost of maintainership and building that variant alongside all others existing doesn't make sense.

If you disagree, please reopen

CC @fhvyhjriur

closed 2026-09-17 by tlaurion