linuxboot/heads #1146: T440p: Easy 5-minute flashing method

githubisnonfree, 2022-04-03:
Hello everyone. Leah here again from osboot/libreboot. I have made a breakthrough:

First, see this thread: https://github.com/osresearch/heads/issues/1143

Now:

Regarding T440p, see pics.
[image]
[image]

1) Remove 2 screws for big door
2) Remove big door
3) Flash 4MB chip with upper 4MB part of ROM with 4MB CBFS
4) Strap HDA_SDO. Just solder a wire*
5) Boot Linux. Flash full 12MB, with >11MB CBFS

Nailed it. Could do this on X230 too, but kinda silly there, since both chips are accessible immediately. With T440p, only SPI2 is immediately accessible, you have to remove the mainboard to access SPI1; this method above saves time by avoiding that. 5 minutes instead of 20-30.

*IFD flash protection is useless anyway. Can just short HDA_SDO easily. Setting Protected Range registers and enabling /dev/mem protection in Linux gives better flash protection.

I also wrote about it in this tweet:

https://twitter.com/n4of7/status/1510761816137347075

Regarding those test points shown in the pic: just remove the HDD/SSD and peel back the tape underneath, to access them. Easy!

Instead of soldering a wire, you could just short with tweezers, turn on then let go. Or maybe hold to them using pogo pins, but meh.

Anyway, just thought I'd share this here. Enjoy!

githubisnonfree, 2022-04-03:
This photo shows the tape being peeled back. I have the board removed in these photos, but you can access this part of the board by just removing the "big door".
[image]

githubisnonfree, 2022-04-05:
[image]

All you need is tweezers.

tlaurion, 2022-04-05:
Hi @githubisnonfree! Could you edit OP to counter-say initial sayings?

githubisnonfree, 2022-04-05:
i'll just find a working method and re-post later

githubisnonfree, 2022-04-05:
[image]

these two pic
[image]

find hda chip, short the hda_sdo pin (pin 5) to dvdd (pin 9 according to schematics i have in front of me)

you'll find the hda chip in roughly the same place as the pics above, but from the other side of the board. the hda_sdo test point (not actually a test point, a via but whatever) is correct but the 3.3v point is not hda dvdd. you want to short hda_sdo to dvdd, but i couldn't find a test point in an easily accessible place. i will try again at a future date, when i get time again

i didn't cut any of the solder mask on the vias that are covered, it's possible that one of those is dvdd
[image]

i've attached a 3rd photo showing the hda chip

githubisnonfree, 2022-04-05:
in my original post, i assumed just using any old 3.3v point would be fine, to pull hda_sdo high. for some reason though didn't work. i found the 3.3v point by just probing around while the system booted, but i had no idea if it was actually a power rail; for example, what i found could have been a signal pulled high via pull-up resistor

i don't have the boardview so i'm limited to just cross referencing schematics with... gutting a donor board to use as a sacrifice, while i buzz around to then test on another board

githubisnonfree, 2022-04-05:
either hda_sdo can't be strapped to disable ifd protections / intel me after buy, OR: or i just screwed up. i don't know which is the case, but it's one of these

githubisnonfree, 2022-04-05:
actually, look at that "bubble" of solder coming off on the track on pin 9 on hda

looks like a via that has solder in it. probably going to the other side of the board, and probably accessible. but probably one of the vias that's covered in soldermask

i will have to just guess which one(s) to check and scrape away solder resistor. will try it at a later date, and then report back

if my hunch is right, it is possible, but you'd need to scrape off soldermask to short hda_sdo to dvdd (after removing the tape under the ssd)

githubisnonfree, 2022-04-09:
This is copied from my report on X230 flashing via https://github.com/osresearch/heads/issues/1143 and I'm going to assume it's the same or worse on T440p:

I discovered: first 8MB theoretically flashable internally (PR regs didn't protect it), but even with HDA_SDO strapped, I still couldn't flash it internally from lenovo's firmware; you'll get errors and if you try to reboot afterwards, the result will be a brick.

The best way is this: https://doc.coreboot.org/mainboard/lenovo/ivb_internal_flashing.html

Then strap HDA_SDO and erase/write the whole flash.

Basically, the conclusion of my research is that it's easier and cheaper in terms of time investment to simply reflash externally with an SPI programmer like the Raspberry Pi.

This is cheaper overall in terms of time, especially if you're flashing a lot of of machines like I do.

That said:

If a method is discovered for T440p to do internal flashing to boot coreboot, and an easy way to strap HDA_SDO can be figured out (and it works), then that might be 5-10 minutes of work, versus 20-30 minutes of work doing disassembly and reassembly on these machines.

I've concluded my research on these. I will not be investigating further, for the time being. If someone else wants to carry on where I left off, please feel free. I will be pleasantly surprised if someone proves my assertion wrong, but I do assert:

Simply flashing externally is easier, and you might aswell just do that.

tlaurion, 2022-07-14:
Not aware of what is directly exposed on the t440p, but the following haxkncoukd also apply as to t530 and w530 https://github.com/osresearch/heads/issues/1189

Let me know @githubisnonfree