linuxboot/heads #1152: w530-dgpu-K2000m-hotp-maximized heads build not initializing /boot

walliams, 2022-04-13:
just attempted an updated build. did a full reset, wipe settings, and verything seemed to work until it got to the point where it would normally update /boot checksums-- it skipped to the prompt where it shows me my new TPM/key passwords. after reboot and new secret, the hotp passed, but i was unable to boot, not even if i chose "Ignore Tampering and Force Boot". it was just skipping back to main menu.

tried multiple OEM Factory Resets, same issue. Reflashed with an older version of the same board cloned from git on 3/22/22 and it works fine. so it appears that something pulled in the past 3 weeks broke boot for me.

tlaurion, 2022-04-13:
@walliams Hey there. Can you give the ROM image full name you do not see the problem with? The commit ID is part of the filename, g***********.rom where the stars are corresponding to the commit ID.

In the last weeks some changes happened directly in the OEM factory reset/Re-Ownership script.

From what you report, it seems that an issue is happening that is not being reported to the user:
> until it got to the point where it would normally update /boot checksums-- it skipped to the prompt where it shows me my new TPM/key passwords. 

The OEM script corresponding to that step is happening at:
https://github.com/osresearch/heads/blob/f3d49246467bab4b44a5bc058a16cd14952fc70e/initrd/bin/oem-factory-reset#L675-L679 

Which should give graphical errors in case something goes wrong:
https://github.com/osresearch/heads/blob/f3d49246467bab4b44a5bc058a16cd14952fc70e/initrd/bin/oem-factory-reset#L150-L223

So i'm a bit in the dark here. Will reflash master CI [build](https://circleci.com/gh/osresearch/heads/3044?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link) for x230-hotp-maximized, retest and reply here.

tlaurion, 2022-04-13:
Found an unrelated bug in master as of 1dd03d0361ad572d8649b5258e049b8e02d4e508, will push corrective (setting a new boot default was not having a functional menu, failing into recovery shell).

On my tests (no LUKS reencryption, no LUKS passphrase change, single custom passphrase, exporting public key to usb drive) I was not able to replicate your issue. Can you be more specific on the issue?

The OEM factory reset wipes /boot/kexec* files. So no more digest of /boot nor default boot options prior of finishing generate_checksums. Then since we just generated a new keypair, and since the OEM factory reset script knows GPG User PIN, it is generating the digests and signing them, unattended. The next screen is the provisioned secrets screen.

On reboot, HOTP/TOTP needs to be regenerated, its normal since we just modified the firmware by injecting a new public key.
Reboot. The system is then configured to boot default grub boot option without a LUKS disk unlock key passphrase. Enrolling such will require, as of now, to go into boot options and to set a new boot option.

Where does your experience differ?

walliams, 2022-04-13:
thankyou for taking a look @tlaurion. 
heads-w530-dgpu-K2000m-hotp-maximized-v0.2.0-1162-g9016ebc-dirty - was the working rom.
heads-w530-dgpu-K2000m-hotp-maximized-v0.2.0-1190-g1dd03d0-dirty - is the build I just did, not working

both dirty i think because am using an extracted gbe.bin for the stock MAC. if i ever want to run both my thinkpads at once, i'm assuming this is for the best so the DEADC0FEE MAC addresses don't confuse the network.

> Where does your experience differ?

between the working rom I have and the not-working-- on the working one, after the 3 minute gpg generation, i am able to see a few lines notifying me that it is generating the digests and signing them before it moves to the provisioned secrets screen. on the build i just did yesterday, it seems to skip over that part-- i see the lines appear and then it instantly jumps to the provisioned secrets page, as if there was an error generating the digests and it just skipped, but i'm not certain.

then i did try to set a new default for boot, and tried to force an unsafe boot, and each time it would return me to the main menu immediately. when attempting default boot, even with the proper boot sector selected from the boot option menu, i saw an error and it dropped me into recovery shell. from there i was able to see my headers in /boot, and was able to mount /dev/sda1 to /tmp/boot as read only and see the same files there.

tlaurion, 2022-04-13:
@walliams dirty means you have differences in your git tree. Since gbe.bin is part of the tree, modifying gbe.bin (extracting it) explains that behavior.

Could you do a `git pull origin` then a `git diff` and make sure nothing else then gbe.bin in tree is different?

From your reported behavior, it seems that something fails between generating the keypair, flashing the public key inside of the firmware and generating checksums, which I was not able to reproduce on my tests, and where the w530 doesn't differ in code outside of coreboot internals.

Pictures speak better then words, while:
- "Options -> GPG menu -> List GPG keys in your keyring" should show your public key information
- "Options-> Update checksums and sign all files in /boot" should generate digests and sign /boot options
- Options-> Boot options-> Show boot options should validate that signed /boot digests are valid and propose you boot options from there.

Where does it hang?

tlaurion, 2022-04-13:
> seems my ifd.bin is also unique, probably from my initial rom dump & extract sometime last year. other than that, everything else checks out.
> 
> just to be safe i will do a clean build, run more tests and report back
> 
> > * Options-> Boot options-> Show boot options should validate that signed /boot digests are valid and propose you boot options from there.
> 
> when i tried this, it just bounced me back to the main menu. i was never able to see my kernel headers from the menu, only in recovery shell did i see anything in /boot

So generating checksums and signing them is faulty, which should be able to be done manually from options shown above:
- "Options -> GPG menu -> List GPG keys in your keyring" should show your public key information (otherwise flashing public key in rom is faulty)
- "Options-> Update checksums and sign all files in /boot" should generate digests and sign /boot options (which would fail if no public key as well).

walliams, 2022-04-13:
well, i just flashed v0.2.0-1192-g3201cd4-dirty and it took me to boot. as a sanity check, i reverted back to v0.2.0-1190-g1dd03d0-dirty, keeping settings, and every boot option errored to recovery shell.

now i'm back on the latest and it is working, so i guess we can close this? i'm a little confused as to what is different, but at least it is working. 

note: i did swap the ifd.bin for the stock blob in the repo, keeping only my custom gbe.bin

> * "Options -> GPG menu -> List GPG keys in your keyring" should show your public key information (otherwise flashing public key in rom is faulty)
> 
> * "Options-> Update checksums and sign all files in /boot" should generate digests and sign /boot options (which would fail if no public key as well).

also note: i did try both these options, was able to see my public key, and yet the second option bounced me back to the main menu. but i guess this is moot, since my latest build is now working properly.

let me know if there are any other tests you'd like me to run

closed 2022-04-13 by walliams

tlaurion, 2022-04-13:
@walliams g1dd03d0 was faulty, since selecting a boot option was going to shell. I pushed a commit to master, correcting that specific bug today.

But this has nothing to do with the error linked to your original posted problem, which was linked to impossibility to either flash new public key inside of the firmware and flash. The ifd.bin that you had may have been the culprit. I remember bad behaviors in the past with flashrom failing to flash (under flash.sh) used by flash-gui.sh, where no error was reported to user.

This seems to be a candidate to #1149 (or adding an error function that waits for input from user to confirm error having been view prior of continuing, which warn is just doing: outputting a warning that should be non-consequential to the console.) If what we have is an error, we should stop at the console for user acknowledgement, not continue as if nothing happened. 

As of now, this issue seems resolved, without being clear of what was the cause of it.

walliams, 2022-04-13:
well, i think i just ruled out ifd.bin as the culprit... did another g3201cd4-dirty build with unique gbe and ifd blobs and it boots fine.

so, i'm relatively certain the last two commits resolved the issue. i just don't understand how.